bdring/FluidNC #671: Problem: SD not work.

DEST1981, 2022-10-15:
### Controller Board

DIY board 4-axis. But setup to 3 axes
[image]

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

External driver TB6600, PWM spindel. Limit switches pull up to +3.3v directly connect.

### Configuration file

```yaml
name: "ESP32 Dev Controller V4"
board: "ESP32 Dev Controller V4"

stepping:
  engine: RMT
  idle_ms: 0
  dir_delay_us: 1
  pulse_us: 10
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.13:high
  
  x:
    steps_per_mm: 100
    max_rate_mm_per_min: 2500
    acceleration_mm_per_sec2: 200
    max_travel_mm: 400
    homing:
      cycle: 2
      positive_direction: false
    
    motor0:
      limit_neg_pin: gpio.35:low
      standard_stepper:
        direction_pin: gpio.22
        step_pin: gpio.21
    motor1:
      null_motor:

  y:
    steps_per_mm: 100
    max_rate_mm_per_min: 2500
    acceleration_mm_per_sec2: 200
    max_travel_mm: 400
    homing:
      cycle: 2
      positive_direction: false

    motor0:
      limit_all_pin: gpio.34:low
      standard_stepper:
        direction_pin: gpio.17:low
        step_pin: gpio.16
    motor1:
      null_motor:

  z:
    steps_per_mm: 100
    max_rate_mm_per_min: 2500
    acceleration_mm_per_sec2: 200
    max_travel_mm: 220
    homing:
      cycle: 1
      allow_single_axis: true
      positive_direction: true
      mpos_mm: 1.000
      seek_mm_per_min: 600
      feed_mm_per_min: 50
      seek_scaler: 1.0
      feed_scaler: 1.0

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.39:low
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        direction_pin: gpio.12
        step_pin: gpio.14
    motor1:
      null_motor:

    
spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN
  
coolant:
  flood_pin: NO_PIN
  mist_pin:  NO_PIN

        
probe:
  pin: gpio.32:low:pu
  check_mode_start: false

pwm:
  pwm_hz: 35000
  output_pin: gpio.2
  enable_pin: gpio.15
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0% 2000=25% 10000=50% 20000=100%
  off_on_alarm: false
  
start:
  must_home: false
```

### Startup Messages

```text
$ss
[MSG:INFO: FluidNC v3.6.3]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config3ax.yaml]
[MSG:DBG: Running after-parse tasks]
[MSG:DBG: Checking configuration]
[MSG:INFO: Machine ESP32 Dev Controller V4]
[MSG:INFO: Board ESP32 Dev Controller V4]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:10us Dsbl Delay:0us Dir Delay:1us Idle Delay:0ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.13]
[MSG:INFO: Axis X (0.000,400.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.21 Dir:gpio.22 Disable:NO_PIN]
[MSG:INFO:  X Neg Limit gpio.35:low]
[MSG:DBG:  X Neg Limit 0]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (0.000,400.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.16 Dir:gpio.17:low Disable:NO_PIN]
[MSG:INFO:  Y All Limit gpio.34:low]
[MSG:DBG:  Y All Limit 0]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-219.000,1.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.14 Dir:gpio.12 Disable:NO_PIN]
[MSG:INFO:  Z Pos Limit gpio.39:low]
[MSG:DBG:  Z Pos Limit 0]
[MSG:INFO:   Motor1]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:gpio.15 Out:gpio.2 Dir:NO_PIN Freq:35000Hz Period:2047]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Probe Pin: gpio.32:low:pu]
[MSG:INFO: Connecting to STA SSID:DEST]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.89.25]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

[MSG:ERR: mount_to_vfs failed code 0xffffffff]
```

### User Interface Software

WebUI, Candle

### What happened?

Sd card not work.

### Other Information

I connected the card directly to the ESP32 and the card did not work.
[image]

MitchBradley, 2022-10-15:
That SD card adapter is designed for 5V signaling.  That can cause problems with the 3.3V signaling of ESP32.  It is better to use a direct-connect adapter without the 5/3 level conversion.

MitchBradley, 2022-10-15:
This is an example of a direct connect SD breakout: https://www.sparkfun.com/products/544 .

DEST1981, 2022-10-15:
This SD card adapter does not have level 5/3 conversion. The resistors on the board are pulled up to 3.3V. The board has only a linear voltage regulator AMS1117/

DEST1981, 2022-10-15:
>

DEST1981, 2022-10-15:
> This is an example of a direct connect SD breakout: https://www.sparkfun.com/products/544 .

Ok, I'll try removing the pull-up resistors.

closed 2022-10-15 by DEST1981

MitchBradley, 2022-10-15:
What capacity is the SD card and how is it formatted?  The library that we use does not support ExFAT

DEST1981, 2022-10-15:
Removing the pull-up resistors didn't change anything.
Power connected to 3.3V directly
2GB card is FAT32 formatted

MitchBradley, 2022-10-15:
If you have a scope or logic analyzer you can look at the signals to see if they are okay.  If not, the only thing I can think of is to check your wiring to make sure the dupont pins are connected to the right pins on the ESP32. Setups like this work routinely for us.

DEST1981, 2022-10-15:
I don't have a logic analyzer and my scope is analog.
I think that the problem may be in the card itself. Now I don't have another suitable one, only 64GB
I took another similar card. And here's what I got.
Format:
FAT32 Cluster 512 - [MSG:ERR: mount_to_vfs failed code 0xffffffff]
FAT32 Cluster 1024
Try Write File - CRUSH
E (43166) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x109
E (43167) diskio_sdmmc: sdmmc_write_blocks failed (265)
[MSG:INFO: Upload failed - file write failed]
Guru Meditation Error: Core 1 panic'ed (LoadProhibited). Exception was unhandled.

Core 1 register dump:
PC : 0x400f5bba PS : 0x00060b30 A0 : 0x800f7438 A1 : 0x3ffb2190
A2 : 0xff000000 A3 : 0x3ffb21ec A4 : 0x00000000 A5 : 0x3ffcca04
A6 : 0x00060b20 A7 : 0x00000001 A8 : 0xff000100 A9 : 0x3ffb2170
A10 : 0x3ffb220c A11 : 0x3ffb220c A12 : 0x00000000 A13 : 0x00000009
A14 : 0x00ff0000 A15 : 0xff000000 SAR : 0x00000008 EXCCAUSE: 0x0000001c
EXCVADDR: 0xff00014c LBEG : 0x4008c739 LEND : 0x4008c749 LCOUNT : 0xfffffffe

Backtrace:0x400f5bb7:0x3ffb21900x400f7435:0x3ffb21b0 0x400f9bad:0x3ffb2240 0x400f9edf:0x3ffb2270 0x400fa02a:0x3ffb22e0 0x401b4d2d:0x3ffb2300 0x401056be:0x3ffb2320 0x401095fe:0x3ffb2340 0x4010692a:0x3ffb2380 0x40107599:0x3ffb23c0 0x401082ee:0x3ffb24c0 0x40109861:0x3ffb25c0 0x400f65d1:0x3ffb2610 0x400fe842:0x3ffb2660 0x400ecd8b:0x3ffb2680 0x400eb8c7:0x3ffb26a0 0x400e0f55 :0x3ffb27e0 0x4010eabd:0x3ffb2820

But if I write a file from the computer to the card, then insert it into the board, the reading goes well. Files open and work.

MitchBradley, 2022-10-15:
One possibility is bad signal quality on the lines.

sotchrys, 2022-10-16:
Please Test it with firmware 3.4.4

DEST1981, 2022-10-16:
SD card problem. I replaced the card with 32GB and everything started working. the 2GB card worked fine with the computer.
In case of problems with the card, advise you to take another card and check.
[image]

MitchBradley, 2022-10-16:
I just remembered a factoid from a previous job where I had to write SD and MMC drivers.  2 GB is strange.  Instead of using the normal transfer sizes at the lowest level, for 2 GB they changed the fundamental block size and thus the interpretation of block numbers - or something like that.  I believe that let them get to 2 GB with FAT16.  It doesn't matter if you format in FAT16 or FAT32; the change is at a lower level.  They realized that was a stopgap measure that was not going to scale, so they went back to the normal block size with larger devices and switched to FAT32.  Very complete drivers like the ones in PCs know about this and cope with it.  I expect that the driver we use does not.