bdring/FluidNC #672: Problem: At high speeds, the tool travels a shorter distance than it should.

DEST1981, 2022-10-15:
### Controller Board

DIY board 4-axis. But setup to 3 axes
[image]

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

External driver TB6600, PWM spindel. Limit switches pull up to +3.3v directly connect.

### Configuration file

```yaml
name: "ESP32 Dev Controller V4"
board: "ESP32 Dev Controller V4"

stepping:
  engine: RMT
  idle_ms: 0
  dir_delay_us: 1
  pulse_us: 10
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.13:high
  
  x:
    steps_per_mm: 100
    max_rate_mm_per_min: 3500
    acceleration_mm_per_sec2: 200
    max_travel_mm: 400
    homing:
      cycle: 2
      positive_direction: false
    
    motor0:
      limit_neg_pin: gpio.35:low
      standard_stepper:
        direction_pin: gpio.22
        step_pin: gpio.21
    motor1:
      null_motor:

  y:
    steps_per_mm: 100
    max_rate_mm_per_min: 2500
    acceleration_mm_per_sec2: 200
    max_travel_mm: 400
    homing:
      cycle: 2
      positive_direction: false

    motor0:
      limit_all_pin: gpio.34:low
      standard_stepper:
        direction_pin: gpio.17:low
        step_pin: gpio.16
    motor1:
      null_motor:

  z:
    steps_per_mm: 100
    max_rate_mm_per_min: 3500
    acceleration_mm_per_sec2: 200
    max_travel_mm: 220
    homing:
      cycle: 1
      allow_single_axis: true
      positive_direction: true
      mpos_mm: 1.000
      seek_mm_per_min: 600
      feed_mm_per_min: 50
      seek_scaler: 1.0
      feed_scaler: 1.0

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.39:low
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        direction_pin: gpio.12
        step_pin: gpio.14
    motor1:
      null_motor:

    
spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN
  
coolant:
  flood_pin: NO_PIN
  mist_pin:  NO_PIN

        
probe:
  pin: gpio.32:low:pu
  check_mode_start: false

pwm:
  pwm_hz: 35000
  output_pin: gpio.2
  enable_pin: gpio.15
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0% 2000=25% 10000=50% 20000=100%
  off_on_alarm: false
  
start:
  must_home: false
```

### Startup Messages

```text
$ss
[MSG:INFO: FluidNC v3.6.3]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config3ax.yaml]
[MSG:DBG: Running after-parse tasks]
[MSG:DBG: Checking configuration]
[MSG:INFO: Machine ESP32 Dev Controller V4]
[MSG:INFO: Board ESP32 Dev Controller V4]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:10us Dsbl Delay:0us Dir Delay:1us Idle Delay:0ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.13]
[MSG:INFO: Axis X (0.000,400.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.21 Dir:gpio.22 Disable:NO_PIN]
[MSG:INFO:  X Neg Limit gpio.35:low]
[MSG:DBG:  X Neg Limit 0]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (0.000,400.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.16 Dir:gpio.17:low Disable:NO_PIN]
[MSG:INFO:  Y All Limit gpio.34:low]
[MSG:DBG:  Y All Limit 0]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-219.000,1.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.14 Dir:gpio.12 Disable:NO_PIN]
[MSG:INFO:  Z Pos Limit gpio.39:low]
[MSG:DBG:  Z Pos Limit 0]
[MSG:INFO:   Motor1]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:gpio.15 Out:gpio.2 Dir:NO_PIN Freq:35000Hz Period:2047]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Probe Pin: gpio.32:low:pu]
[MSG:INFO: Connecting to STA SSID:DEST]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.89.25]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
```

### User Interface Software

WebUI, Candle

### What happened?

I noticed in version 3.6x only:
steps_per_mm: 100
max_rate_mm_per_min: 2500
Works correctly.
If max_rate_mm_per_min is greater than 2500 when the tool travel speed is large (greater than 2500), steps are lost.
Example:
Version 3.6.3
pulse_us: 10
steps_per_mm: 100
max_rate_mm_per_min: 5000
I set the movement speed to 2000, move the tool by 100 mm. Everything is correct
I set the movement speed to 3000, move the tool by 100 mm. The tool moves 80mm.

On version 3.5.x everything worked correctly.
My motor drivers is TB6600

bdring, 2022-10-15:
Use something like this.

[image]

We have tested the SD features quite a lot over the past few weeks. The project developers cannot support DIY and breadboard quality hardware. We can leave this issue open to see if any community member want to help you.

DEST1981, 2022-10-15:
> Use something like this.
> 
> [image]
> 
> We have tested the SD features quite a lot over the past few weeks. The project developers cannot support DIY and breadboard quality hardware. We can leave this issue open to see if any community member want to help you.

My question is not about sd card

MitchBradley, 2022-10-15:
Try removing the pwm section from the config file to see if that influences the situation.

DEST1981, 2022-10-16:
I removed the PVM section.
The travel distance has become normal.
Movement speed is limited to 2500-2600 even if I specify 5000

DEST1981, 2022-10-16:
But when i set max_rate_mm_per_min: 6500
the speed of moving the tool drops to 500 and the distance traveled decreases by 3-4 times. This is independent of the movement speed specified by the commands.

DEST1981, 2022-10-16:
A very interesting experience.
I set max_rate_mm_per_min: 2500, loaded the config file, rebooted the ESP32, but nothing changed.
The tool moves a shorter distance at a slower speed

DEST1981, 2022-10-16:
Only flashing with full formatting helped.

MitchBradley, 2022-10-16:
It is possible that yesterday's PR #673 fixes it.  There was an uninitialized entry in the structure that configures the RMT engine.  Depending on the phase of the moon, sometimes the RMT engine would use the wrong clock source and the step pulse timing would be wildly wrong.  Other times, after a reboot, it might happen to initialize correctly and the timing would be correct.

MitchBradley, 2022-10-16:
Please try this https://github.com/bdring/FluidNC/releases/tag/v3.6.4-pre1

DEST1981, 2022-10-16:
> Please try this https://github.com/bdring/FluidNC/releases/tag/v3.6.4-pre1
I think this helped solve the problem!
I was able to overclock to 7000 without losing accuracy.

MitchBradley, 2022-10-16:
Can we close this ticket now?

DEST1981, 2022-10-16:
> Can we close this ticket now?

Yes

closed 2022-10-16 by MitchBradley

arlimbad222, 2022-10-22:
Hi @MitchBradley 

I am making a 1050mm travel machine and I am using a 3000rpm servo motor I tested pulse yesterday & I got the same problem whenever I send g0x100 I got a very accurate pulse but when I send g0x-800 it almost loose 100to140 pulse. servo motor has a pulse counter inbuilt so I can test the thing. 

name: "4-axis machine"
board: "4-axis machine"

stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 5
  pulse_us: 5
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  
  x:
    steps_per_mm: 400 
    max_rate_mm_per_min: 15000
    acceleration_mm_per_sec2: 300
    max_travel_mm: 317
    soft_limits: true
    homing:
      cycle: 2
      allow_single_axis: true
      mpos_mm: 0
      positive_direction: true
      feed_mm_per_min: 200
      seek_mm_per_min: 1500
      settle_ms: 250
      seek_scaler: 1.5
      feed_scaler: 1.5
    
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.16:low:pu
      hard_limits: false
      pulloff_mm: 2.000
      stepstick:
        direction_pin: gpio.25
        step_pin: gpio.18
    motor1:
      null_motor:

  y:
    steps_per_mm: 400
    max_rate_mm_per_min: 15000
    acceleration_mm_per_sec2: 300
    max_travel_mm: 250
    soft_limits: true
    homing:
      cycle: 2
      allow_single_axis: true
      mpos_mm: 0
      positive_direction: true
      feed_mm_per_min: 200
      seek_mm_per_min: 1500
      settle_ms: 250
      seek_scaler: 1.5
      feed_scaler: 1.5

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.17:low:pu
      hard_limits: false
      pulloff_mm: 2.000
      stepstick:
        direction_pin: gpio.15
        step_pin: gpio.26
    motor1:
      null_motor:

  z:
    steps_per_mm: 400
    max_rate_mm_per_min: 15000
    acceleration_mm_per_sec2: 300
    max_travel_mm: 140
    soft_limits: true
    homing:
      cycle: 1
      allow_single_axis: true
      mpos_mm: 0
      positive_direction: true
      feed_mm_per_min: 200
      seek_mm_per_min: 1500
      settle_ms: 250
      seek_scaler: 1.5
      feed_scaler: 1.5
      

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.4:low:pu
      hard_limits: false
      pulloff_mm: 2.000
      stepstick:
        direction_pin: gpio.33
        step_pin: gpio.27
    motor1:
      null_motor:

  a:
    steps_per_mm: 8.889
    max_rate_mm_per_min: 30000
    acceleration_mm_per_sec2: 20000
    max_travel_mm: 0

    motor0:
      limit_all_pin: NO_PIN
      stepstick:
        direction_pin: gpio.13
        step_pin: gpio.12
    motor1:
      null_motor:        

coolant:
  flood_pin: gpio.21
  mist_pin:  gpio.23

        
probe:
  pin: gpio.32:low:pu

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: gpio.5:low:pu
  cycle_start_pin: gpio.19:low:pu
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

PWM:
  pwm_hz: 5000
  output_pin: gpio.2
  enable_pin: gpio.22
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0% 18000=100%

$ss
[MSG:INFO: FluidNC v3.6.4-pre1]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:4-axis machine.yaml]
[MSG:INFO: Machine 4-axis machine]
[MSG:INFO: Board 4-axis machine]
[MSG:INFO: SPI not defined]
[MSG:INFO: No SD Card CS Pin]
[MSG:INFO: See http://wiki.fluidnc.com/en/config/sd_card#sdfallbackcs-access-sd-without-a-config-file]
[MSG:INFO: Stepping:RMT Pulse:3us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 4]
[MSG:INFO: Axis X (-317.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.18 Dir:gpio.25 Disable:NO_PIN]
[MSG:INFO:  X All Limit gpio.16:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (-250.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.26 Dir:gpio.15 Disable:NO_PIN]
[MSG:INFO:  Y All Limit gpio.17:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-140.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.27 Dir:gpio.33 Disable:NO_PIN]
[MSG:INFO:  Z All Limit gpio.4:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis A (-0.100,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.12 Dir:gpio.13 Disable:NO_PIN]
[MSG:INFO:   Motor1]
[MSG:INFO: feed_hold_pin gpio.5:low:pu]
[MSG:INFO: cycle_start_pin gpio.19:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:gpio.22 Out:gpio.2 Dir:NO_PIN Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Flood coolant gpio.21]
[MSG:INFO: Mist coolant gpio.23]
[MSG:INFO: Probe Pin: gpio.32:low:pu]

MitchBradley, 2022-10-22:
@arlimbad222 Start a new issue.  "Me too" problem reports with different configurations in the same ticket are hard for use to manage.