bdring/FluidNC #676: Problem: Dive into workpiece during long travel moves

matthewfallshaw, 2022-10-20:
### Controller Board

Root Controller

### Help From Board Vendor

- [X] Yes
- [ ] No
- [ ] Not Applicable

### Machine Description

Openbuilds / Maker Store [Lead CNC V3](https://makerhardware.net/wiki/doku.php?id=manuals:lead_v3_cnc) (high Z mod)

### Configuration file

```yaml
board: Root Controller ISO
name: Lead CNC 1010
stepping:
  engine: I2S_STREAM
  idle_ms: 255
  pulse_us: 4
#  dir_delay_us: 1
#  disable_delay_us: 3

axes:
  shared_stepper_disable_pin: NO_PIN
  shared_stepper_reset_pin: NO_PIN
  x:
    steps_per_mm: 200.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 150.000
    max_travel_mm: 730.000
    soft_limits: true
    homing:
      cycle: 2
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 50.000
      seek_mm_per_min: 1000.000
      settle_ms: 250
      seek_scaler: 1.100
      feed_scaler: 1.100
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: GPIO.26:low
      hard_limits: true
      pulloff_mm: 5.000
      standard_stepper:
        step_pin: I2SO.2
        direction_pin: I2SO.1:high
        disable_pin: I2SO.0:high

  y:
    steps_per_mm: 200.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 150.000
    max_travel_mm: 810.000
    soft_limits: true
    homing:
      cycle: 2
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 50.000
      seek_mm_per_min: 1000.000
      settle_ms: 250
      seek_scaler: 1.100
      feed_scaler: 1.100
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: GPIO.2:low
      hard_limits: true
      pulloff_mm: 5.000
      standard_stepper:
        step_pin: I2SO.6
        direction_pin: I2SO.5:low
        disable_pin: I2SO.4:high

    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: GPIO.32:low
      hard_limits: true
      pulloff_mm: 5.000
      standard_stepper:
        step_pin: I2SO.26
        direction_pin: I2SO.25:low
        disable_pin: I2SO.24:high

  z:
    steps_per_mm: 200.000
    max_rate_mm_per_min: 3500.000
    acceleration_mm_per_sec2: 150.000
    max_travel_mm: 210.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 50.000
      seek_mm_per_min: 500.000
      settle_ms: 250
      seek_scaler: 1.100
      feed_scaler: 1.100
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: GPIO.15:low
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 5.000
      standard_stepper:
        step_pin: I2SO.10
        direction_pin: I2SO.9:high
        disable_pin: I2SO.8:high

kinematics:
  Cartesian:

i2so:
  bck_pin: gpio.22
  data_pin: gpio.12
  ws_pin: gpio.21

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: gpio.36
  cycle_start_pin: gpio.39
  macro0_pin: gpio.34
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

coolant:
  flood_pin: i2so.12
  mist_pin: i2so.13
  delay_ms: 0

probe:
  pin: gpio.27
  check_mode_start: true

macros:
  startup_line0:
  startup_line1:
  macro0:
# G21&G54&G17&G94&G90
  macro1:
  macro2:
  macro3:

start:
  must_home: false
  deactivate_parking: false
  check_limits: true

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: i2so.15
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
planner_blocks: 16

# NoSpindle:
# pwm:
#   pwm_hz: 2000
#   direction_pin: i2so.16
#   output_pin: gpio.25
#   enable_pin: i2so.17
#   disable_with_s0: false
#   s0_with_disable: true
#   spinup_ms: 0
#   spindown_ms: 0
#   tool_num: 0
#   speed_map: 0=0.000% 3500=100.000%
# H2A:
#   uart:
#     txd_pin: gpio.17
#     rxd_pin: gpio.16
#     rts_pin: gpio.4
#     baud: 9600
#     mode: 8N1
#   modbus_id: 1
#   tool_num: 0
#   speed_map: 0=0% 0=15% 24000=100%

# Note that the relay spindle is not real — I'm currently manually controlling a Folinn/Fuling VFD
relay:
  direction_pin: NO_PIN
  output_pin: i2so.14
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0.000% 0=100.000% 1=100.000%
  off_on_alarm: false
Laser:
  pwm_hz: 5000
  output_pin: gpio.25
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  tool_num: 100
  speed_map: 0=0.000% 1000=100.000%
```

### Startup Messages

```text
$ss
<Idle|MPos:350.000,400.000,0.000|FS:0,0>
[MSG:INFO: FluidNC v3.6.3]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:DBG: Running after-parse tasks]
[MSG:DBG: Checking configuration]
[MSG:INFO: Machine Lead CNC 1010]
[MSG:INFO: Board Root Controller ISO]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.21 DATA:gpio.12]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: User Digital Output:0 on Pin:I2SO.15]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,730.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.2 Dir:I2SO.1 Disable:I2SO.0]
[MSG:INFO:  X All Limit gpio.26:low]
[MSG:DBG:  X All Limit 0]
[MSG:INFO: Axis Y (0.000,810.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.6 Dir:I2SO.5:low Disable:I2SO.4]
[MSG:INFO:  Y All Limit gpio.2:low]
[MSG:DBG:  Y All Limit 0]
[MSG:INFO:   Motor1]
[MSG:INFO:     standard_stepper Step:I2SO.26 Dir:I2SO.25:low Disable:I2SO.24]
[MSG:INFO:  Y2 All Limit gpio.32:low]
[MSG:DBG:  Y2 All Limit 0]
[MSG:INFO: Axis Z (-210.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.10 Dir:I2SO.9 Disable:I2SO.8]
[MSG:INFO:  Z Pos Limit gpio.15:low]
[MSG:DBG:  Z Pos Limit 0]
[MSG:INFO: feed_hold_pin gpio.36]
[MSG:INFO: cycle_start_pin gpio.39]
[MSG:INFO: macro0_pin gpio.34]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Relay Spindle Ena:NO_PIN Out:I2SO.14 Dir:NO_PIN]
[MSG:INFO: Laser Ena:NO_PIN Out:gpio.25 Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle Relay]
[MSG:INFO: Flood coolant I2SO.12]
[MSG:INFO: Mist coolant I2SO.13]
[MSG:INFO: Probe Pin: gpio.27]
[MSG:INFO: Connecting to STA SSID:blacknode]
[MSG:INFO: Connecting.]
[MSG:INFO: Connected - IP is 192.168.1.9]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
ok
```

### User Interface Software

gSender

### What happened?

Twice, during long `G0` travel moves, the machine has plunged into the workpiece.
Most recently at the following GCODE, from line 24881 of a longer file (full file available):
```gcode
X73.206 Y733.347 Z-6.009
X73.221 Y733.374 Z-5.914
X73.23 Y733.391 Z-5.815
X73.233 Y733.397 Z-5.715
G0 Z5
```

… and all of that was working perfectly, then:
```gcode
X69.401 Y-5.541
```

From the line above it's still in `G0` modal, and this is a long travel move up to the front end of the table. At approximately a Y value of 470, which is about a third of the way through the travel move, it plunged down to at least Z-30 before I hit the E-stop.

The GCODE immediately following this section, in case that helps:
```gcode
Z-3.5
G1 Z-4.5 F500
Z-6.5
G3 X73.508 Y-5.158 I1.67 J4.309
G1 X73.123 Y-5.024
X72.981 Y-4.971
X72.739 Y-4.869
X72.642 Y-4.832
X72.546 Y-4.806
X72.45 Y-4.79
X72.354 Y-4.784
X72.258 Y-4.787
X72.161 Y-4.798
```

The previous time this happened was also during a long `G0` travel move. It plunged directly downwards, far enough that the collet hit the workpiece, then it dragged through an L shaped move that could have been it executing the moves that should have come after the travel move.

### Other Information

I've run several short programs without problems.

Z axis is solid and not slipping. It's **driving** downwards, not slipping.
[image]
I can supply the gcode for this program, but a quick summary can be gleaned from this:
[image]

Unfortunately I've been unable to inspect machine state, since both times I've hit E-stop and that's turned off the controller.

I've described these issues on the Root CNC Discord. They're a very helpful group, but are out of ideas and suggested that I come here.

Any help is greatly appreciated.
In the meantime I'll try uploading future programs via WebUI to cut any comms with a sender out of the picture.

matthewfallshaw, 2022-10-20:
Just in case it's relevant, I often get garbled text restart or on first command after a restart.

See this example of connecting via fluidterm:
```
pi@leadcnc:~/source/fluidterm $ python3 fluidterm.py 

--- Available ports:
---  1: /dev/ttyAMA0         'ttyAMA0'
---  2: /dev/ttyUSB0         'USB Serial'
--- Enter port index or full name: 2
--- Fluidterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+] or Ctrl+Q | Menu: Ctrl+T | Help: Ctrl+T followed by 'H' ---
$SD/list pwd=adminj^@L^LH)xH^H^@N*^N
[MSG:DBG: Bad GCode: J^@L^LHXH^H^@N*^N$SD/LISTPWD=ADMIN]
error:2
```

… and this example of connecting via gSender:
```
gSender - [Grbl]
Connected to /dev/ttyUSB0 with a baud rate of 115200
[MSG:DBG: Bad GCode: ^B^BJ^L^Y^F^H@B^L^HN9LH^H^HJG^T]
error:1 (Expected command letter)
[MSG:DBG: Macro 0]
[MSG:ERR: Macro can only be used in idle state]
[MSG:INFO: Check door]
[MSG:DBG: Spin down only]
[MSG:INFO: Safety door closed.  Issue cycle start to resume]
```

Another example from gSender:
```
gSender - [Grbl]
Connected to /dev/ttyUSB0 with a baud rate of 115200
[MSG:DBG: Bad GCode: XJ^NZ]
error:2 (Bad number format)
[MSG:DBG: Macro 3]
ok
[MSG:DBG: Macro 1]
[MSG:DBG: Macro 1]
[MSG:DBG: Macro 3]
[MSG:DBG: Macro 3]
client $bye
[MSG:DBG: Bad GCode: NNXN9LH^H^L^HHJ8^N^N^N$BYE]
error:2 (Bad number format)
client $bye
[MSG:INFO: Restarting]
```
(That's me seeing some junk on connecting, issuing `$bye` and getting more junk, then issuing `$bye` again immediately and the command working.)

MitchBradley, 2022-10-20:
I2S_STREAM mode has caused some problems in the past.  I2S_STATIC seems to be more reliable.

MitchBradley, 2022-10-20:
How often does the problem occur?  Most of the time, or infrequently?   Problems that happen infrequently can be very difficult to debug.

matthewfallshaw, 2022-10-21:
> I2S_STREAM mode has caused some problems in the past. I2S_STATIC seems to be more reliable.

I'm happy to try switching to I2S_STATIC.
(I'm *very* happy to try, and this parens is chasing my curiosity, not arguing against trying — please feel very free to ignore it! Having read [the docs](http://wiki.fluidnc.com/en/config/axes#stepping) on the two systems, I would have *guessed* that if this was the problem the symptom would be jitter, not… aborting a `G0` travel on X & Y and activating the otherwise idle Z axis to plunge 35mm+ into the table. Iff you're inclined to satisfy my curiosity, can you gesture at what might be going on that could cause this to be relevant? (… and perhaps you don't think it might be relevant, and are only mentioning it as a helpful comment in passing; if that's so, then thank you!))

> How often does the problem occur? Most of the time, or infrequently? Problems that happen infrequently can be very difficult to debug.

It has happened exactly twice. Those are the exactly two times I've run long (>100 lines) programs on this controller.

Absent any further reply in the next few hours I'll try:
1. Switching to I2S_STATIC; and
2. Running the same program as caused the last crash from the SD card, with nothing else (gSender, fluidterm) connected.

MitchBradley, 2022-10-21:
The Root Controller looks like a nice piece of hardware but since I don't have one to play with, deep debugging is not easy.  I2S_STATIC seems to be less prone to triggering signal integrity problems on the shift register chain, but that is guesswork too.  We have some reports of problems with MKS Tinybee that are resolved when you switch to I2S_STATIC.  Signal integrity is my guess for that, but I don't have a Tinybee either so I can't prove it.  I am going batty right now with the wide range of different boards that all seem to have different strange glitches.

matthewfallshaw, 2022-10-21:
It's repeatable, though it didn't happen at exactly the same place this time.

Same program as the last dive, but this time running from WebUI / SD card with stepping/engine switched to I2S_STATIC. It dived into the table again, but this time a little before the big travel move that triggered it last time.
The [WebUI console output](https://docs.google.com/spreadsheets/d/1892th_MkhznURWOBLbuesjNFObXZzgH_y1hHnb7kaRM/edit?usp=sharing), with a visualisation. Full GCODE on the second tab of that sheet.

Other ideas:
The Root controller is driving DM556 stepper motor controllers. At least 10m after the crash I thought to check their temperature (I have them in an enclosure without any forced convection installed yet). The Z axis was at 46°C (115°F)… which seems hotter than I want it, but less than the 70°C spec.

Erm… I don't know what else to check.

matthewfallshaw, 2022-10-21:
Second plunge on the top, third on the bottom:
[image]

matthewfallshaw, 2022-10-23:
Fourth plunge just occurred, and my guess is that fluidnc is cleared of responsibility, but I'd be very grateful for your opinion, @MitchBradley (or anyone else).

This time I didn't hit the E-stop. I physically caught the Z-axis (by the dust shoe… "safely"), and I believe that it was unpowered and falling under its own weight. I then stopped the spindle via the VFD control pad, then hit the control>feed_hold_pin button (but I'm pretty sure the Z stepper was still unpowered). Perhaps 30s later, everything was still unmoving and the Z-axis stepper holding torque had returned (not sure when in that window). I checked the other steppers, and they were also powered and holding.

So, new theory (this is a bad theory, but it's the best I have):
Something is causing my stepper motors (all of them? at least my Z-axis) to lose power for something in the ballpark of 5 to 20s. The Root controller / fluidnc doesn't know that this has happened, so keeps sending move pulses, while the Z-axis is falling under its own weight. The steppers then power back up and resume motion, but the machine is in a very different physical state, so… see first attached image above where I got to the E-stop late and the collet was doing it's best to set my workpiece on fire.
Is this theory plausible? Are stepper pulses "dumb fired", and the fluidnc controller won't know if the stepper driver is even powered on?
The (seemingly, to me) most plausible candidate for what might cause the motors to shut down then resume is an overheat… and sure enough the X & (dual) Y steppers themselves were at 80-85°C… but the Z (the one that definitely lost power) was only reading 40°C (which is the main thing that causes me to label this as a bad theory, and perhaps 85°C isn't very hot for a stepper?) The DM556 motor drivers were all reading ~35°C, and the 36V, 350W Meanwell power supply powering the four stepper drivers was reading 40°C (from its metal outer casing, measured at an angle, with a laser temperature gun… so low confidence on the accuracy of that reading — it felt hotter, but nothing like the steppers).
(Have I underspecified that power supply? It's 350W, which is the same as the 24V power supply Maker Store (AU) sold me with the motors, but [the motor specs say](https://www.makerstore.com.au/product/elec-nema23-635-ht/) 3.6V × 3A (max current; 2.1A RMS), and `4 × 3A` sums to more than `350W / 36V = 9.7A`. I've limited the current on the DM556 drivers to 2.8A… but should that be `3A × 3.6V / 36V = 0.3A`, or something like that… that would seem crazy — surely those motors are running more that `3A × 3.6V = 10.8W`!?)

Things I'd changed since the last run:
1. I swapped the DM556 stepper drivers between my X & Z axes, so that if it was the driver that was responsible, the error would appear on my X axis instead of my Z.
2. I ran with my two enclosure doors open, to limit heat build up.
3. I used the Fusion 360 OpenbuildsGRBL post processor instead of the grblHAL post processor I used last time.

(I still ran from SD via WebUI, with no gSender or other sender connected.)

With this setup it successfully ran a 30 minute program that I'd manually broken into two pieces, than failed 85.43% of the way through a 56 minute program. (This evidence seems to me to support the overheat theory — I lasted longer before the crash 'cos whatever is overheating had more time to cool down, and better convection with the doors open.)

Possibly relevant machine spec details:

Enclosure 1:
Fuling / Folinn VFD
4 × DM556 stepper drivers, all set to a current limit of 2.8A
Power line noise filter

Enclosure 2:
Root controller
36V 350W Meanwell power supply powering the stepper drivers
24V 350W Meanwell power supply powering the Root controller

[Stepper motors](https://www.makerstore.com.au/product/elec-nema23-635-ht/)

matthewfallshaw, 2022-10-23:
Closing this issue now, since my best theory says that it's not a fluidnc problem.
(I'll re-open if I run out of electro-mechanical leads to follow.)

But if anyone has any suggestions, I'd be very grateful to hear them!
(And thank you for your earlier suggestions, @MitchBradley )

closed 2022-10-23 by matthewfallshaw

matthewfallshaw, 2022-10-24:
Thank you for the advice, @MitchBradley. I like your power supply theory (which @readeral independently suggested last night).

> How do you keep your Z axis from falling when the system is off?

A stick. 😁
I have a braked stepper motor (still in its box, waiting to get to the front of my todo list), and needing more outputs to control it was one of the reasons I switched from my old xPro to the Root controller. The [startup_line0](http://wiki.fluidnc.com/en/config/macros) in fluidnc's config looks like an ideal place to put the `M64 P[whatever]` to disable the brake, but I note that the wiki sounds a little disparaging of this config item ("These are legacy features from Grbl, which called them $N0 and $N1"). Am I misreading the disparaging tone?

> Did you check the green power LED… I can understand if you didn't; it's hard to think when you are trying to prevent disaster.

Exactly. I'll try to rustle up a few old webcams or smart phones and try to get a view of everything simultaneously when I next run the machine.

> I suspect that the root controller itself needs less than an amp, so the 350W supply is serious overkill.

Agreed. It's still overkill (it's what I had lying around), but I'm running air solenoids (mist and laser air assist), and my laser wants 12V but I was going to just use a buck converter to power it from the same supply.

Thanks again for your detailed advice, and your work both helping me directly and everywhere else you're making fluidnc as awesome as it is.

MitchBradley, 2022-10-24:
Misread; there's nothing horribly problematic about $N0 and $N1, as far as they go.  There can be concerns about exactly when they should be executed, considering all of the ways the system might be reset.

matthewfallshaw, 2022-10-28:
Update: Thank you, @MitchBradley, your power supply theory seems confirmed.
No new craters or canyons in my workpieces since I stopped pulling much more current than I needed from my power supply and overloading it. FluidNC and my Root controller have been rock solid.

I'm still frequently (~20%) getting junk on powering up ([examples above](#issuecomment-1286230626), but after one (or two) `$bye` attempts it settles down.
WiFi is also a chance to drop out within a few minutes of starting up, but if it survives that long, then it seems rock solid until the next power cycle.
Neither of those two glitches is blocking me, and I'm loving working with FluidNC.

[No reply necessary — the two remaining glitches seem more likely to be hardware related than firmware, but if you happen to have any insights, they're welcome. (I've signed up as a Github sponsor.)]

MitchBradley, 2022-10-28:
There were some reports of power-up junk on MKS Tinybee boards that I was not able to solve since I did not have hardware to test.  WiFi dropouts are tricky.  Some ESP32 modules have insufficient capacitance on the 3.3V rail, but the typical symptom is the system resets continuously; when the WiFi turns on it drags down the power and the brownout detector resets.

At one point there was a problem with some mesh WiFi routers, in which the various routers would try to search for new ones on the mesh and end up spamming the ESP32 with file requests.  I think we managed to solve that one by filtering out spurious URLs.

WiFi problems can be really hard to solve since they are so dependent on environmental factors that are different in every setup.