bdring/FluidNC #677: Problem: SD card not Initialized on version 3.6.3

mstfyghm, 2022-10-20:
### Controller Board

I made it myself using ESP32-WROOM32. I used an old 256MB mmc.

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

My machin is a CO2 laser cnc that has DM542E stepper drivers.

### Configuration file

```yaml
stepping:
  engine: RMT
  idle_ms: 255
  dir_delay_us: 1
  pulse_us: 3
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.27:low
  
  x:
    steps_per_mm: 125.98425
    max_rate_mm_per_min: 60000
    acceleration_mm_per_sec2: 4500
    max_travel_mm: 900
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 895
      feed_mm_per_min: 300
      seek_mm_per_min: 4000
      settle_ms: 200
      seek_scaler: 1.05
      feed_scaler: 1.1
    
    motor0:
      standard_stepper:
        step_pin: gpio.25:low
        direction_pin: gpio.32:low
        disable_pin: NO_PIN
      limit_pos_pin: gpio.35:high
      pulloff_mm: 5.000

  y:
    steps_per_mm: 125.98425
    max_rate_mm_per_min: 50000
    acceleration_mm_per_sec2: 1100
    max_travel_mm: 600
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 595
      feed_mm_per_min: 300
      seek_mm_per_min: 3000
      settle_ms: 250
      seek_scaler: 1.1
      feed_scaler: 1.1
    
    motor0:
      standard_stepper:
        step_pin: gpio.33:low
        direction_pin: gpio.14:low
        disable_pin: NO_PIN
      limit_pos_pin: gpio.34:high
      pulloff_mm: 5.000

coolant:
  flood_pin: gpio.22:low

Laser:
  pwm_hz: 5000
  output_pin: gpio.2:low
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  tool_num: 0
  speed_map: 0=0.000% 1000=100.000%

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

  
start:
  must_home: true
  check_limits: true
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.6.3]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine None]
[MSG:INFO: Board None]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:3us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.27:low]
[MSG:INFO: Axis X (-5.000,895.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.25:low Dir:gpio.32:low Disable:NO_PIN]
[MSG:INFO:  X Pos Limit gpio.35]
[MSG:INFO: Axis Y (-5.000,595.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.33:low Dir:gpio.14:low Disable:NO_PIN]
[MSG:INFO:  Y Pos Limit gpio.34]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Laser Ena:NO_PIN Out:gpio.2:low Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: Flood coolant gpio.22:low]
[MSG:INFO: Connecting to STA SSID:XXXXX]
[MSG:INFO: Connecting.]
[MSG:INFO: Connected - IP is 192.168.1.XXX]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://XXXXXXX.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
ok
```

### User Interface Software

WebUI

### What happened?

My mmc is connected to ESP32 physically but webUI shows that there is no sdcard connected. When I press the "Refresh" button on WebUI, following message appears:
`[MSG:ERR: sdmmc_card_init failed code 0x106]`
As I mentioned in hardware description, I used an old 256MB mmc and everything was OK before v3.6.3.

bdring, 2022-10-20:
Show photos and a schematic of your hardware.

mstfyghm, 2022-10-20:
> Show photos and a schematic of your hardware.

Thanks for your attention. My hardware is based on "D1 mini ESP32". I have used pins 34 & 35 as X and Y limit switch inputs  and obviously they need external pull-up resistors. The limit switches are configured as N.O. and drive PC817 optocoupler LEDs. MMC is directly connected to the ESP32 according to config.yaml content:
[image]
[image]

bdring, 2022-10-20:
Our tests show the SD code works fine. The primary developers of FluidNC do not have time to help with DIY hardware. Maybe others in the community can help.

sjonholle, 2022-10-20:
worth a try: add pullup resistors (if you haven't) 
[image]

Emmpunkt, 2022-10-20:
@mstfyghm i have the same Problem, you can go up to Version 3.6.1.
This works for my hardware....

MitchBradley, 2022-10-20:
It's a software problem.  The MMC and SD protocols are somewhat different.  Earlier versions of FluidNC used the Arduino Framework SD driver, which supports both the MMC and SD protocol variants.  We switched to the ESP-IDF driver, which only works with SD.  MMC cards are pretty rare these days.

MitchBradley, 2022-10-20:
The ESP-IDF driver only supports MMC cards in the native multi-data-line mode, not in SPI mode.  Perhaps you could switch to an SD card using one of these shields
[image]
They plug right on top of a D1 mini.  Both of my production shop FluidNC controllers use D1 Mini with a shield like that.

MitchBradley, 2022-10-20:
This is the stack that powers my production router.  Bottom is a custom "Gecko Blaster" board that brings the ESP32 pins over to a parallel port and has a USB-B connector to bring in 5V.  Next is a D1 Mini, then the SD shield, and finally an OLED shield.
[image]

MitchBradley, 2022-10-20:
The shield uses the "standard" SPI GPIOs 18,19,23 cs 5, same as your config, so in principle it should be  plug and play.

mstfyghm, 2022-10-21:
Thanks @MitchBradley .  and what is the max size for sd card?

bdring, 2022-10-22:
See this wiki page regarding the SD card setup and limitations.

http://wiki.fluidnc.com/en/config/sd_card#card-formatting

sotchrys, 2022-10-27:
on which pins is the connection to the esp?

sotchrys, 2022-11-08:
> The shield uses the "standard" SPI GPIOs 18,19,23 cs 5, same as your config, so in principle it should be plug and play.

The card you see in the photo has 16 contacts,
apart from the 3 contacts of the power supply ( 5v 3.3v GND ), what is the connection to the other 4 contacts (18, 19, 23, 5 ) I searched but did not find the pinouts of the card. Thanks

MitchBradley, 2022-11-08:
That shield stacks on top of a D1 Mini board.  The reason I recommended it is because the OP showed a picture of his setup which has a D1 Mini ESP32 with jumper wires soldered onto the contacts of an MMC card.  Since he already has a D1 Mini, that shield would be completely plug and play in his system, using the standard SPI and SD CS pins.

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5

The pin number labels on the shield - like D1 and D2 - do not correspond to ESP32 GPIO numbers.  The reason is because the D1 Mini started out as an ESP8266 thing and the pin numbers are for the ESP8266 world.  Then someone shoehorned an ESP32 into that form factor, sort of.  The CPU module is wider to accommodate two extra rows of 8 pins, but the inner rows of pins are the same distance apart as for the original ESP8266 D1 Mini, so the shields still work, although they cannot access the extra ESP32 pins.  To find the correspondence between the shield pin numbers and the ESP32 GPIOs, you just have to look at the layouts and match them up.  

Bottom side of ESP32 Mini:
[image]

Top side of original ESP8266 D1 Mini

[image]

Note that the ESP32 Mini designers arranged the pinout so that the pins commonly used for UART, SPI and I2C line up on both versions, despite the fact that their GPIO numbers are different.

I found these by searching for "D1 Mini pinout" and "ESP32 Mini pinout"

If you are not using a Mini, there is nothing special about this shield.  Any breakout without extra circuitry would work the same.

sotchrys, 2022-11-08:
Does this card work?
https://nettigo.eu/products/module-with-microsd-card-reader-shield-for-wemos-d1-mini

MitchBradley, 2022-11-08:
That is the same product as what I have been talking about.  I am closing this ticket now because the OP has been silent for some time and there is no more to say.  @sotchrys - if you need help with SD, make a ticket with the relevant info about your machine.  Asking questions on this ticket without the context about your machine is distracting and hard for us to answer.

closed 2022-11-08 by MitchBradley