bdring/FluidNC #679: Problem: Jog it works incorrectly

DEST1981, 2022-10-22:
### Controller Board

DIY board 4-axis. But setup to 3 axes
[image]

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

External driver TB6600, PWM spindel. Limit switches pull up to +3.3v directly connect.

### Configuration file

```yaml
name: "ESP32 Dev Controller V4"
board: "ESP32 Dev Controller V4"

stepping:
  engine: RMT
  idle_ms: 0
  dir_delay_us: 1
  pulse_us: 10
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.13:high
  
  x:
    steps_per_mm: 100
    max_rate_mm_per_min: 7000
    acceleration_mm_per_sec2: 200
    max_travel_mm: 400
    homing:
      cycle: 2
      positive_direction: false
    
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.35:low
      limit_all_pin: NO_PIN
      standard_stepper:
        direction_pin: gpio.22
        step_pin: gpio.21
    motor1:
      null_motor:

  y:
    steps_per_mm: 100
    max_rate_mm_per_min: 7000
    acceleration_mm_per_sec2: 200
    max_travel_mm: 400
    homing:
      cycle: 2
      positive_direction: false

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.34:low
      limit_all_pin: NO_PIN
      standard_stepper:
        direction_pin: gpio.17:low
        step_pin: gpio.16
    motor1:
      null_motor:

  z:
    steps_per_mm: 100
    max_rate_mm_per_min: 1000
    acceleration_mm_per_sec2: 200
    max_travel_mm: 220
    homing:
      cycle: 1
      allow_single_axis: true
      positive_direction: true
      mpos_mm: 1.000
      seek_mm_per_min: 600
      feed_mm_per_min: 50
      seek_scaler: 1.0
      feed_scaler: 1.0

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.39:low
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        direction_pin: gpio.12
        step_pin: gpio.14
    motor1:
      null_motor:

    
spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN
  
coolant:
  flood_pin: NO_PIN
  mist_pin:  NO_PIN

        
probe:
  pin: gpio.32:low:pu
  check_mode_start: false

pwm:
  pwm_hz: 35000
  output_pin: gpio.2
  enable_pin: gpio.15
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0% 2000=25% 10000=50% 20000=100%
  
start:
  must_home: false
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.6.4-pre1]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config3ax.yaml]
[MSG:INFO: Machine ESP32 Dev Controller V4]
[MSG:INFO: Board ESP32 Dev Controller V4]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:10us Dsbl Delay:0us Dir Delay:1us Idle Delay:0ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.13]
[MSG:INFO: Axis X (0.000,400.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.21 Dir:gpio.22 Disable:NO_PIN]
[MSG:INFO:  X Pos Limit gpio.35:low]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (0.000,400.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.16 Dir:gpio.17:low Disable:NO_PIN]
[MSG:INFO:  Y Pos Limit gpio.34:low]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-219.000,1.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.14 Dir:gpio.12 Disable:NO_PIN]
[MSG:INFO:  Z Pos Limit gpio.39:low]
[MSG:INFO:   Motor1]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:gpio.15 Out:gpio.2 Dir:NO_PIN Freq:35000Hz Period:2047]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Probe Pin: gpio.32:low:pu]
[MSG:INFO: Connecting to STA SSID:DEST]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.89.25]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
```

### User Interface Software

WebUI, Candle

### What happened?

After starting, I use the jog wheel to position the instrument.
I use Candle and WebUi at the same time.
I press the move tool button.
$J=G91 G21 F1000 Z-0.1
The tool moves very slowly at about 50 mm/min and does not stop. I can only stop using a Firmware Reset or a Candle Reset.
log:
$J=G91 G21 F1000 Z-0.1
<Idle|MPos:0.000,0.000,-1.130|FS:0,0>
ok
<Jog|MPos:0.000,0.000,-1.150|FS:nan,0|Ov:200,100,100>
<Jog|MPos:0.000,0.000,-2.090|FS:nan,0|Ov:200,100,100>
<Jog|MPos:0.000,0.000,-3.230|FS:nan,0>
<Jog|MPos:0.000,0.000,-4.390|FS:nan,0>
<Jog|MPos:0.000,0.000,-5.530|FS:nan,0|WCO:3.210,1.250,0.000>
<Jog|MPos:0.000,0.000,-6.670|FS:nan,0>
<Jog|MPos:0.000,0.000,-7.820|FS:nan,0>
<Jog|MPos:0.000,0.000,-8.970|FS:nan,0>

When I select home on the Z-axis, the machine works fine.

The problem does not appear when I first use WebUi only. I move the tool along all axes anywhere. And after that I launch Candle. Then it works fine.

bdring, 2022-10-22:
Does the problem occur if you only use Candle and never load the WebUI?

bdring, 2022-10-22:
The "nan" in `<Jog|MPos:0.000,0.000,-1.150|FS:nan,0|Ov:200,100,100>` is weird. I'll look into that.

bdring, 2022-10-22:
I tried using Candle. It worked once, but after closing it, it would not start again until a reboot.

DEST1981, 2022-10-22:
> I tried using Candle. It worked once, but after closing it, it would not start again until a reboot.

Yes, the next time the problem appears after a reboot.

DEST1981, 2022-10-22:
> Does the problem occur if you only use Candle and never load the WebUI?

There is also a problem without WebUi.  I rarely use WebUi, almost never.

DEST1981, 2022-10-22:
> I tried using Candle. It worked once, but after closing it, it would not start again until a reboot.

Often the problem occurs after a full reset.

bdring, 2022-10-22:
Full PC reboot

DEST1981, 2022-10-22:
> Full PC reboot

This won't fix the problem

DEST1981, 2022-11-04:
Close the ticket. In version 3.6.4, I could not reproduce the problem.

closed 2022-11-04 by MitchBradley