bdring/FluidNC #680: RS 485 issue

lugeral, 2022-10-22:
### Controller Board

6-pack

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

Gantry router with tb6600's
Recommended Huanyang vfd
2.2 Kw water cooled spindle
no end stops connected yet

### Configuration file

```yaml
board: 6 Pack
name: FluidNC
meta: 2022-09-05 Andrew Luger 315469
stepping:
  engine: I2S_STREAM
  idle_ms: 250
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 160.000
    max_rate_mm_per_min: 4000.000
    acceleration_mm_per_sec2: 750.000
    max_travel_mm: 1220.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.33
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0

  y:
    steps_per_mm: 160.000
    max_rate_mm_per_min: 4000.000
    acceleration_mm_per_sec2: 750.000
    max_travel_mm: 640.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.32
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.5
        direction_pin: I2SO.4:low
        disable_pin: I2SO.7
    
    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.10
        direction_pin: I2SO.9:low
        disable_pin: I2SO.8

  z:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 500.000
    max_travel_mm: 105.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.35
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.13
        direction_pin: I2SO.12:low
        disable_pin: I2SO.15
        

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
  
probe:
  pin: gpio.34
  check_mode_start: true

start:
  must_home: false
  
coolant:
  flood_pin: gpio.15
  mist_pin: gpio.12
  delay_ms: 0

Huanyang:
  uart:
    txd_pin: gpio.26
    rxd_pin: gpio.16
    rts_pin: gpio.4
    baud: 19200
    mode: 8N1
  modbus_id: 1
  tool_num: 0
  speed_map: 0=0% 0=25% 6000=25% 24000=100%
  off_on_alarm: true
```

### Startup Messages

```text
FluidNC v1.2.0 using COM4
Exit: Ctrl-C, Ctrl-Q or Ctrl-], Clear screen: CTRL-W
Upload: Ctrl-U, Reset ESP32: Ctrl-R, Send Override: Ctrl-O
[MSG:ERR: Huanyang spindle did not reach device units 1073423424. Reported value is 0]

Grbl 1.1h [FluidNC v3.6.3 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
[MSG:INFO: VFD Queue Full]
[MSG:ERR: Huanyang spindle did not reach device units 1073423424. Reported value is 0]
ALARM:10
[MSG:INFO: VFD Queue Full]
[MSG:ERR: Huanyang spindle did not reach device units 1073423424. Reported value is 0]
[MSG:INFO: VFD Queue Full]
[MSG:ERR: Huanyang spindle did not reach device units 1073423424. Reported value is 0]

Grbl 1.1h [FluidNC v3.6.3 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
[MSG:INFO: VFD Queue Full]
[MSG:ERR: Huanyang spindle did not reach device units 1073423424. Reported value is 0]
ALARM:10
[MSG:INFO: VFD Queue Full]
[MSG:ERR: Huanyang spindle did not reach device units 1073423424. Reported value is 0]
[MSG:INFO: VFD Queue Full]
[MSG:ERR: Huanyang spindle did not reach device units 1073423424. Reported value is 0]

Grbl 1.1h [FluidNC v3.6.3 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
[MSG:INFO: VFD Queue Full]
[MSG:ERR: Huanyang spindle did not reach device units 1073423424. Reported value is 0]
ALARM:10
[MSG:INFO: VFD Queue Full]
[MSG:ERR: Huanyang spindle did not reach device units 1073423424. Reported value is 0]
[MSG:INFO: VFD Queue Full]
Resetting MCU

[MSG:INFO: FluidNC v3.6.3]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine PrintNC]
[MSG:INFO: Board 6 Pack]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,1220.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.2 Dir:I2SO.1 Disable:I2SO.0]
[MSG:INFO:  X Neg Limit gpio.33]
[MSG:INFO: Axis Y (-640.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.5 Dir:I2SO.4:low Disable:I2SO.7]
[MSG:INFO:  Y Neg Limit gpio.32]
[MSG:INFO:   Motor1]
[MSG:INFO:     standard_stepper Step:I2SO.10 Dir:I2SO.9:low Disable:I2SO.8]
[MSG:INFO: Axis Z (-105.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.13 Dir:I2SO.12:low Disable:I2SO.15]
[MSG:INFO:  Z Pos Limit gpio.35]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Huanyang Spindle Uart Tx:gpio.26 Rx:gpio.16 RTS:gpio.4 Baud:19200]
[MSG:INFO: Using spindle Huanyang]
[MSG:INFO: Flood coolant gpio.15]
[MSG:INFO: Mist coolant gpio.12]
[MSG:INFO: Probe Pin: gpio.34]
[MSG:INFO: Connecting to STA SSID:LUGER]
[MSG:INFO: Connecting.]
[MSG:INFO: Connected - IP is 192.168.50.100]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 1.1h [FluidNC v3.6.3 (wifi) '$' for help]
[MSG:INFO: VFD RS485 Unresponsive
```

### What happened?

I cannot start the spindle

Connection with Rs 485 seems to be ok,  however communication is garbled. I have tried this at 9600 and 19200  and i get the exact same results. 

VFD sends back the the stuff at the top. as well as the queue full errors so i'm not sure what is going on

Rs 485 is installed on module 3

thanks

bdring, 2022-10-22:
You did not include the beginning of the start messages.

The start with something similar to...

```
[MSG:INFO: FluidNC v3.6.4-pre1]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
```

bdring, 2022-10-22:
I assume you have setup the VFD with proper register values. Please confirm.

<http://wiki.fluidnc.com/en/config/config_spindles#vfd-rs485>

I would suggest using 9600 baud. That is what we test at.

Send a picture of your wiring at the VFD and Module side.

lugeral, 2022-10-22:
Mr. Dring

Enclosed

[MSG:INFO: FluidNC v3.6.3]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine PrintNC]
[MSG:INFO: Board 6 Pack]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle
Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,1220.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.2 Dir:I2SO.1 Disable:I2SO.0]
[MSG:INFO:  X Neg Limit gpio.33]
[MSG:INFO: Axis Y (-640.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.5 Dir:I2SO.4:low Disable:I2SO.7]
[MSG:INFO:  Y Neg Limit gpio.32]
[MSG:INFO:   Motor1]
[MSG:INFO:     standard_stepper Step:I2SO.10 Dir:I2SO.9:low Disable:I2SO.8]
[MSG:INFO: Axis Z (-105.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.13 Dir:I2SO.12:low
Disable:I2SO.15]
[MSG:INFO:  Z Pos Limit gpio.35]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Huanyang Spindle Uart Tx:gpio.26 Rx:gpio.16 RTS:gpio.4
Baud:19200]
[MSG:INFO: Using spindle Huanyang]
[MSG:INFO: Flood coolant gpio.15]
[MSG:INFO: Mist coolant gpio.12]
[MSG:INFO: Probe Pin: gpio.34]
[MSG:INFO: Connecting to STA SSID:LUGER]
[MSG:INFO: Connecting.]
[MSG:INFO: Connected - IP is 192.168.50.100]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 1.1h [FluidNC v3.6.3 (wifi) '$' for help]
[MSG:INFO: VFD RS485 Unresponsive]

Andy

bdring, 2022-10-22:
I suspect a problem with the wiring or the VFD parameters. The config looks OK.

lugeral, 2022-10-22:
Hi Mr Dring

Yes, I have set the register values exactly as per the web page.
The 19200 baud rate was returned to 9600 after I completed my trials.
Enclosed are the pictures you requested.

The two wire connection from the vfd to the 6pack was temporarily
disconnected so i could show that i was indeed in socket i/o 3 in my
pictures

Andy

4 pictures

2 here
2 more will follow

lugeral, 2022-10-22:
picture 3 and 4

bdring, 2022-10-22:
Drag and drop the files into the editor

lugeral, 2022-10-22:
Pictures as requested

[image]
[image]

lugeral, 2022-10-22:
[image]
[image]

bdring, 2022-10-22:
No connection?

[image]

bdring, 2022-10-22:
This is my connection

[image]

lugeral, 2022-10-22:
My apologies I'm not very good at this stuff

I assure you that I reset the registers to their default values and rekeyed them as per your page many times over.

I checked the wiring many times as well , I have checked the continuity of the cables prior to connecting them , they are good.

Is it possible that I have the wrong gauge of wire??

I noticed your connection between the rs485 and the RS+/- , your wires seem lager than the ones i have used which twisted pair  from a cat 5e cable

lugeral, 2022-10-22:
I also notice that you have a version 3 module , I have version 2

bdring, 2022-10-22:
- Is it connected. Your picture shows there is no terminal block attached.
 - Is A going to RS- and B going to RS+
 - Is the ground wire disconnected?

lugeral, 2022-10-22:
Like yours I do not have a ground wire as per the web page.
I have A to RS+ and B to RS-

bdring, 2022-10-22:
Version 3 only added LEDs to the Rx and Tx and converted the source to OSHW Labs

bdring, 2022-10-22:
Try swapping A and B.

The wire gauge should be OK. Are the wires twisted?

BTW: When I ask 3 questions, please give 3 answers.

lugeral, 2022-10-22:
I swapped A and B........ no change
ok
ok

lugeral, 2022-10-22:
yes twisted

bdring, 2022-10-22:
Is it coneected? Your picture says no.

lugeral, 2022-10-22:
Yes it is connected
it was only temporarily disconnected to show you that the RS 485 was in the correct socket

bdring, 2022-10-22:
It looks like there are 3 wires connected. Orange, Blue and White

It is hard to tell what is twisted with what.

[image]

bdring, 2022-10-22:
Actually maybe 4

lugeral, 2022-10-22:
I had a twisted pair of wires going to RS+ and RS-
I have since fixed that to reflect what you have in your picture, that is a single wire going to each connector. I used the blue and white and got rid of orange and white.

bdring, 2022-10-22:
OK, how about some new pictures. This time exactly how you have it connected.

The RS485 Haunyang setup has been extensively tested as recent as last week. We had a a nearly 50 post issue that turned out to be user error.

https://github.com/bdring/FluidNC/issues/568

lugeral, 2022-10-22:
Ok

lugeral, 2022-10-22:
[image]
[image]
[image]

bdring, 2022-10-22:
The wiring looks OK assuming there are no breaks.

I ran your complete config and it works fine on my setup. I even tried 19200 baud and it worked fine.

I still think it is a setup issue. 

You should recheck the things below. If you think there is a problem with the hardware, I would be happy to test it on my Huanyang. Other than that, I can't help anymore via this issue.

- Double check that both ends use the same baud rate. You changed them a lot, so be sure they agree. 
- What does the display of the VFD show after power on? Mine flashes 00.000 with an "R" like character first. The POWER and HZ LEDs are on. The FOR LED blinks.
- With all power off, check the continuity of the wiring using a meter for the controller terminal block to the VFD terminal block.
- Does the VFD spin on manual (Operator) mode?
- Try new wiring. Maybe temporarily move the VFD close for a test.

lugeral, 2022-10-22:
Thanks Bart appreciate the effort.

- I just finished rechecking the registers everything is exactly per your website
- mine flashes R00.00, the power light is on, the FOR blinks
- item 3 , i have checked continuity of all the wiring from one end to the other for both the input power and power going to the spindle. I also have continuity between your equipment and the vfd.
- the spindle runs fine in operator mode, I have speed control via the knob.
- the vfd is located **outside the cabinet** , i'll try to relocate it further away from your stuff

Bart 
your controller is mounted about 3 inches away from its power supply, is that too close????

Anyway i'll try moving things around, I suppose i could my 0-10 volt spindle control or order another rs 485 .

Andy

lugeral, 2022-11-12:
Afternoon Bart

Just following up to close the issue I was having. Everything I had wiring wise and programmatically was correct.
What I did not realize is that I had damaged the top board of the vfd. I saw that when I tearing down the vfd to start over.

Anyway, long and the short of it is that a receive a new VFD this week installed it today and all is well.

Please accept my apologies for wasting your time!

Andy

closed 2022-11-12 by lugeral