bdring/FluidNC #681: Hard limit during homing

daxliniere, 2022-10-22:
### Controller Board

MKS DLC32

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

SWOLE-CNC mill (modified CNC3040Z)

### Configuration file

```yaml
board: MKS-DLC32
name: SWOLE-CNC
meta: V2.6.4 (2202.10.16) Dax Liniere, originally by Skorpi

kinematics:
  Cartesian:

stepping:
  engine: I2S_STATIC
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
  
axes:
  shared_stepper_disable_pin: I2SO.0
  x:
    steps_per_mm: 804.065
#was 802.341
    max_rate_mm_per_min: 1600.000
#was 1500.000
    acceleration_mm_per_sec2: 80.000
    max_travel_mm: 302.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 20.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.36
      limit_pos_pin: gpio.34
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2:low
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
        reset_pin: NO_PIN

  y:
    steps_per_mm: 804.636
#was 802.322
    max_rate_mm_per_min: 1600.000
    acceleration_mm_per_sec2: 80.000
# was 300.000
    max_travel_mm: 186.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 20.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35
      limit_pos_pin: gpio.23
#this pin doesn't have the same input circuitry as the 3 dedicated on-board limit switch inputs
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      stepstick:
        step_pin: I2SO.5
        direction_pin: I2SO.6:low
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
        reset_pin: NO_PIN
  z:
    steps_per_mm: 802.497
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 50.000
# was 500.000
    max_travel_mm: 100
#46.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 20.000
      seek_mm_per_min: 100.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_pos_pin: gpio.18
#this pin doesn't have the same input circuitry as the 3 dedicated on-board limit switch inputs
      hard_limits: true
      pulloff_mm: 1.000
      stepstick:
        step_pin: I2SO.3
        direction_pin: I2SO.4:low
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
        reset_pin: NO_PIN

i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14

sdcard:
  cs_pin: gpio.15
  card_detect_pin: NO_PIN
# This could be GPIO.39, but Card Detect has no supported functions in FluidNC

control:
  safety_door_pin: gpio.33:low:pu
  cycle_start_pin: NO_PIN
  feed_hold_pin: NO_PIN
  reset_pin: NO_PIN
  macro0_pin: gpio.19:low:pu
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

parking:
  enable: true
  axis: Z
  pullout_distance_mm: 5.000
  pullout_rate_mm_per_min: 250.000
  target_mpos_mm: 0.000
  rate_mm_per_min: 800.000
  
macros:
  startup_line0:
  startup_line1:
  macro0: $X&$H
#unlock and home XYZ
  macro1: G38.2 G91 Z-46 F100&G0 Z1&G38.2 G91 Z-2 F50&G10 L20 P1 Z4.985&G10 L20 P2 Z4.985&G10 L20 P3 Z4.985&G10 L20 P4 Z4.985&G10 L20 P5 Z4.985&G10 L20 P6 Z4.985&G53 G0 Z0
#z-probe
  macro2:
  macro3: $SD/Run=lasertest.gcode

coolant:
  flood_pin: gpio.0
# continuous air
  mist_pin:  gpio.4
# pulsed air 
  delay_ms: 0

probe:
  pin: gpio.22:low
  check_mode_start: true

10V:
# Spindle
  direction_pin: NO_PIN
  forward_pin: NO_PIN
  reverse_pin: NO_PIN
  output_pin: gpio.32
  enable_pin: gpio.5:low
  pwm_hz: 5000
  disable_with_s0: false
  s0_with_disable: false
  off_on_alarm: true
#switch off spindle when in alarm state
  spinup_ms: 10000
  spindown_ms: 12000
  tool_num: 0
  speed_map: 0=0% 0=25% 5868=25% 23237=99% 24000=100%
#DLC32 Spindle TTL max output voltage is 4.89v, so it can never reach 24000rpm/400Hz

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

start:
  must_home: false
# Pins that could be used: 0(SDA), 4(SCL), 5, 18, 19, 22, 23, 25, 26, 27, 32 (TTL spindle control), 33, 39, I2SO.7
# Pins not currently in use: 25 (TMC_CS_Y), 26 (TMC_CS_X), 27 (TMC_CS_Z), (39 SD card detect), I2SO.7
```

### Startup Messages

```text
*** Connecting to jserialcomm://COM5:115200
*** Fetching device status
*** Fetching device firmware version
*** Fetching device status codes
*** Fetching device state
*** Fetching device settings
*** Connected to FluidNC 3.6.4
```

### User Interface Software

UGS dev 2022-09-30

### What happened?

See below. I opened UGS, connected to FNC running 3.6.4pre1, pressed my hardware home XYZ button (macro 0) and got a hard limit warning during the homing cycle.

 **_Connected to FluidNC 3.6.4
DBG: Macro 0
DBG: Homing Cycle Z
DBG: Homing nextPhase FastApproach
DBG: Starting from 301.223,185.952,0.000
DBG: Planned move to 301.223,185.952,110.000 @ 100.000
DBG:  Z Pos Limit 1
DBG: Homing limited X Z
DBG: Homing nextPhase Pulloff0
DBG: Starting from 301.223,185.952,0.999
DBG: Planned move to 301.223,185.952,-0.001 @ 20.000
ERR: Macro can only be used in idle state
DBG:  Z Pos Limit 0
ERR: Macro can only be used in idle state
DBG: CycleStop Pulloff0
DBG: Homing nextPhase SlowApproach
DBG: Starting from 301.223,185.952,0.000
DBG: Planned move to 301.223,185.952,1.100 @ 20.000
DBG:  Z Pos Limit 1
DBG: Homing limited X Z
DBG: Homing nextPhase Pulloff1
DBG: Starting from 301.223,185.952,0.994
DBG: Planned move to 301.223,185.952,-0.006 @ 20.000
DBG:  Z Pos Limit 0
DBG: CycleStop Pulloff1
DBG: Homing nextPhase Pulloff2
DBG: mpos was 301.223,185.952,-0.005
DBG: mpos becomes 301.223,185.952,0.000
DBG: mpos transformed 301.223,185.952,0.000
DBG: Homing Cycle XY
DBG: Starting from 301.223,185.952,0.000
DBG: Planned move to 300.223,185.952,0.000 @ 28.284
DBG:  X Pos Limit 0
DBG:  X Pos Limit 1
DBG:  X Pos Limit 0
DBG: CycleStop PrePulloff
DBG: Homing nextPhase FastApproach
DBG: Starting from 300.223,185.952,0.000
DBG: Planned move to -31.977,-18.648,0.000 @ 282.843
DBG:  X Neg Limit 1
DBG: Homing limited X
DBG:  Y Neg Limit 1
DBG: Homing limited X Y
DBG: Homing nextPhase Pulloff0
DBG: Starting from -0.971,-1.008,0.000
DBG: Planned move to 0.029,-0.008,0.000 @ 28.284
DBG:  X Neg Limit 0
DBG:  X Neg Limit 1
DBG:  X Neg Limit 0
DBG:  Y Neg Limit 0
DBG: CycleStop Pulloff0
DBG: Homing nextPhase SlowApproach
DBG: Starting from 0.029,-0.007,0.000
DBG: Planned move to -1.071,-1.107,0.000 @ 28.284
DBG:  X Neg Limit 1
DBG: Homing limited X
DBG: CycleStop SlowApproach
DBG:  Y Neg Limit 1
DBG: Limit switch tripped for Y motor 0
 Connection closed_**

bdring, 2022-10-23:
Your macro is actually $X&$H slightly less weird than the $X$H that you posted on Discord.

I don't see the hard limit in the messages.

I tested on one of my machines. Here is the relevant config section. 

```yaml
control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: gpio.25
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
  
macros:
  startup_line0:
  startup_line1: 
  macro0: $X&$H
  macro1:
  macro2:
  macro3:
```

A macro can be executed via switch or command

___
- Via command

`$Macros/Run=0` or `$RM=0`

The machine homed perfectly

```
Grbl 3.6 [FluidNC v3.6.4-pre1 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
<Alarm|WPos:0.000,150.000,0.000|Bf:15,128|FS:0,0|WCO:0.000,-150.000,0.000>
$rm=0
[MSG:INFO: Running macro 0]
ok
[MSG:INFO: Caution: Unlocked]
<Idle|WPos:0.000,150.000,0.000|Bf:15,128|FS:0,0>
```
___

- Via Switch

The macro fails because it can only be used in idle state, so the $X is pretty worhless

```
Grbl 3.6 [FluidNC v3.6.4-pre1 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
<Alarm|WPos:0.000,150.000,0.000|Bf:15,128|FS:0,0|WCO:0.000,-150.000,0.000>
[MSG:ERR: Macro can only be used in idle state]
<Alarm|WPos:0.000,150.000,0.000|Bf:15,128|FS:0,0>
```

If the alarm state is cleared first, it homes fine.

```
Grbl 3.6 [FluidNC v3.6.4-pre1 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
<Alarm|WPos:0.000,150.000,0.000|Bf:15,128|FS:0,0|WCO:0.000,-150.000,0.000>
$X
[MSG:INFO: Caution: Unlocked]
ok
<Idle|WPos:0.000,150.000,0.000|Bf:15,128|FS:0,0>
<Home|WPos:0.000,150.000,0.220|Bf:15,128|FS:300,0>
<Home|WPos:0.000,150.000,1.000|Bf:15,128|FS:300,0|Pn:Z>
<Home|WPos:0.000,150.000,0.000|Bf:15,128|FS:533,0>
<Idle|WPos:0.000,150.000,0.000|Bf:15,128|FS:0,0>
```

If I continue to push the macro button during the homing, I get ERR messages, but it completes fine. I noticed you had those error messages. You might have a bouncy switch.
___

- Try homing normally via $H command a bunch of times to see if the macro is not the issue. 
- Try moving each axis to a separate cycle to see if that changes anything.

daxliniere, 2022-10-23:
> Your macro is actually $X&$H slightly less weird than the $X$H that you posted on Discord.
Yeah, I forgot that I had the & in the middle as required. My bad.

> I don't see the hard limit in the messages.
Really? It's the second last line in the log I included above, just before is disconnects from UGS.

**_DBG: Limit switch tripped for Y motor 0
Connection closed_**

> I tested on one of my machines. Here is the relevant config section.
> The machine homed perfectly
Yes, most of the time it works perfectly for me, too. But this time the hard limit tripped.

> The macro fails because it can only be used in idle state, so the $X is pretty worhless
But it doesn't usually fail, hence why I'm reporting this. And I usually have to press unlock in UGS, so it made sense to add Unlock to the macro.

> I noticed you had those error messages. You might have a bouncy switch.
Yes, perhaps. It's just a simple switch configured as **gpio.19:low:pu**. Could you please re-enable software debounce (just for non-limit-switch inputs).

> * Try homing normally via $H command a bunch of times to see if the macro is not the issue.
Why would the macro be an issue?

Is there a higher level of debug I could enable in case this ever happens again?

bdring, 2022-10-23:
We cannot look at debouncing until we have a long period of no issues. Everything is a very delicate balance of timing, resources, threads, etc. That is probably why the new Espressif library introduced so many issues,

daxliniere, 2022-10-23:
Okay, no worries, I totally understand.
The debounce library is already implemented but disabled, though, right? I remember discussions about users enabling it instead of creating proper input circuitry.

MitchBradley, 2022-10-23:
The debounce code was part of code that is long gone.

daxliniere, 2022-10-23:
Ahhh, I see. No worries. Thanks guys.

closed 2022-12-02 by MitchBradley