bdring/FluidNC #685: Problem: New Install install-fs.sh SAT SSID is not set]

xs23933, 2022-10-26:
### Controller Board

MKS-DLC32 V2.1

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [ ] Not Applicable

### Machine Description

MKS-DLC32 V2.1

### Configuration file

```yaml
Configuration file has not been uploaded
```

### Startup Messages

```text
Configuration file has not been uploaded
```

### What happened?

run ./install-fs.sh

select serial device

```shell
--- Available ports:
---  1: /dev/cu.BLTH         'n/a'
---  2: /dev/cu.Bluetooth-Incoming-Port 'n/a'
---  3: /dev/cu.usbserial-144420 'USB Serial'
---  4: /dev/cu.wchusbserial144420 'USB Serial'
--- Enter port index or full name: 3
--- FluidTerm v1.2.0 on /dev/cu.usbserial-144420  115200,8,N,1 ---
--- Quit: Ctrl+] or Ctrl+Q | Upload: Ctrl+U | Reset: Ctrl+R | ClearScreen: Ctrl+W ---

[MSG:INFO: FluidNC v3.6.3]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:WARN: Cannot open configuration file:config.yaml]
[MSG:INFO: Using default configuration]
[MSG:INFO: Axes: using defaults]
[MSG:INFO: Machine Default (Test Drive)]
[MSG:INFO: Board None]
[MSG:INFO: SPI not defined]
[MSG:INFO: No SD Card CS Pin]
[MSG:INFO: See http://wiki.fluidnc.com/en/config/sd_card#sdfallbackcs-access-sd-without-a-config-file]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-1000.000,0.000)]
[MSG:INFO: Axis Y (-1000.000,0.000)]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: STA SSID is not set]
Connection lost
Exception in thread tx:
Traceback (most recent call last):
  File "/Users/caisong/.fluidnc_venv/lib/python3.9/site-packages/serial/serialposix.py", line 621, in write
    n = os.write(self.fd, d)
OSError: [Errno 6] Device not configured

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/Caskroom/miniconda/base/lib/python3.9/threading.py", line 973, in _bootstrap_inner
    self.run()
  File "/usr/local/Caskroom/miniconda/base/lib/python3.9/threading.py", line 910, in run
    self._target(*self._args, **self._kwargs)
  File "/Users/caisong/Downloads/fluidnc-v3.6.3-posix/common/fluidterm.py", line 704, in writer
    self.enable_fluid_echo();
  File "/Users/caisong/Downloads/fluidnc-v3.6.3-posix/common/fluidterm.py", line 919, in enable_fluid_echo
    self.serial.write(self.tx_encoder.encode(right_arrow))
  File "/Users/caisong/.fluidnc_venv/lib/python3.9/site-packages/serial/serialposix.py", line 655, in write
    raise SerialException('write failed: {}'.format(e))
serial.serialutil.SerialException: write failed: [Errno 6] Device not configured

```

bdring, 2022-10-26:
It looks like to connection dropped. It could be randomly at that point or related to the radio tuning on. Some poorly designed ESP32 circuits cannot handle the small power surge when the radio turns on.

- Try restarting again to see if it consistently turns off there.
- Try installing the bluetooth version of the firmware to see if the radios are to the problem.
- Make sure you have a strong power supply

Unless this can be duplicated on other hardware, I would not consider it a firmware problem.

xs23933, 2022-10-26:
`
./install-fs.sh
`
and 
`
./install-wifi.sh
`
board its' working.
low to 115200
Not currently connected to the machine
Only computer usb power supply to flash firmware

closed 2022-10-26 by xs23933