bdring/FluidNC #686: Can't pipe output to fluidterm.exe

PBandParker, 2022-10-26:
### Controller Board

KeeYees ESP32 Development Board.

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

A CNC with a spindle and an X, Y, A axis cutting arm.

### Configuration file

```yaml
name: "ESP32 Dev Controller V4"
board: "ESP32 Dev Controller V4"

stepping:
  engine: RMT
  idle_ms: 255
  dir_delay_us: 0
  pulse_us: 10
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  
  x:
    steps_per_mm: 40
    max_rate_mm_per_min: 7500
    acceleration_mm_per_sec2: 1250
    max_travel_mm: 1000
    homing:
      cycle: 2
      mpos_mm: 0
      positive_direction: false
      seek_mm_per_min: 800
      feed_mm_per_min: 50
      settle_ms: 0
    
    motor0:
      limit_neg_pin: NO_PIN
      standard_stepper:
        direction_pin: gpio.14
        step_pin: gpio.13
    motor1:
      null_motor:
  y:
    steps_per_mm: 40
    max_rate_mm_per_min: 7500
    acceleration_mm_per_sec2: 1250
    max_travel_mm: 1000
    homing:
      cycle: 1
      mpos_mm: 0
      positive_direction: false
      seek_mm_per_min: 800
      feed_mm_per_min: 50
      settle_ms: 0
    
    motor0:
      limit_neg_pin: NO_PIN
      standard_stepper:
        direction_pin: gpio.4
        step_pin: gpio.5
    motor1:
      null_motor:
  
  a:
    steps_per_mm: 1.111
    max_rate_mm_per_min: 100000
    acceleration_mm_per_sec2: 16666
    max_travel_mm: 1000
    homing:
      cycle: 3
      mpos_mm: 0
      positive_direction: false
      seek_mm_per_min: 800
      feed_mm_per_min: 50
      settle_ms: 0
    
    motor0:
      limit_neg_pin: NO_PIN
      standard_stepper:
        direction_pin: gpio.16
        step_pin: gpio.17
    motor1:
      null_motor:
DAC:
  output_pin: gpio.25
  enable_pin: NO_PIN
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 100
  speed_map: 0=0.000% 3300=100.00%
  off_on_alarm: false
```

### Startup Messages

```text
setConsoleModes failed
Press any key to continue...
```

### User Interface Software

Command Line and a Python Script

### What happened?

I'm hoping to be able to send inputs to fluidterm from a program so I could log certain macros.  I tried searching for the setConsoleModes failure error in the wiki but I didn't see anything.

FluidNCControl.py below:
import numpy as np
import time
import os
import subprocess
import sys
p1 = subprocess.Popen(['py', '-3.9', 'FluidNCOutput.py'], stdout=subprocess.PIPE)
p2 = subprocess.run(['C:\\Users\\mylogin\\Documents\\fluidnc-v3.6.4-pre1-win64\\win64\\fluidterm.exe'], stdin=p1.stdout, creationflags=subprocess.CREATE_NEW_CONSOLE)

FluidNCOutput.py below:
print("$X")

MitchBradley, 2022-10-26:
FluidTerm is not designed for use as a component of a pipeline.  If you need this functionality you will need to modify the source code to recognize that input is coming from a pipe and avoid the calls to setConsoleMode that put the input into the modes suitable for character-at-a-time interaction.  The source code for the Windows native version is at https://github.com/MitchBradley/FluidTerm2 . Alternatively, you could modify the python version at fluidterm/fluidterm.py in this (FluidNC) repo and use that from your script, given that you are already using Python.

closed 2022-10-26 by MitchBradley

MitchBradley, 2022-10-26:
Most of the FluidTerm code handles user interaction subtleties.  For your purposes, it might be better to do everything in your Python script, using the Python serial package directly.  You could use fluidterm.py as a source of code snippets for finding,  opening, reading and writing to the correct port.  You would only need a very small portion of the fluidterm code.

PBandParker, 2022-10-26:
Thanks for responding so quickly.  I'll get on that, thanks for pointing me in the right direction.

PBandParker, 2022-11-01:
I ended up using the macro function provided in FluidNC to meet our needs.  As of right now, that looks like it will be sufficient.   Whether adding the ability to pipe stdout from other processes or extending the macros the be more than 4 is easier, I can't say.  And right now it looks like we won't need more than 4 macros, I'm just thinking ahead.

MitchBradley, 2022-11-01:
I am not inclined to add the pipe capability to FluidTerm, since it would essentially bypass 95% of what FluidTerm does.  All that would be left would be a trivial pipe-to-serial bridge.  It would be better to make a dedicated bridge, or find an existing one.