bdring/FluidNC #687: Problem: FluidNC response to $$

that-jim, 2022-10-27:
### Controller Board

6-pack

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

Gantry Router with XYYZ, HY spindle rs485 woring good. Wanting to use GRBL sender having problems with that.

### Configuration file

```yaml
not sure this applies to my question
```

### Startup Messages

```text
not sure this applies to my question
```

### User Interface Software

Openbuilds Control GRBL gcode sender

### What happened?

Is there a was to have FLuidNC respond to an $$ from the gcode sender? I believe the gcode sender wants to see mesages like $22=1 to enable homing button in the control software or $31 and $31 to fill in min and mas speed for the spindle control.

If there was say have a configurable response to $$ in the config.yaml file it would help with backward combability with grbl gcode senders.

### Other Information

na

bdring, 2022-10-27:
Do things like $22=1 need to be done **every time** you home in ***Openbuilds Control GRBL gcode sender***?

We have a vastly larger set of homing options than Grbl has. They are stored in the config file.

bdring, 2022-10-27:
This is our policy on Grbl compatibility.

http://wiki.fluidnc.com/en/features/grbl_compatibility

that-jim, 2022-10-27:
I believe what is happening is that openbuilds control does a $$ when it connects to a grbl controller, then it enables the home button and sets spindle speeds based on the settings that comes back from the grbl board. So I think if fluidnc could just spit out a $22=1 it would take care of it. I did try putting $22=1 in the 'meta:' but it did not work. I understand your policy, just trying to make the sender I'm used to work before I try others. Thanks and thanks for all your work on fluidnc it is pretty cool!

MitchBradley, 2022-10-27:
Rather than implementing a collection of hacks to work around quirks in senders, we are looking into contacting sender developers so they support FluidNC features more fully.  We already have traction with a few.  We will add OpenBuilds to the list.  UGS, CNCjs, and Candle are known to work.  gSender almost works and we have a workaround which we expect will be applied when the gSender developers have time; we are in contact with them.

that-jim, 2022-10-27:
Ok, thanks.

closed 2022-10-27 by that-jim

bdring, 2022-10-27:
OpenBuilds should default to enable homing and turn it off if it receives a $22=0 from old school grbl.

that-jim, 2022-10-27:
Thanks Bart, your right. I played with this more and found that openbuilds control seems to keep the last settings it received from grbl unless you refresh from the grbl settings tab, this must be must be what I did. So if the last grbl board you had plugged in had the settings setting you are looking for your probably good.