bdring/FluidNC #688: Problem: Servo travel intermittently slow when using serial port

MarkJB, 2022-10-30:
### Controller Board

[ESP32_Plotter_Controller (Emerald Dingo RevB)](https://github.com/theworkisthework/ESP32_Plotter_Controller)

### Machine Description

Wallbot

### Configuration file

```yaml
name: ESP32_Plotter_Controller_RevB_Emerald_Dingo
board: Emerald Dingo (RevB)

kinematics:
  WallPlotter:
    left_axis: 0
    left_anchor_x: -326.500
    left_anchor_y: 120.000
    right_axis: 1
    right_anchor_x: 326.000
    right_anchor_y: 120.000
    segment_length: 10.000

stepping:
  engine: RMT
  idle_ms: 255
  dir_delay_us: 0
  pulse_us: 3
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.2
  x:
    # Pully = 100mm circumference
    # Steppers are 1.8deg or 200 divs per rev * 16 micro steps = 3200 steps per rev / 100m = 32 steps per mm
    steps_per_mm: 32.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      mpos_mm: 0.000
      positive_direction: true
      settle_ms: 250.000
      seek_mm_per_min: 2000.000
      feed_mm_per_min: 200.000
      seek_scaler: 1.100
      feed_scaler: 5.000
    motor0:
      limit_all_pin: NO_PIN
      hard_limits: false
      tmc_2130:
        toff_disable: 1
        direction_pin: gpio.26
        step_pin: gpio.25
        run_amps: 0.350
        hold_amps: 0.25
        microsteps: 16
        stallguard: 16
        stallguard_debug: false
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        r_sense_ohms: 0.110
        cs_pin: gpio.17
    motor1:
      null_motor:
  y:
[... 308 lines cut ...]
<Run|MPos:-125.865,-542.939,5.000|FS:545,0>

G1 X30.939 Y3.955 Z5 F1500.000
Waiting for grbl
Command received: ok

Waiting for grbl
<Run|MPos:-133.701,-545.521,5.000|FS:1872,0>

G1 X32.000 Y7.000 Z5 F1500.000
Waiting for grbl
Command received: ok

Waiting for grbl
<Run|MPos:-131.004,-537.010,3.340|FS:1731,0>

G1 X30.061 Y8.955 Z5 F1500.000
Waiting for grbl
Command received: ok

Waiting for grbl
<Run|MPos:-111.643,-536.660,-3.880|FS:1618,0|WCO:-140.000,-550.000,0.000>

G1 X23.245 Y6.455 Z5 F1500.000
Waiting for grbl```

MarkJB, 2022-10-30:
Doesn't seem to be related to acceleration settings or work offsets...

MarkJB, 2022-10-30:
Seems the serial port is a red herring as well.

I wrote the gcode to a file (attached), uploaded it and ran it from the webui and am seeing the same issue.
[mind-glyphs.txt](https://github.com/bdring/FluidNC/files/9896769/mind-glyphs.txt)

bdring, 2022-10-30:
I can replicate the problem. It looks like a feed rate compensation problem.

MarkJB, 2022-10-31:
So far so good.

I have a few longer running plots to try and I'll report back.

MarkJB, 2022-10-31:
Not seen the servo move slowly in a couple of hours, but the next move is happening before the servo finishes its travel. Is there a setting to control the settle time? (Was looking for something like `/axes/z/homing/settle_ms` but for moves rather than homing).

MarkJB, 2022-10-31:
Something odd happened while I was attempting to speed up the servo. After cranking up the z_axis feedrate and acceleration, I was manually sending `G0 Z0` & G0 Z5` to test and the machine went off in a random direction and `nan` shows in the console.

I'm not able to reproduce it yet, but if I do I'll open another issue.

```
<Idle|MPos:-93.937,-544.058,0.000|FS:0,0>
G0 Z5
ok
G0 Z0
ok
<Idle|MPos:-93.937,-544.058,0.000|FS:0,0>
<Idle|MPos:-93.937,-544.058,0.000|FS:0,0>
G0 Z5
ok
<Idle|MPos:-93.937,-544.058,5.000|FS:0,0>
G0 Z0
ok
<Idle|MPos:-93.937,-544.058,0.000|FS:0,0>
G0 Z5
ok
<Idle|MPos:-93.937,-544.058,5.000|FS:0,0|WCO:0.000,-0.000,0.001>
G0 Z5
ok
<Run|MPos:19.352,-449.975,5.000|FS:5175,0|Ov:100,100,100>
<Run|MPos:155.031,-237.756,5.000|FS:5175,0>
G0 Z0
ok
<Run|MPos:201.193,43.803,5.000|FS:5175,0>
<Run|MPos:158.648,nan,5.000|FS:5175,0>
<Run|MPos:167.553,nan,4.380|FS:5175,0>
<Run|MPos:200.435,-1.909,3.110|FS:5175,0>
<Run|MPos:144.513,-263.240,1.840|FS:5175,0>
<Run|MPos:-0.411,-469.699,0.570|FS:5175,0>
<Idle|MPos:-93.937,-544.058,0.000|FS:0,0>
<Idle|MPos:-93.937,-544.058,0.000|FS:0,0|WCO:0.000,-0.000,0.001>
<Idle|MPos:-93.937,-544.058,0.000|FS:0,0|Ov:100,100,100>
G0 Z0
<Idle|MPos:-93.937,-544.058,0.000|FS:0,0>
ok
<Run|MPos:65.364,-396.611,0.000|FS:5175,0>
<Run|MPos:176.207,-173.108,0.000|FS:5175,0>
<Run|MPos:198.476,nan,0.000|FS:5175,0>
<Run|MPos:131.678,nan,0.000|FS:5155,0>
<Idle|MPos:119.671,nan,0.000|FS:0,0>
<Idle|MPos:119.671,nan,0.000|FS:0,0>
<Idle|MPos:119.671,nan,0.000|FS:0,0>
<Idle|MPos:119.671,nan,0.000|FS:0,0|WCO:0.000,-0.000,0.001>
<Idle|MPos:119.671,nan,0.000|FS:0,0|Ov:100,100,100>
<Idle|MPos:119.671,nan,0.000|FS:0,0>
<Idle|MPos:119.671,nan,0.000|FS:0,0>
<Idle|MPos:119.671,nan,0.000|FS:0,0>
<Idle|MPos:119.671,nan,0.000|FS:0,0>
<Idle|MPos:119.671,nan,0.000|FS:0,0>
<Idle|MPos:119.671,nan,0.000|FS:0,0>
<Idle|MPos:119.671,nan,0.000|FS:0,0>
<Idle|MPos:119.671,nan,0.000|FS:0,0>
<Idle|MPos:119.671,nan,0.000|FS:0,0|WCO:0.000,-0.000,0.001>
<Idle|MPos:119.671,nan,0.000|FS:0,0|Ov:100,100,100>
<Idle|MPos:119.671,nan,0.000|FS:0,0>
<Idle|MPos:119.671,nan,0.000|FS:0,0>
<Idle|MPos:119.671,nan,0.000|FS:0,0>

MarkJB, 2022-10-31:
Off the back of tweaking the feeds and speeds for the z-axis, I've noticed the servo its slower going from Z5 -> Z0, than it is the other way.

bdring, 2022-10-31:
The `nan` means something is wrong.  The wallbot kinematics was contributed by someone not on the development team. We are trying to patch it up without a machine to test with. It could be moving into a bad area. The 119 is close to the Y position of the anchor. We can add code to warn you if you are getting out of the valid area of motion.

```yaml
kinematics:
  WallPlotter:
    left_axis: 0
    left_anchor_x: -326.500
    left_anchor_y: 120.000
    right_axis: 1
    right_anchor_x: 326.000
    right_anchor_y: 120.000
    segment_length: 10.000
```
I noticed your servo range is this.

[MSG:INFO: Axis Z (-5.000,0.000)]

Do you have a G54 offset?

MarkJB, 2022-10-31:
Yes, I am using this offset `G10 L2 P2 X-140 Y-550 Z5`

MarkJB, 2022-10-31:
The servo range should be 0.000,5.000.

bdring, 2022-10-31:
I noticed you don't have limit switches.

Can you sent a link to the hardware you are using, exactly how you start a job and why you set that offset?

The range is in machine space. If you want 0 to 5mm, you need to set the homing mpos_mm to 5, because you home positive on Z

MarkJB, 2022-10-31:
The[ controller board](https://github.com/theworkisthework/ESP32_Plotter_Controller) is one of my own design, but pin compatible with one of yours.

The rest of the machine is a self build.

The offset is to make working with the machines coordinate system easier. The semi-circle at the top is 'home' and the processing sketch starts at the 0,0 bottom left of the page, hence the offset.

[image]

MarkJB, 2022-10-31:
Sorry - forgot to say I start a job by homing (if the gondola isn't at the home position when the machine thinks it is the plot will be skewed) then applying the offset, either manually or as part of the gcode (the processing sketch will send the offset and activate it), then it starts streaming the gcode.

But for the recent tests, I've been starting the gcode files from the webui with the play button.

bdring, 2022-10-31:
I am not sure, but I assume you manually move the gondola to the center of Y and 120 below the "anchors" and then start the controller.

MarkJB, 2022-10-31:
> (edited) I am not sure, but I assume you manually move the gondola to the center of X and 120 below the "anchors" and then start the controller.
> 
> 

Yes, that's correct.

MitchBradley, 2022-10-31:
I pushed a change to that branch that probably fixes the problem.  The trigger for the problem was a move that went nowhere, in this case G0 Z5 right after G0 Z5.

MarkJB, 2022-11-02:
So far so good. Been plotting on and off for 24hrs and not seen `nan` or crazy moves.

When this plot is finished I'll try and force some crazy moves and see what happens.

R.e. the issue with the next move starting before the servo reaches its expected position, is this expected? Is there a setting to delay until the servo has likely finished moving? Or should I raise a separate issue for it?

bdring, 2022-11-02:
The servo is not closed loop (with FluidNC) . It is just doing its best to track the position. If you set the speed and acceleration higher than the servo can go, the next move will start before the servo completes it motion.

Try lowering the `max_rate_mm_per_min` for the Z axis.

Most servos can do a full move in a little less than a second. For a 5mm Z move, I would try a rate in the 200-400 range. You can tune that value live by sending...
```
$axes/z/max_rate_mm_per_min=300
```

Once you like the performance you need to save your config file with that value so it reads it a start up.

closed 2022-11-08 by bdring