bdring/FluidNC #689: Criticial error with SDDET pin - does not have expected behaviour

HoverClub, 2022-10-31:
### Controller Board

MKS TinyBee

### Machine Description

Gantry Router XY

### Input Circuits

```yaml
https://github.com/makerbase-mks/MKS-TinyBee/blob/main/hardware/MKS%20TinyBee%20V1.0_001/MKS%20TinyBee%20V1.0_001%20PIN.pdf
```

### Configuration file

```yaml
# last edit 16/8/22
board: MKS TinyBee V1.0
name: HovTek_CNC

kinematics:
  Cartesian:

i2so:
  bck_pin: gpio.25
  data_pin: gpio.27
  ws_pin: gpio.26

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  # uses TH2 IO34 active low - MAKE SURE jumper J2 is set to SDDET!!!
  card_detect_pin: gpio.34:low

stepping:
  # I2S_STATIC to make sure laser is sync'ed with steppers!
  engine: I2S_STATIC
  idle_ms: 255
  # 4usec min. - I2S bit rate 
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 2

axes:
  x:
    # X = 1/8 uStep
    steps_per_mm: 26.66
    max_rate_mm_per_min: 8000.000
    acceleration_mm_per_sec2: 80.000
    max_travel_mm: 2500.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 1500.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.33:low:pu
      hard_limits: true
      pulloff_mm: 4.000
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2
        disable_pin: I2SO.0

    # use E0 driver for 2nd X axis motor!
    motor1:
      hard_limits: true
      pulloff_mm: 4.000
      stepstick:
        step_pin: I2SO.10
        direction_pin: I2SO.11
        disable_pin: I2SO.9

  y:
    # Y = 1/8 uStep
    steps_per_mm: 26.66
    max_rate_mm_per_min: 8000.000
    acceleration_mm_per_sec2: 70.000
    max_travel_mm: 1250.000
    soft_limits: false
    homing:
      cycle: 3
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 2000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.32:low:pu
      hard_limits: true
      pulloff_mm: 4.000
      stepstick:
        step_pin: I2SO.4
        # invert DIR signal from default high
        direction_pin: I2SO.5:low
        disable_pin: I2SO.3

  z:
    # Z = 1/8 uStep
    steps_per_mm: 200.000
    max_rate_mm_per_min: 8000.000
    acceleration_mm_per_sec2: 60.000
    max_travel_mm: 80.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 500.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_pos_pin: gpio.22:low:pu
      hard_limits: true
      pulloff_mm: 3.000
      stepstick:
        step_pin: I2SO.7
        direction_pin: I2SO.8
        disable_pin: I2SO.6

control:
  safety_door_pin: NO_PIN
  # on MT_DET connector
  reset_pin: gpio.35:low
  # on TH1 connector
  feed_hold_pin: gpio.36:low
  # on TB connector
  cycle_start_pin: gpio.39:low
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

# spindle PWM signal
PWM:
  pwm_hz: 2500
  # on EXP1 IO15 connector
  output_pin: gpio.15:high
  s0_with_disable: true
  tool_num: 0
  spinup_ms: 4000
  spindown_ms: 4000
  speed_map: 0=0.000% 12000=100.000%
 
Laser:
  pwm_hz: 5000
  # on 3D Touch connector (has pdwn + PWM)
  output_pin: gpio.2:high:pd
  s0_with_disable: true
  tool_num: 1
  # same scaling as spindle!
  speed_map: 0=0.000% 12000=100.000%

start:
  must_home: false
```

### Startup Messages

```text
[MSG:INFO Init OLED SDA:gpio.21 SCL:gpio.4]
    [MSG:INFO: FluidNC v3.6.2]
    [MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
    [MSG:INFO: Local filesystem type is spiffs]
    [MSG:INFO: Configuration file:config.yaml]
    [MSG:INFO: Machine HovTek_CNC]
    [MSG:INFO: Board MKS TinyBee V1.0]
    [MSG:INFO: I2SO BCK:gpio.25 WS:gpio.26 DATA:gpio.27]
    [MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
    [MSG:INFO: SD Card Detect gpio.34:low]
    [MSG:INFO: SD Card cs_pin:gpio.5 detect:gpio.34:low]
    [MSG:ERR: Critical error in main_init: Requested pin gpio.34:low does not have the expected behavior.]
    [MSG:INFO: STA SSID is not set]
    [MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
    [MSG:INFO: AP started]
    [MSG:INFO: WiFi on]
    [MSG:INFO: Captive Portal Started]
    [MSG:INFO: HTTP started on port 80]
    [MSG:INFO: Telnet started on port 23]

    Grbl 3.6 [FluidNC v3.6.2 (wifi) '$' for help]
    [MSG:INFO: Configuration is invalid. Check boot messages for ERR's.]
```

### User Interface Software

FluidTerm

### What happened?

Throws startup error for SDDETECT pin ("does not have the expected behaviour").   I've verified that the hardware pin is working correctly (and checked the ESP32 reads the correct value).  Works OK on earlier version of FluidNC (3.5) but not on 3.6.2  If I disable the SDDETECT pin in config.yaml then all is good as you'd expect.   

### Other Information

Note - compile includes OLED support using GPIO 4 & 21
oled_basic.cpp:.
`void display_init() {
    init_oled(0x3c, GPIO_NUM_21, GPIO_NUM_4, GEOMETRY_128_64);`

bdring, 2022-10-31:
The pin capabilities are not set correctly in the firmware. `Pin::Capabilities::Output` should be `Pin::Capabilities::Input`

```cpp
    if (_cardDetect.defined()) {
        _cardDetect.setAttr(Pin::Attr::Input);
        auto cdPin = _cardDetect.getNative(Pin::Capabilities::Output | Pin::Capabilities::Native);
        sd_init_slot(csPin, cdPin);
    } else {
        sd_init_slot(csPin);
    }
 ```
 
 Currently the card detect pin is not used in the firmware. That is probably why the issue was not noticed. The plan is to add card detect features in the future.
 
 We are working on a better protocol where state changes are pushed to the clients. This is needed for a good way to deal with an SD card detect feature.
 
 I can fix this issue on the next release, but it still won't so anything for a while.

bdring, 2022-10-31:
This has been added to the devt branch.  You can compile and test that or wait for the next release.

closed 2022-10-31 by bdring