bdring/FluidNC #693: Problem: Steppers wiggle instead of turning smoothly

dustMason, 2022-11-02:
### Controller Board

Pen/Laser TMC2209 ([from tindie](https://www.tindie.com/products/33366583/fluidnc-penlaser-cnc-controller-tmc2209/))

### Machine Description

Polargraph style machine with these [NEMA17 steppers](https://cdn-shop.adafruit.com/product-files/324/C140-A+datasheet.jpg).

### Configuration file

```yaml
name: "TMC2209 XY Servo Laser"
board: "FluidNC Pen/Laser 2209"

#kinematics:
#  WallPlotter:
#    left_axis: 0
#    left_anchor_x: -428.000
#    left_anchor_y: 520.00
#    right_axis: 1
#    right_anchor_x: 428.000
#    right_anchor_y: 520.00
#    segment_length: 2.0

stepping:
  engine: RMT
  idle_ms: 255
  dir_delay_us: 1
  pulse_us: 2
  disable_delay_us: 0
  
start:
  must_home: false
  deactivate_parking: false
  check_limits: false

axes:
  shared_stepper_disable_pin: gpio.13:high
  
  x:
    steps_per_mm: 6.06
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100
    
    motor0:
      limit_neg_pin: gpio.39:low
      limit_neg_pin: gpio.36:low
      #Diag pin is gpio.34
      tmc_2209:
        uart:
          txd_pin: gpio.17
          rxd_pin: gpio.16
          rts_pin: NO_PIN
          cts_pin: NO_PIN
          baud: 115200
          mode: 8N1

        addr: 0
        r_sense_ohms: 0.68
        run_amps: 0.500
        hold_amps: 0.500
        microsteps: 4
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: false
        direction_pin: gpio.12
        step_pin: gpio.14
        disable_pin: NO_PIN
    
    
    motor1:
      null_motor:

  y:
    steps_per_mm: 6.06
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.15:low:pu
      limit_neg_pin: gpio.4:low:pu
      #diag pin is gpio.35
      tmc_2209:
        addr: 1
        r_sense_ohms: 0.68
        run_amps: 0.500
        hold_amps: 0.500
        microsteps: 4
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: false
        direction_pin: gpio.26
        step_pin: gpio.25
        disable_pin: NO_PIN
    motor1:
      null_motor:

  z:
    steps_per_mm: 100
    max_rate_mm_per_min: 1000
    acceleration_mm_per_sec2: 50
    max_travel_mm: 5.00
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 5
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100
    
    motor0:
      limit_neg_pin: gpio.33:low
      limit_neg_pin: gpio.32:low
      rc_servo:
        pwm_hz: 50
        output_pin: gpio.22
        min_pulse_us: 1000
        max_pulse_us: 2100
        
spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

coolant:
  flood_pin: NO_PIN
  mist_pin:  gpio.21
  
  
Laser:
  pwm_hz: 5000
  output_pin: gpio.27
  enable_pin: gpio.2
  disable_with_s0: false
  s0_with_disable: true
  tool_num: 0
  speed_map: 0=0.000% 255=100.000%
```

### Startup Messages

```text
$G
[MSG:ERR: sdmmc_card_init failed code 0x108]
<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:-1.000,0.000,0.000>
$J=G91 G21 F1000 X10
```

### User Interface Software

WebUI

### What happened?

When jogging the motors at any feed rate, they only wiggle back and forth. [This short video](https://www.youtube.com/shorts/uPTQkB-1TW4) shows the behavior. Following advice from folks on discord, I rewired the motor connections and verified with a multimeter that they are hooked up properly.

MitchBradley, 2022-11-02:
We need the startup messages as given by $SS

Please use FluidTerm, as we have better control of exactly what is happening compared to WebUI.

bdring, 2022-11-02:
The stepper motor you are using is odd.

[image]

The typical resistance per phase for a modern motor that size is about 3-8 Ohm per coil.
The voltage is 12V and the current is  0.35A

It appears that motor was design to be driven directly by 12V with no current control.

In order for you to control it with a chopper to the current you want, you must drive it at a much higher voltage.

It cannot reach your requested current at 12V. Most motors have a voltage much lower than 12

bdring, 2022-11-02:
The rated voltage on stepper motors, just means the voltage at which the current is measured. You typically can and should run them at 10x-25x the rated voltage when used with stepper drivers.

bdring, 2022-11-02:
Try lowering the all currents in the config file to 0.20 amps.

bdring, 2022-11-02:
Here is a more modern NEMA 17. It does not show the rated voltage, but you can infer it to be about 2.6 Ohms

[image]

dustMason, 2022-11-02:
Thanks for all the info. I adjusted the currents in the config to `0.2` as suggested, and with `microsteps` set to `1` and `steps_per_mm` at `60.6` and a feed rate of 500, the motors move smoothly once they have accelerated. However when being driven at low speed, they seem to move randomly.

Based on your look at the motors, do you think that they are ultimately incompatible with these drivers?

bdring, 2022-11-02:
The r_sense_ohms should be 0.11.

bdring, 2022-11-02:
Try replacing your X and Y section of the config with this. Be sure to keep the other sections

This will be a basic config for testing.

```yaml
  x:
    steps_per_mm: 100
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100
    
    motor0:
      limit_neg_pin: gpio.36:low
      #Diag pin is gpio.34
      tmc_2209:
        uart:
          txd_pin: gpio.17
          rxd_pin: gpio.16
          rts_pin: NO_PIN
          cts_pin: NO_PIN
          baud: 115200
          mode: 8N1

        addr: 0
        r_sense_ohms: 0.110
        run_amps: 0.200
        hold_amps: 0.200
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: Coolstep
        homing_mode: Coolstep
        use_enable: false
        direction_pin: gpio.12
        step_pin: gpio.14
        disable_pin: NO_PIN
    
    
    motor1:
      null_motor:

  y:
    steps_per_mm: 100
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.39:low
      #diag pin is gpio.35
      tmc_2209:
        addr: 1
        r_sense_ohms: 0.110
        run_amps: 0.200
        hold_amps: 0.200
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: Coolstep
        homing_mode: Coolstep
        use_enable: false
        direction_pin: gpio.26
        step_pin: gpio.25
        disable_pin: NO_PIN
    motor1:
      null_motor:
 ```

bdring, 2022-11-02:
The r_sense is a physical resistor soldered on the board. You cannot change it, but you do need to tell FluidNC what it is.

TLDR...Leave it at 0.11 because that is what is on the board.

bdring, 2022-11-02:
What does the other motor do?

dustMason, 2022-11-02:
Both motors behave the same.

bdring, 2022-11-02:
It looks like a classic mis-wire to me. 

Remove the motor from the mount. Wire the motor wires directly to the PCB terminal block. Wire in this order from left to right Red, Yellow, Grey, Green.

Test and take a picture.

MitchBradley, 2022-11-02:
Also please show the output from $SS.

bdring, 2022-11-02:
Any of those NEMA 17 motors on Stepper online should work.

If you post again in hardware, you will get a timeout.

dustMason, 2022-11-02:
@bdring You were right from the beginning. Wiring the motor as Red/Yellow/Grey/Green has the motor turning smoothly. It makes a lot of noise, though, even while idling. 

To clarify my understanding of the pinout (pictured below) – I suppose this means that 2A and 2B connect to one coil (red & yellow), while 1A and 1B (grey & green) connect to the other? I think I failed to properly clarify this question in the discord.

[image]

Thanks for your help and apologies for the noise in the #hardware channel on the discord

MitchBradley, 2022-11-02:
1a and 1b are one coil, 2a and 2b are the other coil

MitchBradley, 2022-11-02:
is the motor hot?

dustMason, 2022-11-02:
Thanks. I think I got the wrong answer to my initial question! Again, I appreciate the help. I'll close this issue as solved.

closed 2022-11-02 by dustMason