bdring/FluidNC #694: Problem: wallplotter kinematics – how to home the machine?

dustMason, 2022-11-03:
### Controller Board

Pen/Laser TMC2209 ([from tindie](https://www.tindie.com/products/33366583/fluidnc-penlaser-cnc-controller-tmc2209/))

### Machine Description

Polargraph style machine with these [NEMA17 steppers](https://cdn-shop.adafruit.com/product-files/324/C140-A+datasheet.jpg).

### Configuration file

```yaml
name: "TMC2209 XY Servo Laser"
board: "FluidNC Pen/Laser 2209"

kinematics:
  WallPlotter:
    left_axis: 0
    left_anchor_x: -403.22
    left_anchor_y: 265.18
    right_axis: 1
    right_anchor_x: 403.22
    right_anchor_y: 265.18
    segment_length: 2.0

stepping:
  engine: RMT
  idle_ms: 255
  dir_delay_us: 0
  pulse_us: 4
  disable_delay_us: 0
  
start:
  must_home: false
  deactivate_parking: false
  check_limits: false

axes:
  shared_stepper_disable_pin: gpio.13:high
  
  x:
    steps_per_mm: 12.12
    max_rate_mm_per_min: 4000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 406
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.39:low
      limit_neg_pin: gpio.36:low
      #Diag pin is gpio.34
      tmc_2209:
        uart:
          txd_pin: gpio.17
          rxd_pin: gpio.16
          rts_pin: NO_PIN
          cts_pin: NO_PIN
          baud: 115200
          mode: 8N1

        addr: 0
        r_sense_ohms: 0.11
        run_amps: 0.20
        hold_amps: 0.20
        microsteps: 8
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        direction_pin: gpio.12
        step_pin: gpio.14
        disable_pin: NO_PIN

    motor1:
      null_motor:

  y:
    steps_per_mm: 12.12
    max_rate_mm_per_min: 4000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 457
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.15:low:pu
      limit_neg_pin: gpio.4:low:pu
      #diag pin is gpio.35
      tmc_2209:
        addr: 1
        r_sense_ohms: 0.11
        run_amps: 0.20
        hold_amps: 0.20
        microsteps: 8
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        direction_pin: gpio.26
        step_pin: gpio.25
        disable_pin: NO_PIN
    motor1:
      null_motor:
        
spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

coolant:
  flood_pin: NO_PIN
  mist_pin:  gpio.21
  
  
Laser:
  pwm_hz: 5000
  output_pin: gpio.27
  enable_pin: gpio.2
  disable_with_s0: false
  s0_with_disable: true
  tool_num: 0
  speed_map: 0=0.000% 255=100.000%
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.6.3]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:pen_2209_servo_laser.yaml]
[MSG:INFO: Machine TMC2209 XY Servo Laser]
[MSG:INFO: Board FluidNC Pen/Laser 2209]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.13]
[MSG:INFO: Axis X (150.000,556.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Trinamic Stepper Uart Tx:gpio.17 Rx:gpio.16 RTS:NO_PIN Baud:115200]
[MSG:INFO:     tmc_2209 Step:gpio.14 Dir:gpio.12 Disable:NO_PIN Addr:0 R:0.110]
[MSG:INFO:  X Neg Limit gpio.36:low]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (150.000,607.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2209 Step:gpio.25 Dir:gpio.26 Disable:NO_PIN Addr:1 R:0.110]
[MSG:INFO:  Y Neg Limit gpio.4:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: Kinematic system: WallPlotter]
[MSG:INFO: Laser Ena:gpio.2 Out:gpio.27 Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: Mist coolant gpio.21]
[MSG:INFO: Connecting to STA SSID:Geranium]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connecting...]
[MSG:INFO: Connected - IP is 192.168.4.153]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
```

### User Interface Software

WebUI

### What happened?

Before starting the machine, I manually center the pen 403.22mm to the right of and 265.18mm below the left-hand motor. I understood the `anchor` settings under `kinematics/Wallplotter` to be used this way. However, my plots have arced horizontal lines (pictured, amongst other test scribbles below)

I am wondering what the intended flow for setting up a plot should be, using the wallplotter kinematics.
- Should the gcode coordinates be offset, such that 0,0 is at the physical center of the drawing area?
- Where should the 0,0 machine coordinates fall in physical space on the machine?

cc @forbin1ab on advice from folks in the discord, as someone with context here.

[image]

bdring, 2022-11-04:
@MarkJB on this issue may be able to help

https://github.com/bdring/FluidNC/issues/688

MarkJB, 2022-11-04:
On my machine 0,0 is in the top middle.

I define my machine coords as -326, 120 (left motor) and 326, 120 (right motor) which puts 0,0 120 below the line of the motor shafts and equal distance between them.

Homing involves physically moving the gondola to home and resetting the controller.

Then you can set an offset in your gcode software or as a machine work offset.

dustMason, 2022-11-04:
This is very helpful, thanks. I've adjusted my setup and config but the machine still behaves strangely.

My kinematic settings:

```yaml
kinematics:
  WallPlotter:
    left_axis: 0
    left_anchor_x: -403.500
    left_anchor_y: 197.000
    right_axis: 1
    right_anchor_x: 403.000
    right_anchor_y: 197.000
    segment_length: 10.000
```

When I start the machine, the pen rests halfway between the two motors, 197mm below them. The WebUI shows work and machine coordinates each at 0,0. When I send `$J=G91 G21 F4000 Y300`, the pen moves down about 90mm and then _moves back up about 12mm_! After doing this, the machine and work coordinates both read X0 Y93.95

When you physically home your machine, do the machine and work coordinates read 0,0?

bdring, 2022-11-04:
You might have your motors (right vs left) mixed up or running backwards.

I think you can use the homing to control the motors individually. Be ready to click reset, because there is nothing to stop the motors.

You have both X and Y as `positive_direction: false`

Try sending $HX. The left motor should feed out more line. 
- If the wrong motor moves, swap the wires to the other driver.
- If the motor runs the wrong way, add the `:low` attribute to the driection pin.

Once the left (X) is right, do the same for the other motor with $HY

bdring, 2022-11-04:
I might have the direction info backwards. Feeding out line might be a positive move. I have not studied that code.

MarkJB, 2022-11-04:
> When you physically home your machine, do the machine and work coordinates read 0,0?

Yes, both machine and work coordinates read 0.

dustMason, 2022-11-04:
Thanks @bdring, once again your intuition is good :) I wasn't able to use `$HX` to debug (`[MSG:ERR: This kinematic system cannot do homing]`). However your response made me realize that the motors must be connected backwards. Swapping the wires results in straight horizontal lines.

Thanks @MarkJB for your help as well. Closing this issue as resolved.

closed 2022-11-04 by dustMason

bdring, 2022-11-04:
Does positive motion feed out more cord?

MitchBradley, 2022-11-04:
Positive motion in Y would have to retract both cords, assuming positive is up.

In X it would have to feed out the left and retract the right

bdring, 2022-11-04:
I meant in motor space or with cartesian kinematics. 

Would $J=G91 F100 X10 extend the X cord?

dustMason, 2022-11-04:
Yes, positive Y motion retracts both chains. `$J=G91 F100 X10` causes one motor to retract and the other to feed out chain. The rate at which this happens depends on the starting position, in order to create perfectly linear movement.

bdring, 2022-11-04:
Of course it does!

Not my question. I am trying to write a procedure to help the next guy from randomly wiring and asking for help.

Whatever

dustMason, 2022-11-04:
I must have misunderstood your question. To be specific, positive X movement _does_ indeed extend the chain on the motor connected to the X motor. I'll write up my debugging experience as a blog post and if any of it seems useful to you I can add submit a documentation PR about it. Thanks again for the help