bdring/FluidNC #696: Problem: v3.6.5-pre2 jerky motion / inaccurate milling

tgm02281961, 2022-11-07:
### Controller Board

6-pack internal with 8825s and (2) 4X input modules

### Machine Description

V1 Engineering Lowrider 3 (XYYZZ)
[image]

### Configuration file

```yaml
board: 6 Pack
name: Lowrider 6 Pack XYYZZ PMW
meta: 
stepping:
  engine: I2S_static
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
  segments: 12
axes:
  shared_stepper_disable_pin: NO_PIN
  shared_stepper_reset_pin: NO_PIN
  x:
    steps_per_mm: 50.000
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 800.000
    soft_limits: false
    homing:
      cycle: 2
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 1000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100
    motor0:
      limit_neg_pin: gpio.33:pu
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0
  y:
    steps_per_mm: 50.000
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 600.000
    soft_limits: false
    homing:
      cycle: 3
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 1000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100
    motor0:
      limit_neg_pin: gpio.32:pu
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 2.500
      standard_stepper:
        step_pin: I2SO.5
        direction_pin: I2SO.4
        disable_pin: I2SO.7
    motor1:
      limit_neg_pin: gpio.35
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 2.000
      standard_stepper:
        step_pin: I2SO.10
        direction_pin: I2SO.9
        disable_pin: I2SO.8
  z:
    steps_per_mm: 200.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 100.000
    soft_limits: false
    homing:
      cycle: 1
      allow_single_axis: true
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 400.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.34
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 2.000
      standard_stepper:
        step_pin: I2SO.13
        direction_pin: I2SO.12
        disable_pin: I2SO.15
    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.2:pu
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 2.750
      standard_stepper:
        step_pin: I2SO.18
        direction_pin: I2SO.17
        disable_pin: I2SO.16
kinematics:
  Cartesian:
i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17
spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18
sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN
control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0
probe:
  pin: NO_PIN
  check_mode_start: true
macros:
  startup_line0: 
  startup_line1: 
  macro0: 
  macro1: 
  macro2: 
  macro3: 
start:
  must_home: false
  deactivate_parking: false
  check_limits: false
parking:
  enable: false
  axis: Z
  target_mpos_mm: -5.000
  rate_mm_per_min: 800.000
  pullout_distance_mm: 5.000
  pullout_rate_mm_per_min: 250.000
user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN
arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
planner_blocks: 16
PWM:
  pwm_hz: 5000
  direction_pin: NO_PIN
  output_pin: gpio.14
  enable_pin: gpio.13
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 1000
  spindown_ms: 1000
  tool_num: 0
  speed_map: 0=0.000% 1000=100.000%
  off_on_alarm: false
```

### Startup Messages

```text
$ss
[MSG:INFO: FluidNC v3.6.5-pre2]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine Lowrider 6 Pack XYYZZ PMW]
[MSG:INFO: Board 6 Pack]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,800.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.2 Dir:I2SO.1 Disable:I2SO.0]
[MSG:INFO:  X Neg Limit gpio.33:pu]
[MSG:INFO: Axis Y (0.000,600.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.5 Dir:I2SO.4 Disable:I2SO.7]
[MSG:INFO:  Y Neg Limit gpio.32:pu]
[MSG:INFO:   Motor1]
[MSG:INFO:     standard_stepper Step:I2SO.10 Dir:I2SO.9 Disable:I2SO.8]
[MSG:INFO:  Y2 Neg Limit gpio.35]
[MSG:INFO: Axis Z (-100.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.13 Dir:I2SO.12 Disable:I2SO.15]
[MSG:INFO:  Z Pos Limit gpio.34]
[MSG:INFO:   Motor1]
[MSG:INFO:     standard_stepper Step:I2SO.18 Dir:I2SO.17 Disable:I2SO.16]
[MSG:INFO:  Z2 Pos Limit gpio.2:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:gpio.13 Out:gpio.14 Dir:NO_PIN Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Connecting to STA SSID:Starlink]
[MSG:INFO: Connecting.]
[MSG:INFO: Connected - IP is 192.168.1.153]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
ok
$G
$Report/Interval=100
[MSG:INFO: Channel auto report interval set to 100 ms]
ok
<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:200.000,0.000,-61.300>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100>
```

### User Interface Software

Web UI

### What happened?

While doing air cuts on one job I could heard the smooth motion of the steppers when the motion was along an axis but a vibration when the motion involved 2 axes. It sounds like the diagonal motion is broken down into rapidly alternating movements along each axis.
For another test I created a tool path for a tapered hole 20mm on the top, 14.5mm on the bottom, 19.05mm deep with 6.35mm step down. I loaded a scrap of 3/4" poplar and reloaded v3.4.4 firmware/webui and ran the job. It ran smoothly and the results were round as expected. I reloaded v3.6.5-pre2 firmware/webui and reran the job. The motion was jerky and the results were not as smooth.

bdring, 2022-11-07:
Try changing `I2S_static` to `I2S_Stream`

I2S_static is only recommended for laser engraving.

<http://wiki.fluidnc.com/en/config/axes#stepping>

MitchBradley, 2022-11-07:
In order to determine if auto-reporting is implicated, can you try the following:
a) Close the WebUI window
b) Reset the ESP32
c) Connect using FluidTerm
d) Do not enable auto-reporting
e) Run the test with $sd/run=file.gcode

tgm02281961, 2022-11-07:
I was able to break away from my work long enough to make the change that Bart suggested above. I don't think I have seen my machine run better...

tgm02281961, 2022-11-07:
Is there anything else you want me to do/test before I close this issue?

MitchBradley, 2022-11-07:
That's very interesting because I2S_STATIC usually works just fine.  I am setting up a test rig similar to your config to see if I can repro the problem.

Could you try with I2S_STATIC and FluidTerm?  I need to know if there is a problem with auto-reporting.

MitchBradley, 2022-11-07:
Also keep in mind that -pre2 is not intended as a production release, but rather as a way of shaking out problems with auto-reporting.  Only when we are convinced that auto-reporting is reliable will we promote it to a real release.  Until we are sure that your problem is not triggered by auto-reporting, we are in limbo.

tgm02281961, 2022-11-07:
I had to install a UART bridge driver on my laptop before I could use fluidterm. I now have fluidterm connected to my CNC. I don't want to assume next steps...

tgm02281961, 2022-11-07:
firmware set to I2S_STATIC, restarted firmware, boot messages visible in fluidterm...

MitchBradley, 2022-11-07:
$sd/run=FILE.GC

where FILE.GC is a test file that exhibits the problem

MitchBradley, 2022-11-08:
Did you have a chance to run the test?

tgm02281961, 2022-11-08:
After your comment about the test should have been fine with Static, I decided to look closer instead of making assumptions. The feed rate for the initial test is based on a feed rate of 4064mm/min.  So, I did a series of tests, 4 using Static and 4 using Stream, with feed rates of 1016mm/min, 2032mm/min, 3048mm/min and 4064mm/min. To my old eyes there was only a slight improvement using Stream instead of Static. However, there was definite decrease in accuracy at 3048mm/min and 4064mm/min. Mechanical... So, I start shaking down the gantry and there it is. NOTICABLE slop in the spindle mount...
Add me to the list of people wasting your time. My apologies...

closed 2022-11-08 by MitchBradley