bdring/FluidNC #697: Problem: Missing steps

arlimbad222, 2022-11-09:
### Controller Board

diy

### Machine Description

4 axis mill

### Configuration file

```yaml
name: "4 AXIS MILL"
board: "4 AXIS MILL"

stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 1
  pulse_us: 4
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  
  x:
    steps_per_mm: 400
    max_rate_mm_per_min: 15000
    acceleration_mm_per_sec2: 300
    max_travel_mm: 317
    soft_limits: false
    homing:
      cycle: 2
      allow_single_axis: true
      mpos_mm: 0
      positive_direction: true
      feed_mm_per_min: 200
      seek_mm_per_min: 1500
      settle_ms: 250
      seek_scaler: 1.5
      feed_scaler: 1.5
    
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.16:low:pu
      hard_limits: false
      pulloff_mm: 2.000
      stepstick:
        direction_pin: gpio.25
        step_pin: gpio.18
    motor1:
      null_motor:

  y:
    steps_per_mm: 400
    max_rate_mm_per_min: 15000
    acceleration_mm_per_sec2: 300
    max_travel_mm: 250
    soft_limits: false
    homing:
      cycle: 2
      allow_single_axis: true
      mpos_mm: 0
      positive_direction: true
      feed_mm_per_min: 200
      seek_mm_per_min: 1500
      settle_ms: 250
      seek_scaler: 1.5
      feed_scaler: 1.5

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.17:low:pu
      hard_limits: false
      pulloff_mm: 2.000
      stepstick:
        direction_pin: gpio.15
        step_pin: gpio.26
    motor1:
      null_motor:

  z:
    steps_per_mm: 400
    max_rate_mm_per_min: 15000
    acceleration_mm_per_sec2: 300
    max_travel_mm: 140
    soft_limits: false
    homing:
      cycle: 1
      allow_single_axis: true
      mpos_mm: 0
      positive_direction: true
      feed_mm_per_min: 200
      seek_mm_per_min: 1500
      settle_ms: 250
      seek_scaler: 1.5
      feed_scaler: 1.5
      

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.4:low:pu
      hard_limits: false
      pulloff_mm: 2.000
      stepstick:
        direction_pin: gpio.33
        step_pin: gpio.27
    motor1:
      null_motor:

  a:
    steps_per_mm: 8.889
    max_rate_mm_per_min: 30000
    acceleration_mm_per_sec2: 20000
    max_travel_mm: 0

    motor0:
      limit_all_pin: NO_PIN
      stepstick:
        direction_pin: gpio.13
        step_pin: gpio.12
    motor1:
      null_motor:        

coolant:
  flood_pin: gpio.21
  mist_pin:  gpio.23

        
probe:
  pin: gpio.32:low:pu

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: gpio.5:low:pu
  cycle_start_pin: gpio.19:low:pu
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

PWM:
  pwm_hz: 5000
  output_pin: gpio.2
  enable_pin: gpio.22
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0% 18000=100%
```

### Startup Messages

```text
$SS
[MSG:INFO: FluidNC v3.6.4]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:4_AXIS_MILL.YAML]
[MSG:WARN: max_travel_mm value 0.000 constrained to range (0.100,10000000.000)]
[MSG:INFO: Machine 4 AXIS MILL]
[MSG:INFO: Board 4 AXIS MILL]
[MSG:INFO: SPI not defined]
[MSG:INFO: No SD Card CS Pin]
[MSG:INFO: See http://wiki.fluidnc.com/en/config/sd_card#sdfallbackcs-access-sd-without-a-config-file]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 4]
[MSG:INFO: Axis X (-317.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.18 Dir:gpio.25 Disable:NO_PIN]
[MSG:INFO:  X All Limit gpio.16:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (-250.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.26 Dir:gpio.15 Disable:NO_PIN]
[MSG:INFO:  Y All Limit gpio.17:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-140.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.27 Dir:gpio.33 Disable:NO_PIN]
[MSG:INFO:  Z All Limit gpio.4:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis A (-0.100,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.12 Dir:gpio.13 Disable:NO_PIN]
[MSG:INFO:   Motor1]
[MSG:INFO: feed_hold_pin gpio.5:low:pu]
[MSG:INFO: cycle_start_pin gpio.19:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:gpio.22 Out:gpio.2 Dir:NO_PIN Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Flood coolant gpio.21]
[MSG:INFO: Mist coolant gpio.23]
[MSG:INFO: Probe Pin: gpio.32:low:pu]
[MSG:INFO: BT Started with FluidNC]
```

### User Interface Software

UGS

### What happened?

Hi everyone 

I just switch fluid NC v3.4.4 to v3.6.4 and I faced some missing steps on long moves, for example, I sent (G0 X500 F13000) machine moves 490mm I test lots of times but it is not the same error every time the error was changed like some time it moves 485 or 495 or 492  I am very sure that it's not about hardware because everything works fine with v3.4.4.

secondly, I tested esp 32 with plc I have one extra plc so I just program it for counting steps, and then I got the point I switch again to v3.4.4 and test with plc everything was fine, and then again I switch to v3.6.4 and test again every time I send (G0 X500 F13000) the pulse is different as per my calculations if machine movies 500mm the pulse generation is 2,00,000 pulse but with V3.6.4 I got the 1,98,000 or 198068 or different different result.

MitchBradley, 2022-11-09:
What user interface are you using to send the commands?  A sender?  A serial terminal?  Over USB Serial or Bluetooth?

MitchBradley, 2022-11-09:
This is probably not related to your problem, but I see that you have max_travel_mm: 0  for the A axis.  0 is not a valid value for max_travel_mm .

MitchBradley, 2022-11-09:
Bart and I just spent all morning trying to reproduce the problem, using DSLogic logic analyzers with a step-counting module to count the steps.  Neither of us can make it fail.  We tried v3.6.4, a build from the current main, and my latest development code.  We tried both WiFi and Bluetooth versions, using FluidTerm to access the BT serial bridge.

At this point, to make further progress, we need a lot of information about your setup so we can discover what is different between your environment and what we have tried.  Photos of your hardware, complete info about the devices and connections through which you are sending the commands, etc.

Also, please be aware that, in "G0 X500 F13000", the "F13000" part is meaningless to the G0 command.  G0 runs at the maximum axis rate and does not use the F value.  F is only meaningful for G1, G2, G3.  When you add "Fn" to a G0 command, it will affect the next G1, G2, or G3, but it does not affect the G0 move that is on the same line. That is not causing your problem, but I though you should know.

MitchBradley, 2022-11-09:
You are probably thinking "but it works with 3.4.4 so it cannot be my hardware".  The answer to that is "yes, but even if your hardware is okay, there could be something about your setup that is interacting with a subtle difference between 3.4.4 and 3.6.4".

arlimbad222, 2022-11-10:
Hi @MitchBradley 

I am not at my workshop today so I draw one sample schematic to show how I connect esp32 with plc and that "G0" Is my typing mistake actual I send (G1 X500 F13000),

I am using the A axis as a rotary and I want that A axis infinite that's why I use 0 as maximum travel if anything else I can do please guide me.

Sender I try - UGS,  OpenBuildsCONTROL, and grbl controller app for android, I try both USB and Bluetooth
[image]

MitchBradley, 2022-11-10:
Regarding the A axis: the best thing to do is to leave max_travel_mm unspecified (do not include the line in the config file) and set soft_limits: false for that axis.  FluidNC uses max_travel_mm in only two places
1. If the axis has soft_limits: true, the value of max_travel_mm is used, along with mpos_mm and positive_direction, to calculate the axis limit values.  If soft_limits is false for an axis, the limit check code does not use max_travel_mm
2. If homing is enabled for the axis, max_travel_mm is used to determine how far the initial homing approach will go before it gives up, in case the limit switch is not found.  If homing is not enabled or performed on an axis, max_travel_mm is not used for that purpose.

Some UIs might use max_travel_mm to determine machine limits, for the purpose of displaying machine boundaries.  Most UIs have a separate way to configure machine boundaries, without asking the controller firmware.

arlimbad222, 2022-11-10:
Thankyou so much for the guide.

If anything else required for missing step issue please let me know I'll share all the information here.

I am using 6n137 optocoupler for high low signal because it's very fast optocoupler and plc only handle 24v pulse that's why optocoupler is required.

MitchBradley, 2022-11-10:
Regarding the original problem of lost steps: I am out of ideas.  We tried to reproduce the problem in many ways and never saw a failure that could be attributed to lost steps.

I did see a case where, on the very first move after a reset, the machine moved less than the expected amount.  I tracked that down to a work coordinate offset.  The G54 coordinate system had a persistent offset of X: -10.  The initial value for mpos after reset, if you do not perform a homing step, is 0.  With WCO = -10, the initial value of the work position is thus 10, so a "G1 X500 Fn" move goes from 10 to 500, thus the distance is 490.  Subsequent moves, say to "G1 X0", travel the full 500 mm.

arlimbad222, 2022-11-10:
Every time plc read different pulse so i think it's not that 10mm problem because when I switch back to 3.4.4 everything works correctly and all the reading is correct.

I ordered new esp 32 tomorrow I'll update when it arrives to my shop and i test all the things with new ESP.

arlimbad222, 2022-11-10:
hi @ModuloFs 

I am using it Just for a test here is my pcb schematic. for testing purpose 24v will works fine for me with Fluid NC v3.4.4
[image]

ModuloFS, 2022-11-10:
Maybe with version 3.4.4 you get a slightly different pulse width timing and because of this it is working. If you are using this type of optocoupler as a level shifter to 24 Volt you probably get strange outputsignalshape.

bdring, 2022-11-10:
The main difference between 3.4.4 and later releases is the use of the latest Espressif libraries. That broke a lot of things. Until this issue we have been successful in reproducing user issues as they pop up.

One issue was UART communications interfered with the RMT after 3.4.4. It was causing lost steps. It is possible another feature of the chip is still causing issues. Until we can reproduce it, it is hard to know.

MitchBradley, 2022-11-10:
After 3.4.4 we switched to a newer version of the ESP-IDF SDK and the corresponding new version of the Arduino Framework that sits on top of it.  The goal of that switch was threefold.  One was to avoid being stuck on an old version that would become unsupported - and indeed the first trigger was when someone complained that our build scripts did not work when PlatformIO auto-updated to the new Arduino Framework.  Secondly, we wanted to migrate from the SPIFFS FLASH filesystem which is flaky to the newer and better LittleFS filesystem; the old SDK does not support LittleFS.  Thirdly, we want to support some of the newer ESP32 variants like ESP32-S2, and the new SDK is necessary for that.

The switchover did not go smoothly at all.  There were bugs in the new Arduino framework, changes in default behavior of drivers, compiler version changes that obsoleted things like atomic access primitives, API changes in core drivers, WiFi authentication changes, and more.  Versions 3.4.5 - 3.5.x were riddled with hard to diagnose problems.  Things started to get better with 3.6.1, but we are still finding API differences that cause subtle problems.

Meanwhile we have been overwhelmed with support requests from people wanting completely new features, people with hardware design flaws claiming the that firmware is bad, people who fundamentally misunderstand how GCode works, people who ask questions about Chinese knockoffs that violate the GPL license terms, incomplete issues where we have to beg and beg to get the information we need to evaluate the problem, and many more distractions.  A small fraction of those people donate to the project, so we are, for the most part, doing all this development and support work for free.

There have been 275 commits since 3.4.4 so the "main difference" is hard to identify.  Even if I did pick something in particular, the chances of that being the right guess are about zero.

Please respond to the question about Android and the FluidTerm suggestion.  We need a specific setup that we can reproduce.

arlimbad222, 2022-11-11:
Hi@MitchBradley 

I am using Android app and bluetooth communication with fluid nc.

I am not good with coding can you tell me how you are counting pulse, can I do it with Arduino like program one Arduino for counting pulse and read it with serial monitor if it possible can you help me with Arduino code. so i avoid the plc and that optocoupler module and directly count 3.3v pulse with Arduino.

MitchBradley, 2022-11-11:
I am using a dslogic logic analyzer with a stepper motor decoding module. I will write a pulse counting sketch for ESP32 tomorrow. It is late here

arlimbad222, 2022-11-11:
Hi @MitchBradley 

thank you for your help, post the code here whenever you write it. I will use second esp 32 for counting the pulse and update here.

MitchBradley, 2022-11-11:
https://github.com/MitchBradley/PulseCounter is a pulse counter for ESP32

arlimbad222, 2022-11-12:
Hi@MitchBradley 

Thanks for the efforts I'll taste it tomorrow and update here I am waiting for my new esp 32 I am pretty sure it will delever tomorrow.

MitchBradley, 2022-11-12:
I found a difference between 3.6.4 and 3.4.4 - 3.6.4 is faster.  With 3.6.4, when stepping at the cruising speed of 100K steps/sec, some pulses are exactly 10 us apart, some are a little longer, and some a little shorter, maintaining an average speed of 100K/sec.  The longest interval is 12 us and the shortest is 8.

[image]

With 3.4.4, when nominally cruising at 100K/sec, the average speed is closer to 80K/sec, ranging from 12 us to 14 us between pulses.

[image]

In both cases, the logic analyzer shows that all 200K pulses were issued.  In 3.4.4, a `g0 x 500` move took 3.5 sec, vs 2.84 sec for the same move in 3.6.4.

The config file was the same in both cases, with 4us pulse, 400 steps/mm, 15000 mm/min max rate (= 250 mm/sec), 300 mm/sec2 acceleration.    The 2.9 sec time is correct - 0.833 seconds to accelerate to to 250 mm/sec at 300 mm/sec2, plus another 0.83 sec to decelerate.  The distance traveled during accel is 104 mm, so 208 mm for accel + decel.  That leaves 292 mm at cruising speed, taking 1.17 sec.  So a total of 2.84 sec  

At feedrate 13000 (`g1 x500 f13000`), 3.6.4 remains faster than 3.4.4, taking 3.1 sec for 500 mm.

With 3.6.4, you have drop the F value to 12000 to make it behave like 3.4.4 at cruising speed.

My guess is that your external hardware - either external drivers or the PLC - can barely keep up with 100K/sec stepping, especially when 100K is the average rate with some pulses a little closer together and some a little farther apart.

It is unfortunate that, in the initial report, the only thing you told us about your external hardware was just "diy" and 'plc".  That is not nearly enough information to determine how fast such hardware is likely to be.

arlimbad222, 2022-11-15:
Hi @MitchBradley 

I tested with 2 devices tomorrow the first one is with "esp32" and the second is a high-speed Servo drive,

With esp32 - G01 x500 f15000 , pulse - 200000
                      G01 x0 f15000    , pulse - 199998

With servo drive - G01 X500 F15000 , pulse - 200000
                              G01 x0 f15000 ,      pulse - 200000

I found that F14000 is creating some mess it loses 1 or 2 steps otherwise everything works well. 

Finally, I found that it was my plc fault maybe the plc input could not handle that much fast pulse rate, thank you for your time and he
[image]
lp, and also thank you for that pulse counter code.
![16779729-aa84-452f-b848-3af48790d8d4](https://user
[image]
-images.githubusercontent.com/34487708/201821599-0e2481f1-4427-40cd-bcb8-4472ca73c857.jpg)

closed 2022-11-15 by MitchBradley