bdring/FluidNC #701: Problem: fluidnc-v3.6.5-pre2 freezes

ipsod, 2022-11-15:
### Controller Board

Makerbase MKS TinyBee V1.0

### Machine Description

  no hardware hooked up.  just a board with no drivers

### Input Circuits

```yaml
n/a
```

### Configuration file

```yaml
name: "ESP32 Dev Controller V4"
board: "ESP32 Dev Controller V4"

stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 1
  pulse_us: 2
  disable_delay_us: 0

axes:
  # shared_stepper_disable_pin: gpio.13:low

  x:
    steps_per_mm: 3200
    max_rate_mm_per_min: 60
    acceleration_mm_per_sec2: 120
    max_travel_mm: 100
    motor0:
      # limit_neg_pin: gpio.17:low:pu
      stepstick:
        step_pin: gpio.12
        direction_pin: gpio.13
        disable_pin: gpio.5
    motor1:
      null_motor:

  y:
    steps_per_mm: 800
    max_rate_mm_per_min: 2000.0
    acceleration_mm_per_sec2: 1000.0
    max_travel_mm: 1000
    homing:
      cycle: 2
      mpos_mm: 10
      positive_direction: false

    motor0:
      stepstick:
        step_pin: gpio.27
        direction_pin: gpio.14
        disable_pin: gpio.16
    motor1:
      null_motor:

  z:
    steps_per_mm: 800
    max_rate_mm_per_min: 200.0
    acceleration_mm_per_sec2: 200.0
    max_travel_mm: 1000
    homing:
      cycle: 1
      mpos_mm: 10
      positive_direction: true

    motor0:
      stepstick:
        step_pin: gpio.19
        direction_pin: gpio.23
        disable_pin: gpio.18
    motor1:
      null_motor:
```

### Startup Messages

```text
$Bye  
[MSG:INFO: Restarting]
ok
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12812
load:0x40080400,len:3032
entry 0x400805e4

[MSG:INFO: FluidNC v3.6.5-pre2]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine ESP32 Dev Controller V4]
[MSG:INFO: Board ESP32 Dev Controller V4]
[MSG:INFO: SPI not defined]
[MSG:INFO: No SD Card CS Pin]
[MSG:INFO: See http://wiki.fluidnc.com/en/config/sd_card#sdfallbackcs-access-sd-without-a-
config-file]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-100.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.12 Dir:gpio.13 Disable:gpio.5]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (10.000,1010.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.27 Dir:gpio.14 Disable:gpio.16]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-990.000,10.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.19 Dir:gpio.23 Disable:gpio.18]
[MSG:INFO:   Motor1]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.6 [FluidNC v3.6.5-pre2 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
```

### User Interface Software

FluidTerm

### What happened?

input and output freeze after a pause
To recreate, open FluidTerm, and send

```
$ri=50
$j=g91y0.00167f1000
$j=g91y0.00333f1000
$j=g91y0.005f1000
$j=g91y0.00667f1000
$j=g91y0.00833f1000
$j=g91y0.01f1000
```

Then, wait some amount of time and re-send the jog commands

```
$j=g91y0.00167f1000
$j=g91y0.00333f1000
$j=g91y0.005f1000
$j=g91y0.00667f1000
$j=g91y0.00833f1000
$j=g91y0.01f1000
```

If it works, wait 5 minutes and try again.  Eventually, it will freeze.  At least, that's what I'm experiencing.

MitchBradley, 2022-11-15:
I will look into it.  Thanks for finding a simple test case.

MitchBradley, 2022-11-15:
Are you using the Windows native version of FluidTerm, or the Python version?

MitchBradley, 2022-11-15:
Okay, so the Python version.  And what OS are your running?

MitchBradley, 2022-11-15:
I am trying to determine if the problem is in the FluidNC end or the host end.  The fact that your two issues both have host Python in command makes it plausible that the hang could be on either end.

ipsod, 2022-11-15:
Linux Mint 20.2, running on a Thinkpad T530.

The one thing that makes me think it's on the FLuidNC end, is that I can disconnect, reconnect, and it is still frozen.  

I added info to my post just above, too.

ipsod, 2022-11-15:
I had thought that `ctrl+r` always worked, but it's frozen now and that isn't working.  So, jury's still out, definitely could be Python/USB, especially if you've tried more than a few times on Windows and haven't encountered this.  It is quite frequent, for me.

MitchBradley, 2022-11-15:
I haven't tried at all, yet.  I am still catching up with the Discord traffic.

If ctrl-r says "Resetting MCU" I am inclined to suspect some clog in the USB driver, but that is just a guess.

MitchBradley, 2022-11-15:
This is, again, not a diagnosis but rather an observation.  I remember a situation from several years ago whereby a USB serial driver - I think it was the CH340 one - tended to crash MacOS.  It might have had a memory leak.  Tinybee uses CH340.

MitchBradley, 2022-11-15:
How did you send those jog commands?  Did you type them one by one, or did you paste the entire batch into FluidTerm in one go, or what?

ipsod, 2022-11-15:
Pasted the entire batch.  Character count is 114, which, as I understand, is lower than GRBL's serial buffer limit.  I've been treating the serial buffer character limit as 120, though I think the docs say 125 or 126.

MitchBradley, 2022-11-15:
Do you have echoing on?  So that FluidNC echoes the characters as they are typed?

I tried it both ways.

ipsod, 2022-11-15:
I had it fail both ways.  My sender sends the `echo off` command at startup, but sometimes I triggered echo in a FluidTerm session so I could see output.

After restarting my computer, this issue seems to have went away.  Hopefully for good.

closed 2022-11-15 by ipsod

MitchBradley, 2022-11-15:
Best guess: Another process had the port open and was stealing the serial data sometimes.  Linux does not enforce exclusive open on serial ports.  That hypothesis explains all the symptoms, and is plausible considering that you were trying various ways of sending.

ipsod, 2022-11-15:
I had lots of python processes holding usb connections die, while working on my sender.  Wouldn't be surprised if one of their ghosts was haunting me.  I thought I'd checked for that, by killing all "python" processes I could find.

Thankfully, it's still running fine.