bdring/FluidNC #702: Negative homing bails out before max travel is reached

arhuaco, 2022-11-15:
### Controller Board

6 Pack External Driver CNC Controller

### Machine Description

3 axis Chinese 3040 machine.

### Configuration file

```yaml
board: 6 Pack
name: 6 Pack External XYYZ No Square - v2
meta: Based on 2022-09-19 B. Dring for Dean Embrey

stepping:
  engine: I2S_STREAM
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 500
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 50
    max_travel_mm: 285
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      allow_single_axis: true
      mpos_mm: 0
      seek_mm_per_min: 800

    motor0:
      limit_all_pin: gpio.34:low
      hard_limits: true
      pulloff_mm: 2
      standard_stepper:
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0

  y:
    steps_per_mm: 500
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 50
    max_travel_mm: 382
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      allow_single_axis: true
      mpos_mm: 382
      seek_mm_per_min: 1000
    motor0:
      limit_all_pin: gpio.35:low
      hard_limits: true
      pulloff_mm: 2
      standard_stepper:
        step_pin: I2SO.5
        direction_pin: I2SO.4:low
        disable_pin: I2SO.7

  z:
    steps_per_mm: 400
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 50
    max_travel_mm: 48
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      allow_single_axis: true
      mpos_mm: 48
      seek_mm_per_min: 400

    motor0:
      limit_all_pin: gpio.36:low
      hard_limits: true
      pulloff_mm: 2
      standard_stepper:
        step_pin: I2SO.10
        direction_pin: I2SO.9:low
        disable_pin: I2SO.8

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

#sdcard:
#  card_detect_pin: NO_PIN
#  cs_pin: gpio.5

start:
  must_home: false
  deactivate_parking: false
  check_limits: false

arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: true
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
planner_blocks: 16

NoSpindle:

coolant:
  mist_pin: i2so.3
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.6.4]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine 6 Pack External XYYZ No Square - v2]
[MSG:INFO: Board 6 Pack]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: No SD Card CS Pin]
[MSG:INFO: See http://wiki.fluidnc.com/en/config/sd_card#sdfallbackcs-access-sd-without-a-config-file]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,285.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.2 Dir:I2SO.1 Disable:I2SO.0]
[MSG:INFO:  X All Limit gpio.34:low]
[MSG:INFO: Axis Y (0.000,382.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.5 Dir:I2SO.4:low Disable:I2SO.7]
[MSG:INFO:  Y All Limit gpio.35:low]
[MSG:INFO: Axis Z (0.000,48.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.10 Dir:I2SO.9:low Disable:I2SO.8]
[MSG:INFO:  Z All Limit gpio.36:low]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Mist coolant I2SO.3]
[MSG:INFO: Connecting to STA SSID:3_Primo]
[MSG:INFO: Connecting.]
[MSG:INFO: Connected - IP is 192.168.1.34]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
```

### User Interface Software

WebUI

### What happened?

H there!

Using latest release v3.6.4.

After clearing up switches, X is set to 280 and max travel is 285. But  it bails when -250 is reached (traveling 250 mm).

Unless I am not getting something, it should bail out after 285.

$h
  <Home|MPos:0.000,0.000,4.293|FS:400,0>
  <Home|MPos:0.000,0.000,24.255|FS:400,0>
  <Home|MPos:0.000,0.000,44.255|FS:400,0>
  <Home|MPos:0.000,0.000,48.472|FS:50,0>
  <Home|MPos:0.000,0.000,49.865|FS:50,0>
  <Home|MPos:-0.002,0.002,48.000|FS:419,0>
  <Home|MPos:-30.310,50.784,48.000|FS:1281,0|WCO:0.000,0.000,0.000>
  <Home|MPos:-63.106,105.730,48.000|FS:1281,0|Ov:100,100,100>
  <Home|MPos:-95.872,160.628,48.000|FS:1281,0>
  <Home|MPos:-128.676,215.588,48.000|FS:1281,0>
  <Home|MPos:-161.488,270.564,48.000|FS:1281,0>
  <Home|MPos:-194.288,325.516,48.000|FS:1281,0>
  <Home|MPos:-227.088,380.472,48.000|FS:1281,0>
  <Home|MPos:-250.800,384.170,48.000|FS:0,0|Pn:Y>
  ALARM:9
  Homing fail. Could not find limit switch within search distances. Try increasing max travel, decreasing pull-off distance, or check wiring.
  ok

arhuaco, 2022-11-15:
Oh, I think the problem might be with Y?!

max_travel is 382 and it's at 384.170. Before the test I had taken Y to 0.

Wondering if I should make max_travel_mm = 386 and mpos_mm = 382.

But also wondering if that means that I can take Y below 0. I have a limit switch there anyway.

I will try, but any hint about the best practice here is more than welcome.

bdring, 2022-11-15:
$h should be homing z first.

Try homing axes independently with $HX then $HY and $HZ.

It also looks like Y was triggered in the last status message `<Home|MPos:-250.800,384.170,48.000|FS:0,0|Pn:Y>`. Was it touching the switch?

arhuaco, 2022-11-15:
Z is homed OK before X and Y start.

Y It was touching the switch.

Will test Y on its own.

bdring, 2022-11-15:
Double check the switches are reporting the correct axes.

http://wiki.fluidnc.com/en/support/help_with_switch_problems

arhuaco, 2022-11-15:
Homing works OK but not when Y is the furthest. Will debug. When I did the report I thought the issue was X but reading again it seems it's Y.

But homing works if Y is not that far.

MitchBradley, 2022-11-15:
Try homing x and y individually with $hx and $hy.  Also set $message/level=debug .

bdring, 2022-11-15:
Is it possible the Y steps per mm is wrong? Check the move distance with a jog. This should produce a 10mm positive move.

`$J=G91 G21 Y10 F500`

arhuaco, 2022-11-15:
Hi there! It's calibrated. I move Y 380mm and it's end to end in the 3040 machine. X (280) and Y (50) are OK to.

Y worked OK after this change.

`+++ b/3040/FluidNC/config.yaml
@@ -37,7 +37,7 @@ axes:
     steps_per_mm: 500
     max_rate_mm_per_min: 5000
     acceleration_mm_per_sec2: 50
-    max_travel_mm: 382
+    max_travel_mm: 387
     soft_limits: false
     homing:
       cycle: 2
len@shift:~/git/`

But then when  I home the 3 axis it stops with the message in the report.

I hope to debug more later. A bit puzzled.

bdring, 2022-11-15:
Do what @MitchBradley said.

MitchBradley, 2022-11-15:
With $message/level=debug, homing will display a lot of info about its target positions and whatnot.

arhuaco, 2022-11-15:
Thanks @MitchBradley . Just tested.

Here's a homing log. It works for $hx, for $hy, but not for $h. I hope it helps!

I moved X and Y far from HOME with g1 y0 x280 f3000 before testing $h.

[homing-session.txt](https://github.com/bdring/FluidNC/files/10016605/homing-session.txt)

arhuaco, 2022-11-15:
Only X.

    [MSG:DBG: Starting from 0.000,0.000,0.000]
    [MSG:DBG: Planned move to -313.500,0.000,0.000 @ 800.000]

XYZ.

    MSG:DBG: Starting from 0.000,0.000,48.000]
    [MSG:DBG: Planned move to -250.800,425.700,48.000 @ 1280.625]

Interesting that planned 250 is consistent with what I saw in the report I sent above. Nice to have those messages.
This is less than the max_travel_mm: 285.

MitchBradley, 2022-11-15:
250 is wrong.  I am trying to figure out why it picked that value.

MitchBradley, 2022-11-15:
I think the problem is in the code that computes the target based on maxx, maxy, and the rates for x and y.  I'll bet that it will work if you set the max rates for x and y the same.

MitchBradley, 2022-11-15:
Yes that code is wrong.  I have to do some head scratching.  When you are homing two axes simultaneously at different distances and rates, you have to find the axis that, at its seek rate, will take the longest time to reach its limit, and scale everything accordingly.  Seems like I got that computation wrong.

arhuaco, 2022-11-15:
That works! Thanks.

Here's a log. Interesting that planning info looks different if I restart from UI or power cycle (I guess because it does a reset or something like that).

[homing-session-xy-same-seek.txt](https://github.com/bdring/FluidNC/files/10016786/homing-session-xy-same-seek.txt)

arhuaco, 2022-11-21:
Hello there.

With v3.6.5-pre3 X homing direction is reversed. It started homing to the right. And before it was homing to the left. Y and Z are homing OK, as before.

Note that only X searches for a 0 when homing, both Y and Z go to the maximum value that dimension can reach.

Startup messages and config file follow.

[MSG:INFO: FluidNC v3.6.5-pre3]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:DBG: Running after-parse tasks]
[MSG:DBG: Checking configuration]
[MSG:INFO: Machine 6 Pack External XYYZ No Square - v2]
[MSG:INFO: Board 6 Pack]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: No SD Card CS Pin]
[MSG:INFO: See http://wiki.fluidnc.com/en/config/sd_card#sdfallbackcs-access-sd-without-a-config-file]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,285.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.2 Dir:I2SO.1 Disable:I2SO.0]
[MSG:INFO:  X All Limit gpio.34:low]
[MSG:DBG:  X All Limit 0]
[MSG:INFO: Axis Y (-5.000,382.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.5 Dir:I2SO.4:low Disable:I2SO.7]
[MSG:INFO:  Y All Limit gpio.35:low]
[MSG:DBG:  Y All Limit 0]
[MSG:INFO: Axis Z (0.000,48.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.10 Dir:I2SO.9:low Disable:I2SO.8]
[MSG:INFO:  Z All Limit gpio.36:low]
[MSG:DBG:  Z All Limit 0]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:NO_PIN Out:gpio.25 Dir:NO_PIN Freq:2000Hz Period:32767]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Mist coolant I2SO.3]
[MSG:INFO: Connecting to STA SSID:3_Primo]
[MSG:INFO: Connecting.]
[MSG:INFO: Connected - IP is 192.168.1.34]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

[config.txt](https://github.com/bdring/FluidNC/files/10060088/config.txt)

bdring, 2022-11-21:
Show the homing debug messages with the new release

arhuaco, 2022-11-21:
Hi there.

I didn't let it go to the end of X but you can see the planner messages.

Here I have 500 as seek speed for both axis. Attaching config anyway, the only difference with the previous one is that that one doesn't have the same seek rate for X and Y, but I see reversed X homing in both.
[config.txt](https://github.com/bdring/FluidNC/files/10060357/config.txt)

IMHO this line suggests that the limit for Y is affecting the limit for X at homing time.

[MSG:DBG: Planned move to 425.700,425.700,48.000 @ 707.107]

$hx
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100>
[MSG:DBG: Homing Cycle X]
[MSG:DBG: Homing nextPhase FastApproach]
[MSG:DBG: Starting from 0.000,0.000,0.000]
[MSG:DBG: Planned move to 313.500,0.000,0.000 @ 500.000]
<Home|MPos:17.302,0.000,0.000|FS:500,0>

$h
[MSG:DBG: Homing Cycle Z]
[MSG:DBG: Homing nextPhase FastApproach]
[MSG:DBG: Starting from 0.000,0.000,0.000]
[MSG:DBG: Planned move to 0.000,0.000,52.800 @ 400.000]
[MSG:DBG:  Z All Limit 1]
[MSG:DBG: Homing limited Z]
[MSG:DBG: Homing nextPhase Pulloff0]
[MSG:DBG: Starting from 0.000,0.000,2.053]
[MSG:DBG: Planned move to 0.000,0.000,0.053 @ 50.000]
[MSG:DBG:  Z All Limit 0]
<Home|MPos:0.000,0.000,1.150|FS:50,0>
[MSG:DBG: CycleStop Pulloff0]
[MSG:DBG: Homing nextPhase SlowApproach]
[MSG:DBG: Starting from 0.000,0.000,0.052]
[MSG:DBG: Planned move to 0.000,0.000,2.253 @ 50.000]
<Home|MPos:0.000,0.000,1.220|FS:50,0>
[MSG:DBG:  Z All Limit 1]
[MSG:DBG: Homing limited Z]
[MSG:DBG: Homing nextPhase Pulloff1]
[MSG:DBG: Starting from 0.000,0.000,1.985]
[MSG:DBG: Planned move to 0.000,0.000,-0.015 @ 50.000]
[MSG:DBG:  Z All Limit 0]
<Home|MPos:0.000,0.000,0.477|FS:50,0>
[MSG:DBG: CycleStop Pulloff1]
[MSG:DBG: Homing nextPhase Pulloff2]
[MSG:DBG: mpos was 0.000,0.000,-0.015]
[MSG:DBG: mpos becomes 0.000,0.000,48.000]
[MSG:DBG: mpos transformed 0.000,0.000,48.000]
[MSG:DBG: Homing Cycle XY]
[MSG:DBG: Homing nextPhase FastApproach]
[MSG:DBG: Starting from 0.000,0.000,48.000]
[MSG:DBG: Planned move to 425.700,425.700,48.000 @ 707.107]
[MSG:DBG:  Y All Limit 1]
[MSG:DBG: Homing limited Y]
<Home|MPos:17.078,17.076,48.000|FS:707,0|Pn:Y>

(I turned it off here, Y was indeed touching the switch and X was traveling right and I expected it to be traveling left).

arhuaco, 2022-11-22:
Homing worked with v3.6.5-pre4. Thanks!

[log-home.txt](https://github.com/bdring/FluidNC/files/10071921/log-home.txt)

arhuaco, 2022-11-23:
Hi, I'm sorry I reported it worked but now I tried and it didn't work.

It's stopping early and Y is reacting to X switch and I didn't notice because of how I ran the  test.

So, when the X switch is reached, Y start bouncing and acts as if the X switch is its own and FluidNC thinks homing its OK when it's not.

[homing.txt](https://github.com/bdring/FluidNC/files/10077380/homing.txt)
[config.txt](https://github.com/bdring/FluidNC/files/10077382/config.txt)

arhuaco, 2022-11-23:
Doing more tests I figured out that what I just wrote only happens when I'm using USB cable. Otherwise things work well.

I have an issue at home, old building without ground line. So maybe this is bringing some noise. Weird that it didn't happen before.

So, via Telnet it works well.

MitchBradley, 2022-11-23:
Possibly a ground loop through the usb cable and the computer

MitchBradley, 2022-11-23:
Could be a safety issue if there is significant potential between the controller and PC grounds.

arhuaco, 2022-11-23:
I won't use it anymore with the computer.  I will prepare a laptop I have around later. For now I'll use UGS+Telnet and finish the setup. I think FluidNC firmware is finally configured :-)

MitchBradley, 2022-11-23:
Can we close the issue now?

arhuaco, 2022-11-23:
Yes! Thanks again.

commit ccf514a3 (2022-11-20): Fix #702 - incorrect homing target with different axis rates
commit 65cab18e (2022-11-22): Fix #702 - homing direction bug introduced by #703
commit e5913710 (2022-11-22): Merge pull request #715 from bdring/ReFix702

Fix #702 - homing direction bug introduced by #703
commit 821198a4 (2022-11-27): Fix #702 - homing direction bug introduced by #703
closed 2022-11-23 by MitchBradley