bdring/FluidNC #705: gcode clustering

distebia, 2022-11-17:
### Machine Context

mks dlc32

### Feature Description

autocostruita

### Other Approaches

.

### How I Can Help

Salve, utilizzo lightburn e una mks dlc32 con firmware FluidNC. Controllando le impostazioni macchina di LB mi accorgo che c'è un'opzione che consente una maggiore velocità per le incisioni a tono di grigio, Leggo poi che bisogna avere un firmware supportato. Non so se è una particolarità dell'hardware o del firmware e mi chiedo se fluidnc ha questa funzionalità.

MitchBradley, 2022-11-17:
No, FluidNC does not have specific support for grayscale engravings.

distebia, 2022-11-18:
ma pensate di implementare questa possibilità? leggo in giro che grblhal e lasergrbl hanno aggiunto questo. Sinceramernte mi trovo bene con FluidNC e non vorrei comprare un ruida solo per questo

distebia, 2022-11-18:
allora attendo le cose che sono più interessanti e, magari, anche, prima o poi, il grbl clustering

distebia, 2023-01-14:
scusate se ritorno sul clustering gcode ma, leggendo la wiki, ho notato la voce PLANNER_BLOCKS e da 16 l'ho portata a 120.
Un'altra voce che ho notato sotto Stepping è SEGMENTS e l'ho settato a 20. ho provato a fare qualche foto in scala di grigio e ho notato un leggero miglioramento nella velocità di scansione. esiste qualche altra opzione che mi aiuta in questo?

MitchBradley, 2023-01-14:
No

tatarize, 2023-11-01:
It wouldn't just allow grayscale encoding speed you could add in:
`$I`
`[CLUSTER:16]`

And still only accept `S0:1:1:1:1:0:0:0:0:1:1:0:1:0:0:0:1` (assuming an S value between 0-1.

Basically when you make a move with that set as the power, it goes through each of those values between during that movement with a lerp. It doesn't need to be specifically grayscale specific. 

It might need to check lightburns implementation to see if they do clustering mode with grayscale off but they likely should.

closed 2024-12-25 by distebia