bdring/FluidNC #709: Problem: El servidor local no carga

Kenedy-Hanes, 2022-11-20:
### Controller Board

Esp32 -WROOM-32

### Machine Description

La máquina que he ensmblado tiene una dimension de 3m X 2.5m, esta diseñado para trabajar en campo libre.

### Configuration file

```yaml
name: "ESP32"
board: "Wemos_CNCshield"

stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 1
  pulse_us: 2
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN

  x:
    steps_per_mm: 40.650
    max_rate_mm_per_min: 350
    acceleration_mm_per_sec2: 20
    max_travel_mm: 100
    homing:
      cycle: 2
      mpos_mm: -3.000
      feed_mm_per_min: 350
      positive_direction: true
    
    motor0:
      limit_all_pin: gpio.36:low
      hard_limits: false
      pulloff_mm: 3.000
      stepstick:
        direction_pin: gpio.16
        step_pin: gpio.26

  y:
    steps_per_mm: 40.650
    max_rate_mm_per_min: 350
    acceleration_mm_per_sec2: 20
    max_travel_mm: 100
    homing:
      cycle: 2
      mpos_mm: -3
      feed_mm_per_min: 350
      positive_direction: true
    
    motor0:
      limit_all_pin: gpio.34:low
      hard_limits: false
      pulloff_mm: 3
      stepstick:
        direction_pin: gpio.27
        step_pin: gpio.25

    motor1:
      limit_all_pin: gpio.39:low
      hard_limits: false
      pulloff_mm: 3
      stepstick:
        direction_pin: gpio.32
        step_pin: gpio.33

  z:
    steps_per_mm: 40.650
    max_rate_mm_per_min: 350
    acceleration_mm_per_sec2: 20
    max_travel_mm: 10
    homing:
      cycle: 1
      mpos_mm: 3
      feed_mm_per_min: 350
      positive_direction: true
    
    motor0:
      limit_all_pin: gpio.35:low
      hard_limits: false
      pulloff_mm: 3
      stepstick:
        direction_pin: gpio.14
        step_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

coolant:
  flood_pin: gpio.15

probe:
  pin: gpio.13:low:pu

user_outputs:
  digital0_pin: gpio.12

relay:
  direction_pin: NO_PIN
  output_pin: gpio.2: low
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0% 0=100% 1=100%
  off_on_alarm: false
```

### Startup Messages

```text
Start/Message=Grbl \V [FluidNC \B (\R) \H]
```

### User Interface Software

WebUI

### What happened?

Mi máquina está diseñada para trabajar en campo libre sin cobertura, estoy probando FluidNC ya casi durante 2 mese, el problema que presenta es que de noche mi maquina trabaja perfectamente pero de día tengo dificultades para poder controlar la maquina porque la interfaz WebUI no carga, tengo la sospecha de que la radiacion solar tiene algo que ver, probe colocando el router lo mas cerca posible pero sin resultados, la placa se conecta correctamente a la red pero WebUI no responde.

La razon por la que la maquina debe estar en campo libre es porque realiza tallado en rocas.
 Necesito ayuda ya no se que mas hacer.

Les agradesco y Felicito por compartir su magnifico trabajo FluidNC.

MitchBradley, 2022-11-20:
Maybe something is overheating

bdring, 2024-12-12:
Closing inactive issue

closed 2024-12-12 by bdring