bdring/FluidNC #718: Problem:

cranefist, 2022-11-23:
### Controller Board

6-packdasd

### Machine Description

ada

### Configuration file

```yaml
add
```

### Startup Messages

```text
daa
```

### What happened?

eerro

bdring, 2022-11-23:
Try probing with FluidTerm and paste everything here. It helps to see everything without the sender filtering anything.

Check status before testing so we see the initial state.

Also provide the startup messages.

cranefist, 2022-11-23:
[image]

```
G91;
G21;
F300 Y10;
G90;
F100 Z-2;
G91;
G38.2 Y-20 F50;
G10 L20 P0 Y1.725;
G21 F300 Y1;
G21 F300 Z12;
```

bdring, 2022-11-23:
Please use cut and paste rather than screenshots.

I will wait for the other info to comment. Headed out of ski now.

cranefist, 2022-11-23:
I have no idea how to cut & paste on phones, i suck at phones. And this wifi is not connected to my PC:s wifi.

Have fun skiing, i'm not in a hurry and probably eventually will figure this out. But its clear that the error messages are somehow wrong. It cant fail at probing after probing.

cranefist, 2022-11-23:
I now have 3 separate working probing macros for ZYX, and i realized that for order them to work.. i always need to hit the reset button between them. If i do not do that, the probing fails. 

And it always gives the error as a probing fail, or probing related error.

Even if it fails before one movement that happens before the actual probing happens.

So my problem is somehow related to some state that is left in a different position that is the returned to normal by resetting.

And the wrong error message is just weird outcome of this.

bdring, 2022-11-23:
Here is how I sort that out.

Manually turn one motor at a time with the other held firmly until the the end effector moves in positive X and Y (45 deg). Remember which motor you turned and which direction you turn it.

No send small jogs like this `$J=G91 G21 X5 Y5 F200`

You want the motors to do the same thing you did manually. 
- If the wrong motor moves, swap the motors
- If the correct motor moves, but the wrong way then change the direction attribute (:high vs. :low) of that motor.

Restart and test again.

If that jog works, then try jogging other directions. If that is not right, the only thing that could be wrong is the other motor's direction.

MitchBradley, 2022-11-23:
Please show us all of your macros and tell us about one specific sequence that fails, in great detail.  Include information about whether the probe is still touching at various parts of the sequence.

bdring, 2022-11-23:
Without a decent log of what happens, I can't help much until I have access to a machine. That will be in about 10 days.

Maybe someone else who has time and a similar machine can help.

cranefist, 2022-11-23:
What i would be most interested, is what does the soft reset reset? That seems to be the key.

So here are my 3 macros now:

Z:
```
G38.2 Z-100 F50;
G10 L20 P0 Z1.725;
$J=G91 G21 F300 Z10;

```

Y:
```
G91;
G21;
F300 Y10;
G90;
F100 Z-2;
G91;
G38.2 Y-20 F50;
G10 L20 P0 Y1.725;
$J=G21 F300 Y1;
$J=G21 F300 Z12;
```

X:
```
G91;
G21;
G90 F300 Y-5;
G91 G21 F300 X10;
G90 F100 Z-2;
G91;
G38.2 X-20 F50;
G10 L20 P0 X1.725;
$J=G21 F300 X1;
$J=G21 F300 Z12;

```

The Z macro i can do repeatedly, it does not cause errors.

But when i do the Y after i have done the Z. It fails, unless it do a soft reset between.

This is how it fails:
```
$LocalFS/Run=/probey.g
<Idle|MPos:101.817,65.893,41.632|FS:0,0|SD:0.00,/spiffs/probey.g>
<Run|MPos:101.817,66.760,41.632|FS:300,0|SD:19.83,/spiffs/probey.g>
ALARM:5
Probe fail. Probe did not contact the workpiece within the programmed travel for G38.2 and G38.4.
[PRB:101.817,65.893,31.750:0]
[MSG: ERR:9 (GCode cannot be executed in lock or alarm state) in /spiffs/probey.g at line 4]
```

It claims to fail at probing, but it never does the `F100 Z-2;` movement.

MitchBradley, 2022-11-23:
> What i would be most interested, is what does the soft reset reset? That seems to be the key.

About a million things.

MitchBradley, 2022-11-23:
This issue is incomplete.  It lacks the machine and other information that the issue template asks for.  We need all that information to be able to help you efficiently.

cranefist, 2022-11-23:
Also, why would i need to have these:
```
$J=G21 F300 Y1;
$J=G21 F300 Z12;
```

Instead of these:
```
G21 F300 Y1;
G21 F300 Z12;
```

What is the actual difference here?

Without the $J, i get a fail. And again it claims it was a probing error, even though the probing was done and it was successful.
```
$LocalFS/Run=/probey.g
<Idle|MPos:101.817,62.893,41.580|FS:0,0|Ov:100,100,100|SD:0.00,/spiffs/probey.g>
<Run|MPos:101.817,66.688,41.580|FS:300,0|SD:57.27,/spiffs/probey.g>
<Run|MPos:101.817,72.893,34.680|FS:300,0|SD:57.27,/spiffs/probey.g>
<Run|MPos:101.817,71.775,27.855|FS:50,0|SD:57.27,/spiffs/probey.g>
[PRB:101.817,71.738,27.855:1]
ALARM:4
Probe fail. Probe is not in the expected initial state before starting probe cycle.
[MSG: ERR:9 (GCode cannot be executed in.... problem at line 10...

```

cranefist, 2022-11-23:
Without a soft reset, X fails like this
```
$LocalFS/Run=/probex.g
<Idle|MPos:101.817,72.380,39.875|FS:0,0>
<Idle|MPos:101.817,72.380,39.875|FS:0,0|SD:0.00,/spiffs/probex.g>
ALARM:5
Probe fail. Probe did not contact the workpiece within the programmed travel for G38.2 and G38.4.
[PRB:101.817,71.503,27.875:0]
[MSG: ERR:9 (GCode cannot be executed in lock or alarm state) in /spiffs/probex.g at line 4]
```

Same thing as on Y,  fails at G90 movement and claims it was probe error.

So it clearly somehow related to these coordinate systems.

MitchBradley, 2022-11-23:
Aha!
```
G91
G21
F300 Y10
```
There is no G1 there.  Lacking an explicit G0 or G1, the current mode is still G38 from the previous probe cycle, so the F300 Y10 is treated as a probe, not a move.

$J does not inherit the current mode, which is why it works.

MitchBradley, 2022-11-23:
Note that G90 and G91 are not "movement", but rather absolute/incremental modifications that apply to the current motion mode.

Per https://www.linuxcnc.org/docs/devel/html/gcode/overview.html#gcode:modal-groups , G38 is in the same modal group as G1 and G1, so it is correct for it to be "sticky".

cranefist, 2022-11-23:
Ok, this is exactly what i was looking for.. as i just started this and haven't dealt with g-code this way before.

Now this makes sense... it was so weird :)

So do i need to start every command with a G1?

MitchBradley, 2022-11-23:
You only need to start a motion block with G1 if the previous mode was different.  You can say:

```
G1 X1 F100
Y2
X2
G0 Y10
X10
G1 X0
```

Consult the reference above for which motion modes replace each other.

MitchBradley, 2022-11-23:
In a macro or other "packaged" sequence of GCode, it is good practice not to depend on assumptions of what the mode is, so such sequences should establish a known state at the beginning.

closed 2022-11-23 by MitchBradley

cranefist, 2022-11-23:
Great, i think i got all working now. Perfect!

If you are curious, this is my machine:
[image]

I think its a good example of how nice FluidNC is, a proxxon MF70 conversion.
The whole controller is built into the machine, and as it runs via WIFI all i need is a 24v power brick and it works through a phone. So quite minimalistic setup for a tiny machine.

Great software overall. I use klipper on 3d printers, so not going to use anything where you need to compile stuff every time you change machine settings.

I just knew it was a software issue, so i concentrated on that. Thanks for your help again!

bdring, 2022-11-23:
I also suggest using the P parameter with G38. It is designed for use with macros. It is typically more accurate because the deceleration is not an issue.

<http://wiki.fluidnc.com/en/config/probe#optional-p-parameter>

MitchBradley, 2022-11-24:
I have one of those Proxxon mills that someone gave me.  Unfortunately he did a really crappy job of trying to convert it to CNC so it is essentially useless.  There are no thrust bearings on the screws so the couplers take all the thrust load and stretch or come loose, and he glued the leadscrew nuts to the bottom of the table.