bdring/FluidNC #721: Problem: Y axis motor keeps driving into limit switch while X axis seeks home

ColPaulR, 2022-11-26:
### Controller Board

MKS Tinybee

### Machine Description

4 Axis dual Y-axis motor gantry (XYYZA)  

### Input Circuits

```yaml
1 limit switch on Y with 2 DM556 drivers for Y using separate IO pins (e.g. 2 DM556s are not wired in parallel but controlled with unique io pins)
```

### Configuration file

```yaml
board: MKS TinyBee V1.0_001
name: CNC Router XYZ no spindle
meta: (01.02.2022) by Skorpi

kinematics:
  Cartesian:

stepping:
  engine: I2S_STATIC
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  x:
    # X
    steps_per_mm: 250
    max_rate_mm_per_min: 5250.000
    acceleration_mm_per_sec2: 300.000
    max_travel_mm: 1300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 2000.000
      settle_ms: 250
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.33:low
      hard_limits: true
      pulloff_mm: 2.500
      standard_stepper:
        step_pin: I2SO.1
        direction_pin: I2SO.2:low
        disable_pin: I2SO.0

  y:
    # Y
    steps_per_mm: 250
    max_rate_mm_per_min: 5250.000
    acceleration_mm_per_sec2: 300.000
    max_travel_mm: 1300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
[... 256 lines cut ...]
[MSG:DBG:  X Neg Limit 1]
[MSG:DBG: Homing limited X Y Y2]
[MSG:DBG: Homing nextPhase Pulloff1]
[MSG:DBG: Starting from -152.504,-101.976,0.000]
[MSG:DBG: Planned move to -150.004,-99.976,0.000 @ 424.264]
[MSG:DBG:  X Neg Limit 0]
[MSG:DBG:  Y Neg Limit 0]
[MSG:DBG: CycleStop Pulloff1]
[MSG:DBG: Homing nextPhase Pulloff2]
[MSG:DBG: mpos was -150.004,-99.976,0.000]
[MSG:DBG: mpos becomes 0.000,0.000,0.000]
[MSG:DBG: mpos transformed 0.000,0.000,0.000]
[MSG:DBG: Homing done]
ok
```

### User Interface Software

Fluidterm and  CNCjs

### What happened?

I manually moved to X150 Y100 Z-50 after homing so that the X axis has to move a longer distance than the Y when homing. I reset (WebGUI) the Tinybee and then executed a homing cycle. Both X and Y are homing cycle 2. As expected, Y gets to the limit switch first and Y keeps driving (griding noise) until X hits the home switch. After that, they both reverse, slow seek the switches and then pull back. I would expect the Y axis to quit driving once Y hits the Y limit switch, but it does not sound like that is happening.

I reversed the condition so that Y has to travel further than X, and everything behaves correctly (X quits driving while waiting for Y to find the switch). This leads me to believe that this has something to do with me only currently having 1 Y limit switch for 1 motors. My goal is to add another limit switch to the other Y motor and execute axis squaring, but I'm not there yet. I tried adding the same Y limit switch pin (gpio.32:low) to motor1 but got an error message ([MSG:ERR: Configuration error at /machine/axes/y/motor1: ERR: gpio.32:low - Pin is already used.]).

bdring, 2022-11-26:
Try using this release. 

https://github.com/bdring/FluidNC/releases/tag/v3.6.5-pre4

It has a few fixes for homing 2 axes at once. If it does not work, we would rather debug that firmware revision.

Be sure the send the same debug text for the homing test.

ColPaulR, 2022-11-27:
I updated to v3.6.5-pre4 via WebGUI. Below is the debug log. My theory is that the second motor (with limit_neg_pin: NO_PIN) is still driving until X limit is triggered. I confirmed that the Y axis was square. Moved to X150Y100 and then rehomed. I measured the square before and after with dial caliper. Motor without switch is now out of square by several millimeters.

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12812
load:0x40080400,len:3032
entry 0x400805e4

[MSG:INFO: FluidNC v3.6.5-pre4]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:MKS_TinyBee_1_XYZAB.yaml]
[MSG:DBG: Running after-parse tasks]
[MSG:DBG: Checking configuration]
[MSG:INFO: Machine CNC Router XYZ no spindle]
[MSG:INFO: Board MKS TinyBee V1.0_001]
[MSG:INFO: I2SO BCK:gpio.25 WS:gpio.26 DATA:gpio.27]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 4]
[MSG:INFO: Axis X (0.000,1300.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.1 Dir:I2SO.2:low Disable:I2SO.0]
[MSG:INFO:  X Neg Limit gpio.33:low]
[MSG:DBG:  X Neg Limit 0]
[MSG:INFO: Axis Y (0.000,1300.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.4 Dir:I2SO.5:low Disable:I2SO.3]
[MSG:INFO:  Y Neg Limit gpio.32:low]
[MSG:DBG:  Y Neg Limit 0]
[MSG:INFO:   Motor1]
[MSG:INFO:     standard_stepper Step:I2SO.7 Dir:I2SO.8:low Disable:I2SO.6]
[MSG:INFO: Axis Z (-200.000,100.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.10 Dir:I2SO.11 Disable:I2SO.9]
[MSG:INFO:  Z Neg Limit gpio.22:low]
[MSG:DBG:  Z Neg Limit 0]
[MSG:INFO: Axis A (-325.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.13 Dir:I2SO.14 Disable:I2SO.12]
[MSG:INFO: feed_hold_pin gpio.39:low]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Flood coolant I2SO.20]
[MSG:INFO: Mist coolant I2SO.19]
[MSG:INFO: Probe Pin: gpio.36:low]
[MSG:INFO: Connecting to STA SSID:RounsavallWPA2]
[MSG:INFO: Connecting.]
[MSG:INFO: Connected - IP is 192.168.0.235]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.6 [FluidNC v3.6.5-pre4 (wifi) '$' for help]
[MSG:DBG:  X Neg Limit 0]
[MSG:DBG:  Y Neg Limit 0]
[MSG:DBG:  Z Neg Limit 0]
[MSG:DBG: WebSocket 0 from 192.168.0.230 uri /]
[MSG:DBG: Homing Cycle Z]
[MSG:DBG: Homing nextPhase FastApproach]
[MSG:DBG: Starting from 0.000,0.000,0.000]
[MSG:DBG: Planned move to 0.000,0.000,330.000 @ 1000.000]
[MSG:DBG:  Z Neg Limit 1]
[MSG:DBG: Homing limited Z]
[MSG:DBG: Homing nextPhase Pulloff0]
[MSG:DBG: Starting from 0.000,0.000,68.648]
[MSG:DBG: Planned move to 0.000,0.000,67.148 @ 100.000]
[MSG:DBG:  Z Neg Limit 0]
[MSG:DBG: CycleStop Pulloff0]
[MSG:DBG: Homing nextPhase SlowApproach]
[MSG:DBG: Starting from 0.000,0.000,67.148]
[MSG:DBG: Planned move to 0.000,0.000,68.798 @ 100.000]
[MSG:DBG:  Z Neg Limit 1]
[MSG:DBG: Homing limited Z]
[MSG:DBG: Homing nextPhase Pulloff1]
[MSG:DBG: Starting from 0.000,0.000,68.452]
[MSG:DBG: Planned move to 0.000,0.000,66.952 @ 100.000]
[MSG:DBG:  Z Neg Limit 0]
[MSG:DBG: CycleStop Pulloff1]
[MSG:DBG: Homing nextPhase Pulloff2]
[MSG:DBG: mpos was 0.000,0.000,66.952]
[MSG:DBG: mpos becomes 0.000,0.000,100.000]
[MSG:DBG: mpos transformed 0.000,0.000,100.000]
[MSG:DBG: Homing Cycle XY]
[MSG:DBG: Homing nextPhase FastApproach]
[MSG:DBG: Starting from 0.000,0.000,100.000]
[MSG:DBG: Planned move to -1430.000,-1430.000,100.000 @ 2828.427]
[MSG:DBG:  Y Neg Limit 1]
[MSG:DBG: Homing limited Y Y2]
[MSG:DBG:  X Neg Limit 1]
[MSG:DBG: Homing limited X Y Y2]
[MSG:DBG: Homing nextPhase Pulloff0]
[MSG:DBG: Starting from -152.580,-102.104,100.000]
[MSG:DBG: Planned move to -150.080,-100.104,100.000 @ 424.264]
[MSG:DBG:  X Neg Limit 0]
[MSG:DBG:  Y Neg Limit 0]
[MSG:DBG: CycleStop Pulloff0]
[MSG:DBG: Homing nextPhase SlowApproach]
[MSG:DBG: Starting from -150.080,-100.104,100.000]
[MSG:DBG: Planned move to -152.830,-102.304,100.000 @ 424.264]
[MSG:DBG:  Y Neg Limit 1]
[MSG:DBG: Homing limited Y Y2]
[MSG:DBG:  X Neg Limit 1]
[MSG:DBG: Homing limited X Y Y2]
[MSG:DBG: Homing nextPhase Pulloff1]
[MSG:DBG: Starting from -152.492,-101.868,100.000]
[MSG:DBG: Planned move to -149.992,-99.868,100.000 @ 424.264]
[MSG:DBG:  X Neg Limit 0]
[MSG:DBG:  Y Neg Limit 0]
[MSG:DBG: CycleStop Pulloff1]
[MSG:DBG: Homing nextPhase Pulloff2]
[MSG:DBG: mpos was -149.992,-99.868,100.000]
[MSG:DBG: mpos becomes 0.000,0.000,100.000]
[MSG:DBG: mpos transformed 0.000,0.000,100.000]
[MSG:DBG: Homing done]
ok

bdring, 2022-11-27:
When Y is grinding do both Y motors grind?

ColPaulR, 2022-11-27:
It sounds like it only one motor. If I separate the homing cycle so that Y has its own homing cycle, everything works.  Can a single switch be assigned to the axes vs the motor?

MitchBradley, 2022-11-29:
I have reproduced the problem.  I only happens during multi-axis homing when the dual-motor axis has only one switch; dual-motor/dual-switch works fine.  I am looing for a solution.

ColPaulR, 2022-11-29:
Thanks. I added the 2nd axis switch already to do the axis squaring and avoid the issue. I'd be happy to revert to the single-axis switch to troubleshoot or confirm the fix if you need. Thanks for all you do for the community. I love open-source solutions.

MitchBradley, 2022-11-29:
What I really want is for someone to step forward and be a tester for all things related to homing.  The number of combinations is ridiculous, what with CoreXY vs Cartesian vs (future kinematics) X single vs multi-axis cycles X switch-per-motor vs single-switch X pos limit vs neg limit vs pos+neg vs all limit X single pulloff vs per-motor pulloff X pre-pulloff vs no-pre-pulloff.  Not to mention the frequent problems people have just wiring their switches correctly.  So few people give back to the project...

ColPaulR, 2022-11-29:
What does being a tester involve? I'll help if I can but I only have Cartesian machines (prusa 3d and queenbee gantry with 4th axis rotary).

MitchBradley, 2022-11-30:
Mainly testing involves taking the time to systematically work through the matrix of possibilities, generating a test report for each release or pre-release showing what worked as expected and what did not.  Most if not all of it can be done with just three bare motors and some switches that you can activate by hand.  In fact that is how I developed most of the code.

You just have to learn what you expect to see happening.  For CoreXY, when two motors are moving that is a single-axis move, while one motor is a diagonal.  Cartesian, of course, is just the opposite.

You would need config files for the various cases of limit switch configuration.

It is slightly easier if you have a full machine representing a configuration, but one quickly learns to infer what is happening from bare motors.

ColPaulR, 2022-11-30:
I'm willing to give back, depending on the time demands. I'll need to get some motors and another EPS_32 based board from aliexpress. Any recommendations on the board? If not, I'll just pick up another TinyBee to keep as a spare. I don't have any ESP_32 debugging hardware, but sounds like it's not needed.

MitchBradley, 2022-11-30:
MKS DLC32 is probably the least expensive, with Tinybee being only a couple of dollars more.  Since you already have Tinybee, that makes sense.  Debugging hardware is unnecessary; I have it and never use it.  Any motors and drivers will do for the purpose of testing homing.  Mainly you need some switches mounted close together so you can activate two at once with your thumb.  I have a bank of 4 switches but I typically one use 3 at a time when testing homing.  I can reconfigure to put them a various places on the "machine".

ColPaulR, 2022-11-30:
Roger. I've got another TinyBee on the way, along with 4 steppers. Already have limit switches laying around.

commit 64498121 (2022-11-29): Fix #721 - multi-axis homing with dual-motor/single-switch
commit 86dc6f68 (2022-12-08): Fix #721 - multi-axis homing with dual-motor/single-switch
commit f247c00c (2022-12-08): Merge pull request #728 from bdring/Fix721

Fix #721 - multi-axis homing with dual-motor/single-switch
closed 2022-11-30 by MitchBradley