bdring/FluidNC #722: Problem:

JHarding86, 2022-11-26:
### Controller Board

[4 Axis TMC2209 CNC Controller](https://www.tindie.com/products/33366583/4-axis-tmc2209-cnc-controller/)

### Machine Description

CO2 laser engraver. 3 axes x,y,z all single motor. Microswitch endstops

### Configuration file

```yaml
board: FluidNC 4X 2209
name: TMC2209 XYZ  Laser
meta: 2022-08-09 B. Dring for James Harding 340638
stepping:
  engine: I2S_static
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
  segments: 12

axes:
  shared_stepper_disable_pin: NO_PIN
  shared_stepper_reset_pin: NO_PIN
  x:
    steps_per_mm: 177.800
    max_rate_mm_per_min: 18000.000
    acceleration_mm_per_sec2: 1500.000
    max_travel_mm: 750.000
    soft_limits: false
    homing:
      cycle: 1
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 240.000
      seek_mm_per_min: 3000.000
      settle_ms: 250
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.36
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 5.000
      tmc_2209:
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 0
        stallguard_debug: false
        toff_coolstep: 3
        step_pin: I2SO.2
        direction_pin: I2SO.1:low
        disable_pin: I2SO.0
        r_sense_ohms: 0.110
        run_amps: 2.000
        hold_amps: 2.000
        microsteps: 64
        toff_disable: 0
        toff_stealthchop: 5
        use_enable: false
        addr: 0
        uart:
          txd_pin: gpio.16
          rxd_pin: gpio.4
          rts_pin: NO_PIN
          cts_pin: NO_PIN
          baud: 115200
          mode: 8N1

  y:
    steps_per_mm: 106.670
    max_rate_mm_per_min: 12000.000
    acceleration_mm_per_sec2: 1500.000
    max_travel_mm: 410.000
    soft_limits: false
    homing:
      cycle: 1
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 240.000
      seek_mm_per_min: 3000.000
      settle_ms: 250
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.39
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 10.000
      tmc_2209:
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 0
        stallguard_debug: false
        toff_coolstep: 3
        step_pin: I2SO.5
        direction_pin: I2SO.4
        disable_pin: I2SO.7
        r_sense_ohms: 0.110
        run_amps: 2.000
        hold_amps: 2.000
        microsteps: 32
        toff_disable: 0
        toff_stealthchop: 5
        use_enable: false
        addr: 1

  z:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 200.000
    acceleration_mm_per_sec2: 25.000
    max_travel_mm: 200.000
    soft_limits: false
    homing:
      cycle: 2
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 250
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 6.000
      tmc_2209:
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 0
        stallguard_debug: false
        toff_coolstep: 3
        step_pin: I2SO.10
        direction_pin: I2SO.9
        disable_pin: I2SO.8
        r_sense_ohms: 0.110
        run_amps: 2.000
        hold_amps: 1.000
        microsteps: 16
        toff_disable: 0
        toff_stealthchop: 5
        use_enable: false
        addr: 2

kinematics:
  Cartesian:

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: NO_PIN
  card_detect_pin: NO_PIN

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

coolant:
  flood_pin: gpio.25
  mist_pin: NO_PIN
  delay_ms: 5000

probe:
  pin: NO_PIN
  check_mode_start: true

macros:
  startup_line0: 
  startup_line1: 
  macro0: 
  macro1: 
  macro2: 
  macro3: 

start:
  must_home: true
  deactivate_parking: false
  check_limits: false

parking:
  enable: false
  axis: Z
  target_mpos_mm: -5.000
  rate_mm_per_min: 800.000
  pullout_distance_mm: 5.000
  pullout_rate_mm_per_min: 250.000

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
planner_blocks: 16
Laser:
  pwm_hz: 5000
  output_pin: gpio.26
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  tool_num: 0
  speed_map: 0=0.000% 200=3.000% 500=21.000% 1000=100.000%
  off_on_alarm: false
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.6.4-pre3]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:laser.yaml]
[MSG:INFO: Machine TMC2209 XYZ  Laser]
[MSG:INFO: Board FluidNC 4X 2209]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: No SD Card CS Pin]
[MSG:INFO: See http://wiki.fluidnc.com/en/config/sd_card#sdfallbackcs-access-sd-without-a-config-file]
[MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,750.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Trinamic Stepper Uart Tx:gpio.16 Rx:gpio.4 RTS:NO_PIN Baud:115200]
[MSG:INFO:     tmc_2209 Step:I2SO.2 Dir:I2SO.1:low Disable:I2SO.0 Addr:0 R:0.110]
[MSG:INFO:  X Neg Limit gpio.36]
[MSG:INFO: Axis Y (0.000,410.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2209 Step:I2SO.5 Dir:I2SO.4 Disable:I2SO.7 Addr:1 R:0.110]
[MSG:INFO:  Y Neg Limit gpio.39]
[MSG:INFO: Axis Z (0.000,200.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2209 Step:I2SO.10 Dir:I2SO.9 Disable:I2SO.8 Addr:2 R:0.110]
[MSG:INFO:  Z Neg Limit gpio.34]
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: Z Axis driver test passed]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Laser Ena:NO_PIN Out:gpio.26 Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: Flood coolant gpio.25]
[MSG:INFO: WiFi is disabled]
```

### User Interface Software

Lightburn 1.2.04

### What happened?

My 4 axis board uses an external 24v power supply.
I stream my gcode via USB which powers the ESP32.

To get the 4 axis board to function correctly, I have to shut off the power to the 24v power supply and disconnect the USB cable to my laptop. Then I can power on the 24v power supply wait a few seconds, and then plug in the USB to my laptop.

**If I do not power up in this sequence, the steps/mm are the most obvious thing that is wrong, as the x and y axis will make huge moves when commanding very small moves.**

I would like to be able to leave my USB cable plugged in and just power on the machine.

alvarotorijano, 2022-11-26:
This seems not to be a fw issue. What board are you using can you share or reference the schematics? 
Have you tried to power up the ESP using the same voltage supply using a buck converter for example?

JHarding86, 2022-11-26:
The board I have is linked in the issue. I bought it from @bdring

bdring, 2022-11-26:
This is normal behavior. If you power via USB first, the motor drivers will not respond. They need motor voltage to do anything. If you apply the motor voltage later they will work, but only with the default settings, because the motor configuration phase has passed. This is a limitation of the ESP32 module. There is no way to force it to be powered from the motor voltage alone.

You will be able to see this if you run via USB only and hit the reset button on the ESP32 module.

You can send the `$Motors/Init` or `$MI` at any time resent the configuration information to the motors. Motor power must be on for it to work, of course. 

I added a note about this to the wiki.

http://wiki.fluidnc.com/en/hardware/official/TMC2209_4_Axis

closed 2022-12-02 by MitchBradley

JHarding86, 2022-12-14:
After having used this methodology for about a month, I am back to having to disconnect the USB from my computer and then power on the 4 Axis board with the 24v power supply, then plug in the USB again.

@bdring @MitchBradley 
Using $MI or $Motors/Init does not work reliably. Sometimes the axes home in the wrong direction causing the machine to crash.

bdring, 2022-12-14:
Does it work if you hit the reset button after applying main power? That could give a clue why $MI is not working?

AussieMakerGeek, 2023-03-08:
The other option would be to have a 5v regulator from the 24v to feed the board USB 5v power, then use a custom USB cable that has the 5v wire cut (or cut the track on the board so power from USB does not go to it).  I do this with my 3d printer and it works fine.