bdring/FluidNC #723: Problem: XPro V5 Machine file Converter problems & TMC Driver issues

synman, 2022-11-27:
### Controller Board

XPro V5

### Machine Description

Queen Bee Pro - 750mm x 750mm - XYYZ 

### Configuration file

```yaml
board: XPro V5
name: xPro V5 XYYZ
stepping:
  engine: RMT
  idle_ms: 255
  pulse_us: 2
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 200.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      tmc_5160:
        step_pin: gpio.12
        direction_pin: gpio.14
        use_enable: true
        cs_pin: gpio.17
        spi_index: 1
        r_sense_ohms: 0.050
        run_amps: 1.800
        hold_amps: 1.250
        microsteps: 8
        toff_disable: 0
        toff_stealthchop: 5
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 16
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4

  y:
    steps_per_mm: 200.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      tmc_5160:
        step_pin: gpio.27
        direction_pin: gpio.26
        use_enable: true
        cs_pin: NO_PIN
        spi_index: 2
        r_sense_ohms: 0.050
        run_amps: 1.800
        hold_amps: 1.250
        microsteps: 8
        toff_disable: 0
        toff_stealthchop: 5
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 16
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4

    motor1:
      limit_neg_pin: gpio.36:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      tmc_5160:
        step_pin: gpio.33
        direction_pin: gpio.32
        use_enable: true
        cs_pin: NO_PIN
        spi_index: 3
        r_sense_ohms: 0.050
        run_amps: 1.800
        hold_amps: 1.250
        microsteps: 16
        toff_disable: 0
        toff_stealthchop: 5
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 16
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4
  z:
    steps_per_mm: 200.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.39:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      tmc_5160:
        step_pin: gpio.15
        direction_pin: gpio.2
        use_enable: true
        cs_pin: NO_PIN
        spi_index: 4
        r_sense_ohms: 0.050
        run_amps: 1.800
        hold_amps: 1.250
        microsteps: 16
        toff_disable: 0
        toff_stealthchop: 5
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 16
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

probe:
  pin: gpio.22:low
  check_mode_start: true

control:
  safety_door_pin: gpio.16:pu
  macro0_pin: gpio.13:low
  macro1_pin: gpio.0:low

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:

start:
  must_home: false

coolant:
  mist_pin: gpio.21
  delay_ms: 0
```

### Startup Messages

```text
Failed to capture them <details observed in ticket)
```

### User Interface Software

WebUI and CNCjs

### What happened?

The machine file converter did not work (https://github.com/bdring/Grbl_Esp32/issues/1324).  I was able to work around this by using the XPro V5 config.yaml template on the wiki.

However, a couple issues:

- I had to fall back to 3.4.4 to get the TMC drivers to work.  https://github.com/bdring/FluidNC/issues/485 for reference
- I was able to revert X axis movement by setting the motor0/tmc_5160/direction_pin to low, however when I tried to do the same for Y motors 0 and 1, the driver would crash.

bdring, 2022-11-27:
Startup messages are saved and can be displayed with $SS. Please get those messages and edit your first post.

synman, 2022-11-27:
I have since reverted to ESP32 grbl.  I'll pick this back up after #485 closes and be sure to collect the necessary start-up data if I have issues then.

Unless there's something key here you'd like to see.  I am willing to go back to fluidnc temporarily if there is a piece of data there needed to unblock it.

bdring, 2022-11-27:
I have an xPro at home. I should be able to check it when I get home from vacation.

bdring, 2022-12-05:
It works perfectly for me. I restarted in 10 times without issues

```
[MSG:INFO: FluidNC v3.6.5-pre4]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:xPro_XYZA.yaml]
[MSG:INFO: Machine xPro V5 XYZA]
[MSG:INFO: Board XPro V5]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 4]
[MSG:INFO: Axis X (150.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_5160 Step:gpio.12 Dir:gpio.14 CS:gpio.17 Disable:NO_PIN Index:1 R:0.050]
[MSG:INFO:  X Neg Limit gpio.35:low]
[MSG:INFO: Axis Y (150.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_5160 Step:gpio.27 Dir:gpio.26 CS:NO_PIN Disable:NO_PIN Index:2 R:0.050]
[MSG:INFO:  Y Neg Limit gpio.34:low]
[MSG:INFO: Axis Z (150.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_5160 Step:gpio.15 Dir:gpio.2 CS:NO_PIN Disable:NO_PIN Index:4 R:0.050]
[MSG:INFO:  Z Neg Limit gpio.39:low]
[MSG:INFO: Axis A (150.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_5160 Step:gpio.33 Dir:gpio.32 CS:NO_PIN Disable:NO_PIN Index:3 R:0.050]
[MSG:INFO:  A Neg Limit gpio.36:low]
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: Z Axis driver test passed]
[MSG:INFO: A Axis driver test passed]
[MSG:INFO: safety_door_pin gpio.16:low]
[MSG:INFO: macro0_pin gpio.13:low]
[MSG:INFO: macro1_pin gpio.0:low]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Mist coolant gpio.21]
[MSG:INFO: Probe Pin: gpio.22:low]
[MSG:INFO: WiFi is disabled]
```

synman, 2022-12-06:
Please note the orientation / configuration of my machine.  Y has two motors and my config follows the XYYZ nomenclature.

You'll find most use cases for the xPro (at least those where it is bundled with the BM QB) to use this configuration.

bdring, 2022-12-06:
With your config

```
[MSG:INFO: FluidNC v3.6.5-pre4]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:723.yaml]
[MSG:INFO: Machine xPro V5 XYYZ]
[MSG:INFO: Board XPro V5]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (150.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_5160 Step:gpio.12 Dir:gpio.14 CS:gpio.17 Disable:NO_PIN Index:1 R:0.050]
[MSG:INFO:  X Neg Limit gpio.35:low]
[MSG:INFO: Axis Y (150.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_5160 Step:gpio.27 Dir:gpio.26 CS:NO_PIN Disable:NO_PIN Index:2 R:0.050]
[MSG:INFO:  Y Neg Limit gpio.34:low]
[MSG:INFO:   Motor1]
[MSG:INFO:     tmc_5160 Step:gpio.33 Dir:gpio.32 CS:NO_PIN Disable:NO_PIN Index:3 R:0.050]
[MSG:INFO:  Y2 Neg Limit gpio.36:low]
[MSG:INFO: Axis Z (150.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_5160 Step:gpio.15 Dir:gpio.2 CS:NO_PIN Disable:NO_PIN Index:4 R:0.050]
[MSG:INFO:  Z Neg Limit gpio.39:low]
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: Y2 Axis driver test passed]
[MSG:INFO: Z Axis driver test passed]
[MSG:INFO: safety_door_pin gpio.16:pu]
[MSG:INFO: macro0_pin gpio.13:low]
[MSG:INFO: macro1_pin gpio.0:low]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Mist coolant gpio.21]
[MSG:INFO: Probe Pin: gpio.22:low]
[MSG:INFO: WiFi is disabled]
```

brianredbeard, 2022-12-10:
@synman:  In the past I regularly have had to issue a motor disable command (`$MD`) followed by a motor enable command (`$ME`) on boot up due to enumeration problems with the motors.  I had not run the update to `v3.6.5-pre4` yet, but given @bdring's testing I would have a high degree of confidence in his measurements.

synman, 2022-12-10:
I promise to revisit this just as soon as I'm back on the xPro.  I'm working on another controller at the moment.

synman, 2022-12-23:
I can pick this back up again.  Will wire up my xPro over the next couple days and see what I can see.  I've learned quite bit about FluidNC since opening this so yeah... don't be surprised if this gets closed as user error.

synman, 2023-02-26:
Finally was able to find the time to power up the xPro V5 and play around.   Everything is working fine with it.  

In fact, it appears FluidNC fixes a bug that is present in (current and possibly order) GrblESP32 builds for it that make holding and running current unreliable.  The Spark-Concepts issue page is full of complaints of fried drivers and I believe this to be root cause.

Actual Holding Current (commanding 1.25A):

[image]

Actual Holding Current (commanding 1.8A):

[image]

Actual Running Current (commanding 1.8A):

[image]

[image]

Here's what a 1.25A holding current under FluidNC looks like:

[image]

And a 1.8A running current under FluidNC:

[image]

[image]

synman, 2023-02-26:
@bdring closing this out -- thought you might find the "current" situation interesting.

closed 2023-02-26 by synman