bdring/FluidNC #724: Problem: If a feedhold is issued during homing, FluidNC reports its state as Hold(1), but doesn't stop.

milo-brandt, 2022-11-27:
### Controller Board

A board of my own design - but the problem is not hardware related.

### Machine Description

Gantry router with dual Y-axis and a rotary A-axis.

### Configuration file

```yaml
board: None
name: None
meta: 
stepping:
  engine: RMT
  idle_ms: 255
  pulse_us: 5
  dir_delay_us: 5
  disable_delay_us: 0
  segments: 6

axes:
  shared_stepper_disable_pin: NO_PIN
  shared_stepper_reset_pin: NO_PIN
  x:
    steps_per_mm: 320.000
    max_rate_mm_per_min: 6000.000
    acceleration_mm_per_sec2: 250.000
    max_travel_mm: 1200.000
    soft_limits: true
    homing:
      cycle: 2
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 25.000
      seek_mm_per_min: 500.000
      settle_ms: 250
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.36
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: gpio.32
        direction_pin: gpio.33
        disable_pin: NO_PIN

  y:
    steps_per_mm: 320.000
    max_rate_mm_per_min: 6000.000
    acceleration_mm_per_sec2: 250.000
    max_travel_mm: 900.000
    soft_limits: true
    homing:
      cycle: 2
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 25.000
      seek_mm_per_min: 500.000
      settle_ms: 250
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.39
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: gpio.25
        direction_pin: gpio.26
        disable_pin: NO_PIN

    motor1:
      limit_neg_pin: gpio.34
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: gpio.27
        direction_pin: gpio.14
        disable_pin: NO_PIN

  z:
    steps_per_mm: 320.000
    max_rate_mm_per_min: 6000.000
    acceleration_mm_per_sec2: 250.000
    max_travel_mm: 120.000
    soft_limits: true
    homing:
      cycle: 1
      allow_single_axis: true
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 25.000
      seek_mm_per_min: 500.000
      settle_ms: 250
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.35
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: gpio.12
        direction_pin: gpio.13
        disable_pin: NO_PIN

  a:
    steps_per_mm: 6.667
    max_rate_mm_per_min: 100000.000
    acceleration_mm_per_sec2: 1440.000
    max_travel_mm: 1000.000
    soft_limits: false
    homing:
      cycle: 0
      allow_single_axis: true
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 50.000
      seek_mm_per_min: 200.000
      settle_ms: 250
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: gpio.23
        direction_pin: gpio.22
        disable_pin: NO_PIN

kinematics:
  Cartesian:

spi:
  miso_pin: NO_PIN
  mosi_pin: NO_PIN
  sck_pin: NO_PIN

sdcard:
  cs_pin: NO_PIN
  card_detect_pin: NO_PIN

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

probe:
  pin: gpio.21:low
  check_mode_start: true

macros:
  startup_line0: 
  startup_line1: 
  macro0: 
  macro1: 
  macro2: 
  macro3: 

start:
  must_home: true
  deactivate_parking: false
  check_limits: false

parking:
  enable: false
  axis: Z
  target_mpos_mm: -5.000
  rate_mm_per_min: 800.000
  pullout_distance_mm: 5.000
  pullout_rate_mm_per_min: 250.000

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: true
planner_blocks: 16
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.6.5-pre4]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config.yml]
[MSG:INFO: Machine None]
[MSG:INFO: Board None]
[MSG:INFO: SPI not defined]
[MSG:INFO: No SD Card CS Pin]
[MSG:INFO: See http://wiki.fluidnc.com/en/config/sd_card#sdfallbackcs-access-sd-without-a-config-file]
[MSG:INFO: Stepping:RMT Pulse:5us Dsbl Delay:0us Dir Delay:5us Idle Delay:255ms]
[MSG:INFO: Axis count 4]
[MSG:INFO: Axis X (0.000,1200.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.32 Dir:gpio.33 Disable:NO_PIN]
[MSG:INFO:  X Neg Limit gpio.36]
[MSG:INFO: Axis Y (0.000,900.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.25 Dir:gpio.26 Disable:NO_PIN]
[MSG:INFO:  Y Neg Limit gpio.39]
[MSG:INFO:   Motor1]
[MSG:INFO:     standard_stepper Step:gpio.27 Dir:gpio.14 Disable:NO_PIN]
[MSG:INFO:  Y2 Neg Limit gpio.34]
[MSG:INFO: Axis Z (-120.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.12 Dir:gpio.13 Disable:NO_PIN]
[MSG:INFO:  Z Pos Limit gpio.35]
[MSG:INFO: Axis A (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.23 Dir:gpio.22 Disable:NO_PIN]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Probe Pin: gpio.21:low]
[MSG:INFO: WiFi is disabled]
```

### User Interface Software

FluidTerm

### What happened?

If you send the following lines to FluidNC (with some delay between)

```
$H
!
?
```

FluidNC will respond with something like this:

```
<Hold:1|MPos:371.469,460.263,30.084,0.000|FS:500,0>
```

Despite reporting being in the Hold:1 state, the machine does not actually stop its motion.

I  ran into this while implementing an interface and wanted a "graceful stop" feature whereby I could issue a feedhold command, wait while the machine was in Hold:1 (up to some reasonable time limit), then issue the reset command to allow the machine to come to a controlled stop and reset without losing position.

The current behavior where `Hold:1` doesn't necessarily mean that the machine is decelerating seems counterintuitive. I would prefer that the commands I listed above would either behave as...

1. The machine ignores the feedhold and reports `<Homing|...>` as its status.
2. The machine stops (either with the behavior of a feedhold or a reset).

It looks like (1) is pretty easy to implement (i.e. I think one can just move `Homing` to be with states like `Sleep` in `protocol_do_feedhold`) and (2) doesn't look terrible either (and there are some comments about it in the code already). I could likely write the code to make a change either way, but I'm unsure of what the best behavior here would be.

bdring, 2022-11-27:
Feedhold is generally not done during homing. We can look into handling it better.

Most people issue a reset to cancel homing.

MitchBradley, 2022-12-02:
PR #735 adopts solution (1) - feedhold is ignored during homing, with a message saying to use Reset.

commit 33d7fd08 (2022-12-07): Fix #724 - ignore feedhold while homing
commit 43dfd244 (2022-12-07): Merge pull request #735 from bdring/Fix724

Fix #724 - ignore feedhold while homing
commit 2109673b (2022-12-07): Fix #724 - ignore feedhold while homing
closed 2022-12-08 by bdring