bdring/FluidNC #725: Problem:

ipsod, 2022-11-28:
### Controller Board

Makerbase Tiny Bee v1.0

### Machine Description

board only, no motors or switches

### Configuration file

```yaml
name: "ESP32 Dev Controller V4"
board: "ESP32 Dev Controller V4"

stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 1
  pulse_us: 2
  disable_delay_us: 0

axes:
  # shared_stepper_disable_pin: gpio.13:low

  x:
    steps_per_mm: 200
    max_rate_mm_per_min: 60
    acceleration_mm_per_sec2: 120
    max_travel_mm: 100
    motor0:
      # limit_neg_pin: gpio.17:low:pu
      stepstick:
        step_pin: gpio.12
        direction_pin: gpio.13
        disable_pin: gpio.5
    motor1:
      null_motor:

  y:
    steps_per_mm: 200
    max_rate_mm_per_min: 2000.0
    acceleration_mm_per_sec2: 2000.0
    max_travel_mm: 1000
    homing:
      cycle: 2
      mpos_mm: 10
      positive_direction: false

    motor0:
      stepstick:
        step_pin: gpio.27
        direction_pin: gpio.14
        disable_pin: gpio.16
    motor1:
      null_motor:

  z:
    steps_per_mm: 200
    max_rate_mm_per_min: 2000.0
    acceleration_mm_per_sec2: 2000.0
    max_travel_mm: 1000
    homing:
      cycle: 1
      mpos_mm: 10
      positive_direction: true

    motor0:
      stepstick:
        step_pin: gpio.19
        direction_pin: gpio.23
[... 203 lines cut ...]
```1669611683.740916<< SerialConn(1669611620.0) ok
======ok======
======receiving reply to request relative_jog(3187, pos=XYZABC(x=-0.0014, y=-0.01, z=0.0, a=None, b=None, c=None),) T_1669611683741348282======
======RECEIVED GRBL REPLY TO REQUEST: b'$j=g91x-0.0014y-0.01f2000.8998\n'======
1669611683.7419417<< SerialConn(1669611620.0) ok
======ok======
1669611683.742353>> $j=g91x-0.0018y-0.01667f2000.8998
======receiving reply to request relative_jog(3188, pos=XYZABC(x=-0.0016, y=-0.013333333333333334, z=0.0, a=None, b=None, c=None),) T_1669611683742670631======
======RECEIVED GRBL REPLY TO REQUEST: b'$j=g91x-0.0016y-0.01333f2000.8998\n'======
!!error_note!!  error:  _jog_wait_for_ready()   WRONG STATE
!!error_note!!  error:  _jog_wait_for_ready()   WRONG STATE
1669611683.744877<< SerialConn(1669611620.0) <Jog|WPos:-0.415,4.575,0.000|Bf:15,128|FS:0,0>
1669611683.7458942>> $j=g91x-0.002y-0.02f2000.8998
1669611683.7548559<< SerialConn(1669611620.0) <Idle|WPos:-0.415,4.560,0.000|Bf:15,128|FS:0,0>
1669611683.7552912<< SerialConn(1669611620.0) ok
======ok======
======receiving reply to request relative_jog(3189, pos=XYZABC(x=-0.0017999999999999997, y=-0.016666666666666666, z=0.0, a=None, b=None, c=None),) T_1669611683755678654======
======RECEIVED GRBL REPLY TO REQUEST: b'$j=g91x-0.0018y-0.01667f2000.8998\n'======
1669611683.7562137<< SerialConn(1669611620.0) ok
1669611683.7564247>> $j=g91x-0.0022y-0.02333f2000.8998
======ok======
======receiving reply to request relative_jog(3190, pos=XYZABC(x=-0.0019999999999999996, y=-0.02, z=0.0, a=None, b=None, c=None),) T_1669611683757138575======
======RECEIVED GRBL REPLY TO REQUEST: b'$j=g91x-0.002y-0.02f2000.8998\n'======
1669611683.758382>> $j=g91x-0.0024y-0.02667f2000.8998
```

ipsod, 2022-11-28:
Another error log:

```
======RECEIVED GRBL REPLY TO REQUEST: b'$j=g91x-0.0098f60\n'======
1669613703.2001262>> $j=g91x-0.01f60
1669613703.2010846>> $j=g91x-0.01f60
1669613703.2020416>> $j=g91x-0.01f60
1669613703.2094696<< SerialConn(1669613685.0) ok
======ok======
======receiving reply to request relative_jog(1302, pos=XYZABC(x=-0.01, y=0.0, z=0.0, a=None, b=None, c=None),) T_1669613703209867227======
======RECEIVED GRBL REPLY TO REQUEST: b'$j=g91x-0.01f60\n'======
1669613703.2107263>> $j=g91x-0.01f60
1669613703.2136009<< SerialConn(1669613685.0) <Jog|WPos:-0.595,-3.050,0.000|Bf:0,128|FS:60,0>
1669613703.2202148<< SerialConn(1669613685.0) ok
======ok======
======receiving reply to request relative_jog(1303, pos=XYZABC(x=-0.01, y=0.0, z=0.0, a=None, b=None, c=None),) T_1669613703220657929======
======RECEIVED GRBL REPLY TO REQUEST: b'$j=g91x-0.01f60\n'======
1669613703.2216742>> $j=g91x-0.01f60
1669613703.2333229<< SerialConn(1669613685.0) ok
======ok======
======receiving reply to request relative_jog(1304, pos=XYZABC(x=-0.01, y=0.0, z=0.0, a=None, b=None, c=None),) T_1669613703233679684======
======RECEIVED GRBL REPLY TO REQUEST: b'$j=g91x-0.01f60\n'======
1669613703.2349503>> $j=g91x-0.01f60
1669613703.2463036<< SerialConn(1669613685.0) ok
======ok======
======receiving reply to request relative_jog(1305, pos=XYZABC(x=-0.01, y=0.0, z=0.0, a=None, b=None, c=None),) T_1669613703246704421======
======RECEIVED GRBL REPLY TO REQUEST: b'$j=g91x-0.01f60\n'======
1669613703.2472463>> $j=g91x-0.01f60
$$$  MachineServer().jog_stop()
1669613703.261019<< SerialConn(1669613685.0) ok
======ok======
======receiving reply to request relative_jog(1306, pos=XYZABC(x=-0.01, y=0.0, z=0.0, a=None, b=None, c=None),) T_1669613703261357043======
======RECEIVED GRBL REPLY TO REQUEST: b'$j=g91x-0.01f60\n'======
1669613703.2620792>> $j=g91x-0.01f60
$$$  send_jog_cancel()
1669613703.2654614<< SerialConn(1669613685.0) <Jog|WPos:-0.615,-3.050,0.000|Bf:0,128|FS:60,0>
1669613703.272241<< SerialConn(1669613685.0) ok
======ok======
======receiving reply to request relative_jog(1307, pos=XYZABC(x=-0.01, y=0.0, z=0.0, a=None, b=None, c=None),) T_1669613703272531638======
======RECEIVED GRBL REPLY TO REQUEST: b'$j=g91x-0.01f60\n'======
1669613703.2868586<< SerialConn(1669613685.0) ok
======ok======
======receiving reply to request relative_jog(1308, pos=XYZABC(x=-0.01, y=0.0, z=0.0, a=None, b=None, c=None),) T_1669613703287225123======
======RECEIVED GRBL REPLY TO REQUEST: b'$j=g91x-0.01f60\n'======
1669613703.292296<< SerialConn(1669613685.0) ok
======ok======
======receiving reply to request relative_jog(1309, pos=XYZABC(x=-0.01, y=0.0, z=0.0, a=None, b=None, c=None),) T_1669613703292680736======
======RECEIVED GRBL REPLY TO REQUEST: b'$j=g91x-0.01f60\n'======
1669613703.2939055>> b'\x85'
1669613703.2972727<< SerialConn(1669613685.0) ok
======ok======
======receiving reply to request relative_jog(1310, pos=XYZABC(x=-0.01, y=0.0, z=0.0, a=None, b=None, c=None),) T_1669613703297612710======
======RECEIVED GRBL REPLY TO REQUEST: b'$j=g91x-0.01f60\n'======
1669613703.316961<< SerialConn(1669613685.0) <Jog|WPos:-0.655,-3.050,0.000|Bf:0,128|FS:0,0|WCO:0.000,0.000,0.000>
1669613703.366934<< SerialConn(1669613685.0) <Jog|WPos:-0.705,-3.050,0.000|Bf:0,128|FS:0,0|Ov:100,100,100>
1669613703.386607<< SerialConn(1669613685.0) ok
======ok======
======receiving reply to request relative_jog(1311, pos=XYZABC(x=-0.01, y=0.0, z=0.0, a=None, b=None, c=None),) T_1669613703387141162======
======RECEIVED GRBL REPLY TO REQUEST: b'$j=g91x-0.01f60\n'======
1669613703.3882177<< SerialConn(1669613685.0) ok
======ok======
======receiving reply to request relative_jog(1312, pos=XYZABC(x=-0.01, y=0.0, z=0.0, a=None, b=None, c=None),) T_1669613703388543987======
======RECEIVED GRBL REPLY TO REQUEST: b'$j=g91x-0.01f60\n'======
1669613703.3891356<< SerialConn(1669613685.0) ok
======ok======
======receiving reply to request relative_jog(1313, pos=XYZABC(x=-0.01, y=0.0, z=0.0, a=None, b=None, c=None),) T_1669613703389572859======
======RECEIVED GRBL REPLY TO REQUEST: b'$j=g91x-0.01f60\n'======
$$$  END jog_stop()  120
1669613703.4160016<< SerialConn(1669613685.0) <Jog|WPos:-0.740,-3.050,0.000|Bf:15,128|FS:0,0>
$$$  jog_run()
[... 76 lines cut ...]
1669613703.494512>> $j=g91x0.0034f60
1669613703.497458<< SerialConn(1669613685.0) <Jog|WPos:-0.735,-3.050,0.000|Bf:15,128|FS:0,0>
1669613703.5015671<< SerialConn(1669613685.0) ok
======ok======
======receiving reply to request relative_jog(1331, pos=XYZABC(x=0.0033999999999999994, y=0.0, z=0.0, a=None, b=None, c=None),) T_1669613703501862298======
======RECEIVED GRBL REPLY TO REQUEST: b'$j=g91x0.0034f60\n'======
1669613703.5023794>> $j=g91x0.0036f60
1669613703.5094838<< SerialConn(1669613685.0) ok
======ok======
======receiving reply to request relative_jog(1332, pos=XYZABC(x=0.003599999999999999, y=0.0, z=0.0, a=None, b=None, c=None),) T_1669613703509810815======
======RECEIVED GRBL REPLY TO REQUEST: b'$j=g91x0.0036f60\n'======
1669613703.5113237>> $j=g91x0.0038f60
1669613703.518952<< SerialConn(1669613685.0) ok
======ok======
======receiving reply to request relative_jog(1333, pos=XYZABC(x=0.003799999999999999, y=0.0, z=0.0, a=None, b=None, c=None),) T_1669613703519283009======
======RECEIVED GRBL REPLY TO REQUEST: b'$j=g91x0.0038f60\n'======
1669613703.520938>> $j=g91x0.004f60
1669613703.5228796<< SerialConn(1669613685.0) <Idle|WPos:-0.725,-3.050,0.000|Bf:14,128|FS:0,0>
1669613703.5278277<< SerialConn(1669613685.0) <Run|WPos:-0.725,-3.050,0.000|Bf:15,128|FS:0,0|WCO:0.000,0.000,0.000>
1669613703.5297353<< SerialConn(1669613685.0) error:8
!!error_note!!  error:  RECEIVED ERROR: Command requires idle state  MachineServer()
1669613703.5393167<< SerialConn(1669613685.0) <Idle|WPos:-0.720,-3.050,0.000|Bf:15,128|FS:0,0|Ov:100,100,100>
$$$  MachineServer().jog_stop()
!!error_note!!  error:  jog_stop() join thread FAILED
```

ipsod, 2022-11-28:
and another

```
1669613703.2001262>> $j=g91x-0.01f60
1669613703.2010846>> $j=g91x-0.01f60
1669613703.2020416>> $j=g91x-0.01f60
1669613703.2094696<< SerialConn(1669613685.0) ok
======ok======
======receiving reply to request relative_jog(1302, pos=XYZABC(x=-0.01, y=0.0, z=0.0, a=None, b=None, c=None),) T_1669613703209867227======
======RECEIVED GRBL REPLY TO REQUEST: b'$j=g91x-0.01f60\n'======
1669613703.2107263>> $j=g91x-0.01f60
1669613703.2136009<< SerialConn(1669613685.0) <Jog|WPos:-0.595,-3.050,0.000|Bf:0,128|FS:60,0>
1669613703.2202148<< SerialConn(1669613685.0) ok
======ok======
======receiving reply to request relative_jog(1303, pos=XYZABC(x=-0.01, y=0.0, z=0.0, a=None, b=None, c=None),) T_1669613703220657929======
======RECEIVED GRBL REPLY TO REQUEST: b'$j=g91x-0.01f60\n'======
1669613703.2216742>> $j=g91x-0.01f60
1669613703.2333229<< SerialConn(1669613685.0) ok
======ok======
======receiving reply to request relative_jog(1304, pos=XYZABC(x=-0.01, y=0.0, z=0.0, a=None, b=None, c=None),) T_1669613703233679684======
======RECEIVED GRBL REPLY TO REQUEST: b'$j=g91x-0.01f60\n'======
1669613703.2349503>> $j=g91x-0.01f60
1669613703.2463036<< SerialConn(1669613685.0) ok
======ok======
======receiving reply to request relative_jog(1305, pos=XYZABC(x=-0.01, y=0.0, z=0.0, a=None, b=None, c=None),) T_1669613703246704421======
======RECEIVED GRBL REPLY TO REQUEST: b'$j=g91x-0.01f60\n'======
1669613703.2472463>> $j=g91x-0.01f60
$$$  MachineServer().jog_stop()
1669613703.261019<< SerialConn(1669613685.0) ok
======ok======
======receiving reply to request relative_jog(1306, pos=XYZABC(x=-0.01, y=0.0, z=0.0, a=None, b=None, c=None),) T_1669613703261357043======
======RECEIVED GRBL REPLY TO REQUEST: b'$j=g91x-0.01f60\n'======
1669613703.2620792>> $j=g91x-0.01f60
$$$  send_jog_cancel()
1669613703.2654614<< SerialConn(1669613685.0) <Jog|WPos:-0.615,-3.050,0.000|Bf:0,128|FS:60,0>
1669613703.272241<< SerialConn(1669613685.0) ok
======ok======
======receiving reply to request relative_jog(1307, pos=XYZABC(x=-0.01, y=0.0, z=0.0, a=None, b=None, c=None),) T_1669613703272531638======
======RECEIVED GRBL REPLY TO REQUEST: b'$j=g91x-0.01f60\n'======
1669613703.2868586<< SerialConn(1669613685.0) ok
======ok======
======receiving reply to request relative_jog(1308, pos=XYZABC(x=-0.01, y=0.0, z=0.0, a=None, b=None, c=None),) T_1669613703287225123======
======RECEIVED GRBL REPLY TO REQUEST: b'$j=g91x-0.01f60\n'======
1669613703.292296<< SerialConn(1669613685.0) ok
======ok======
======receiving reply to request relative_jog(1309, pos=XYZABC(x=-0.01, y=0.0, z=0.0, a=None, b=None, c=None),) T_1669613703292680736======
======RECEIVED GRBL REPLY TO REQUEST: b'$j=g91x-0.01f60\n'======
1669613703.2939055>> b'\x85'
1669613703.2972727<< SerialConn(1669613685.0) ok
======ok======
======receiving reply to request relative_jog(1310, pos=XYZABC(x=-0.01, y=0.0, z=0.0, a=None, b=None, c=None),) T_1669613703297612710======
======RECEIVED GRBL REPLY TO REQUEST: b'$j=g91x-0.01f60\n'======
1669613703.316961<< SerialConn(1669613685.0) <Jog|WPos:-0.655,-3.050,0.000|Bf:0,128|FS:0,0|WCO:0.000,0.000,0.000>
1669613703.366934<< SerialConn(1669613685.0) <Jog|WPos:-0.705,-3.050,0.000|Bf:0,128|FS:0,0|Ov:100,100,100>
1669613703.386607<< SerialConn(1669613685.0) ok
======ok======
======receiving reply to request relative_jog(1311, pos=XYZABC(x=-0.01, y=0.0, z=0.0, a=None, b=None, c=None),) T_1669613703387141162======
======RECEIVED GRBL REPLY TO REQUEST: b'$j=g91x-0.01f60\n'======
1669613703.3882177<< SerialConn(1669613685.0) ok
======ok======
======receiving reply to request relative_jog(1312, pos=XYZABC(x=-0.01, y=0.0, z=0.0, a=None, b=None, c=None),) T_1669613703388543987======
======RECEIVED GRBL REPLY TO REQUEST: b'$j=g91x-0.01f60\n'======
1669613703.3891356<< SerialConn(1669613685.0) ok
======ok======
======receiving reply to request relative_jog(1313, pos=XYZABC(x=-0.01, y=0.0, z=0.0, a=None, b=None, c=None),) T_1669613703389572859======
======RECEIVED GRBL REPLY TO REQUEST: b'$j=g91x-0.01f60\n'======
$$$  END jog_stop()  120
1669613703.4160016<< SerialConn(1669613685.0) <Jog|WPos:-0.740,-3.050,0.000|Bf:15,128|FS:0,0>
$$$  jog_run()
$$$  jog_running
[... 81 lines cut ...]
1669613703.5023794>> $j=g91x0.0036f60
1669613703.5094838<< SerialConn(1669613685.0) ok
======ok======
======receiving reply to request relative_jog(1332, pos=XYZABC(x=0.003599999999999999, y=0.0, z=0.0, a=None, b=None, c=None),) T_1669613703509810815======
======RECEIVED GRBL REPLY TO REQUEST: b'$j=g91x0.0036f60\n'======
1669613703.5113237>> $j=g91x0.0038f60
1669613703.518952<< SerialConn(1669613685.0) ok
======ok======
======receiving reply to request relative_jog(1333, pos=XYZABC(x=0.003799999999999999, y=0.0, z=0.0, a=None, b=None, c=None),) T_1669613703519283009======
======RECEIVED GRBL REPLY TO REQUEST: b'$j=g91x0.0038f60\n'======
1669613703.520938>> $j=g91x0.004f60
1669613703.5228796<< SerialConn(1669613685.0) <Idle|WPos:-0.725,-3.050,0.000|Bf:14,128|FS:0,0>
1669613703.5278277<< SerialConn(1669613685.0) <Run|WPos:-0.725,-3.050,0.000|Bf:15,128|FS:0,0|WCO:0.000,0.000,0.000>
1669613703.5297353<< SerialConn(1669613685.0) error:8
!!error_note!!  error:  RECEIVED ERROR: Command requires idle state  MachineServer()
1669613703.5393167<< SerialConn(1669613685.0) <Idle|WPos:-0.720,-3.050,0.000|Bf:15,128|FS:0,0|Ov:100,100,100>
$$$  MachineServer().jog_stop()
!!error_note!!  error:  jog_stop() join thread FAILED
$$$  send_jog_cancel()
1669613704.6358695>> b'\x85'
$$$  END jog_stop()  14
$$$  jog_run()
$$$  MachineServer().jog_stop()
$$$  send_jog_cancel()
```

bdring, 2022-11-28:
TLDR;

Can you make the most simple example to repeat the problem?

Are you 100% sure you are waiting for an `ok` before sending a jog command? I notice `Bf:0` in a few places.

ipsod, 2022-11-28:
I'll try to make a simple example.

I'm not 100% sure, but I don't know see how it could be otherwise.  All commands sent go through a function that waits for the buffer to have room, and buffer availability is only restored after "ok" is received.  Everything looks ok on my end - I use 120 as the buffer size, and buffer availability never drops below 0.  Perhaps if some "ok" messages get sent out of order, it could mess up my count (ie: receive `ok` to a message that has 35 characters, but the message FluidNC meant to `ok` only had 15 characters).  Does `0x85` jog cancel do anything funny regarding the buffer and sending `ok` messages?

ipsod, 2022-11-28:
Does `0x85` jog cancel do anything funny regarding the buffer and sending `ok` messages?

I typically queue up a serial-buffer-full of commands.  Before those are planned, while there just in the serial buffer, I may then send a `0x85` jog cancel.  What would happen to those jog commands in the serial buffer?  Am I doing it wrong?

MitchBradley, 2022-11-28:
The jog cancel character "jumps the queue", bypassing any partial line that has already advanced from the hardware FIFO to the line collection buffer.

You really should send jogs synchronously instead of queuing them up - and you should not send other, non-jog, commands until the job has been completed or canceled.

MitchBradley, 2022-11-28:
jog cancel has no relation whatsoever to "ok".  ok indicates that a complete line has been received, passed to the GCode parser, and interpreted without error.  The result of that interpretation causes the indicated operation - perhaps motion - to be passed on to the planner, where it is aggregated with any previous motion.

The jog cancel character is recognized anywhere, even in the middle of a line, before the line is handed over to the GCode parser.  It sends an event to the protocol state machine that results in the cancellation of motion that is already in progress - motion that is typically well past the planner queue.  The feedback to the sender is in the form of status reports that show state changes, completely independent of and separate from the "ok" acknowledgement of the line-oriented GCode parser.

ipsod, 2022-11-29:
Would this list of timestamps (in milliseconds) and commands sent suffice to try and reproduce the error?

```
0001104424, b'\x18'
0001104728, b'\x0c'
0002111029, b'$ri=50\n'
0002122378, b'$Alarm/Disable\n'
0002133496, b'G10L20 X0Y0Z0\n'
0002225694, b'$10=2\n'
0002234257, b'\n$Bye\n'
0003851715, b'$T\n'
0003894207, b'\x85'
0005971033, b'\x17'
0005971358, b'\x0c'
0005975080, b'\x17'
0005975287, b'\x17'
0006076116, b'\x0c'
0006377325, b'\x17'
0006482553, b'\x17'
0006482840, b'\x0c'
0007488691, b'\x18'
0007488913, b'\x0c'
0008497874, b'$ri=50\n'
0008508713, b'$Alarm/Disable\n'
0008527685, b'G10L20 X0Y0Z0\n'
0008541794, b'$10=2\n'
0008549644, b'$T\n'
0008559413, b'$Commands/List\n'
0008719920, b'$Settings/List\n'
0008774405, b'$Settings/ListChanged\n'
0008787879, b'$Alarms/List\n'
0008822019, b'$Errors/List\n'
0009009464, b'$Config/Dump\n'
0009422404, b'$Alarm/Disable\n'
0009430830, b'$j=g91x0.0002y0.00333z0.00333f2829.06345\n'
0009442155, b'$j=g91x0.0004y0.00667z0.00667f2829.06345\n'
0009452593, b'$j=g91x0.0006y0.01z0.01f2829.06345\n'
0009462160, b'$j=g91x0.0008y0.01333z0.01333f2829.06345\n'
0009472766, b'$j=g91x0.001y0.01667z0.01667f2829.06345\n'
0009483306, b'$j=g91x0.0012y0.02z0.02f2829.06345\n'
0009492657, b'$j=g91x0.0014y0.02333z0.02333f2829.06345\n'
0009503597, b'$j=g91x0.0016y0.02667z0.02667f2829.06345\n'
0009512853, b'$j=g91x0.0018y0.03z0.03f2829.06345\n'
0009522771, b'$j=g91x0.002y0.03333z0.03333f2829.06345\n'
0009532953, b'$j=g91x0.0022y0.03667z0.03667f2829.06345\n'
0009543331, b'$j=g91x0.0024y0.04z0.04f2829.06345\n'
0009553657, b'$j=g91x0.0026y0.04333z0.04333f2829.06345\n'
0009566061, b'$j=g91x0.0028y0.04667z0.04667f2829.06345\n'
0009577291, b'$j=g91x0.003y0.05z0.05f2829.06345\n'
0009586751, b'$j=g91x0.0032y0.05333z0.05333f2829.06345\n'
0009598614, b'$j=g91x0.0034y0.05667z0.05667f2829.06345\n'
0009610146, b'$j=g91x0.0036y0.06z0.06f2829.06345\n'
0009620304, b'$j=g91x0.0038y0.06333z0.06333f2829.06345\n'
0009631838, b'$j=g91x0.004y0.06667z0.06667f2829.06345\n'
0009643195, b'$j=g91x0.0042y0.07z0.07f2829.06345\n'
0009653522, b'$j=g91x0.0044y0.07333z0.07333f2829.06345\n'
0009666049, b'$j=g91x0.0046y0.07667z0.07667f2829.06345\n'
0009675647, b'$j=g91x0.0048y0.08z0.08f2829.06345\n'
0009684830, b'$j=g91x0.005y0.08333z0.08333f2829.06345\n'
0009696591, b'$j=g91x0.0052y0.08667z0.08667f2829.06345\n'
0009708508, b'$j=g91x0.0054y0.09z0.09f2829.06345\n'
0009720386, b'$j=g91x0.0056y0.09333z0.09333f2829.06345\n'
0009732658, b'$j=g91x0.0058y0.09667z0.09667f2829.06345\n'
0009744235, b'$j=g91x0.006y0.1z0.1f2829.06345\n'
0009753664, b'$j=g91x0.0062y0.10333z0.10333f2829.06345\n'
0009766080, b'$j=g91x0.0064y0.10667z0.10667f2829.06345\n'
0009777698, b'$j=g91x0.0066y0.11z0.11f2829.06345\n'
0009787161, b'$j=g91x0.0068y0.11333z0.11333f2829.06345\n'
0009798596, b'$j=g91x0.007y0.11667z0.11667f2829.06345\n'
0009810307, b'$j=g91x0.0072y0.12z0.12f2829.06345\n'
0009820800, b'$j=g91x0.0074y0.12333z0.12333f2829.06345\n'
0009832646, b'$j=g91x0.0076y0.12667z0.12667f2829.06345\n'
0009863908, b'\x85'
0009868098, b'$j=g91x0.0002y0.00333z0.00333f2829.06345\n'
0009930934, b'$j=g91x0.0004y0.00667z0.00667f2829.06345\n'
0009941558, b'$j=g91x0.0006y0.01z0.01f2829.06345\n'
0009951801, b'$j=g91x0.0008y0.01333z0.01333f2829.06345\n'
0009962073, b'$j=g91x0.001y0.01667z0.01667f2829.06345\n'
0009971877, b'$j=g91x0.0012y0.02z0.02f2829.06345\n'
0009982339, b'$j=g91x0.0014y0.02333z0.02333f2829.06345\n'
0009994157, b'$j=g91x0.0016y0.02667z0.02667f2829.06345\n'
0010006050, b'$j=g91x0.0018y0.03z0.03f2829.06345\n'
0010017547, b'$j=g91x0.002y0.03333z0.03333f2829.06345\n'
0010027719, b'$j=g91x0.0022y0.03667z0.03667f2829.06345\n'
0010039530, b'$j=g91x0.0024y0.04z0.04f2829.06345\n'
```

ipsod, 2022-11-29:
@MitchBradley Ok, I'll try and modify my jog code to comply with that.  So far, it seems to not work great - it is jerky, as though it's not keeping the buffer full enough, though I believe I'm using the GRBL1.1 jogging formula as is described in their docs here: https://github.com/gnea/grbl/wiki/Grbl-v1.1-Jogging.  It's also causing the error described above (with the commands pasted above)

ipsod, 2022-11-29:
> TLDR;
> 
> Can you make the most simple example to repeat the problem?
> 
> Are you 100% sure you are waiting for an `ok` before sending a jog command? I notice `Bf:0` in a few places.

I've altered my sender to wait for "ok" before sending each subsequent command, and no longer get `Bf:0` in my status, but still get the error described in the issue.  Here's a cleaned up log that's easier to see what's happening in, and it shows that each command waits for a response.

```
======CONNECTED=====
0001106344>> 
0001106532>> 
======starting interface=====
0001116469<<   Grbl 3.6 [FluidNC v3.6.5-pre2 (wifi) '$' for help]
======LINES_NOT_MATCHED=====

===========
0002112660>> $ri=50
0002122445<<   [MSG:INFO: Channel auto report interval set to 50 ms]
0002122928<<   ok
======receiving reply to request mode_transmit_frequency T_1669684359178130977=====
======RECEIVED GRBL REPLY TO REQUEST: b'$ri=50\n'=====

0002125212>> $Alarm/Disable
0002129914<<   <Idle|WPos:0.000,-0.150,0.000|Bf:15,128|FS:0,0|WCO:0.165,2.690,2.690>
0002133979<<   [GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
0002134419<<   ok
======receiving reply to request alarm_disable T_1669684359189576872=====
======RECEIVED GRBL REPLY TO REQUEST: b'$Alarm/Disable\n'=====

0002136562>> G10L20 X0Y0Z0
0002149457<<   ok
======receiving reply to request set_wpos_zero T_1669684359204727340=====
======RECEIVED GRBL REPLY TO REQUEST: b'G10L20 X0Y0Z0\n'=====

0002151178>> $10=2
0002156420<<   <Idle|WPos:0.000,0.000,0.000|Bf:15,128|FS:0,0|WCO:0.165,2.540,2.690>
0002156722<<   ok
======receiving reply to request mode_transmit_wpos T_1669684359211881629=====
======RECEIVED GRBL REPLY TO REQUEST: b'$10=2\n'=====

======test_reboot_mcu()=====
$$$  REBOOT_MCU
0002159639>> 
$Bye
0002166789<<   ok
======LINES_NOT_MATCHED=====

======reset=====
0002168093<<   [MSG:INFO: Restarting]
0002168422<<   ok
======LINES_NOT_MATCHED=====

======reset=====
0002170577<<   ets Jun  8 2016 00:22:57
0002175986<<   rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
0002178562<<   configsip: 0, SPIWP:0xee
0002184421<<   clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
0002186799<<   mode:DIO, clock div:1
0002189499<<   load:0x3fff0030,len:1184
0002192210<<   load:0x40078000,len:12812
0002193296<<   load:0x40080400,len:3032
0002195501<<   entry 0x400805e4
0002725144<<   [MSG:INFO: FluidNC v3.6.5-pre2]
0002730234<<   [MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
0002744153<<   [MSG:INFO: Local filesystem type is spiffs]
0002748597<<   [MSG:INFO: Configuration file:config.yaml]
0002753797<<   [MSG:INFO: Machine ESP32 Dev Controller V4]
0002756608<<   [MSG:INFO: Board ESP32 Dev Controller V4]
0002759526<<   [MSG:INFO: SPI not defined]
[... 812 lines cut ...]
0009956760<<   ok
======receiving reply to request relative_jog(75, pos=XYZABC(x=0.0019999999999999996, y=0.03333333333333333, z=0.03333333333333333, a=None, b=None, c=None),) T_1669684367011973215=====
======RECEIVED GRBL REPLY TO REQUEST: b'$j=g91x0.002y0.03333z0.03333f2829.06345\n'=====

0009957454>> $j=g91x0.0022y0.03667z0.03667f2829.06345
0009968148<<   ok
======receiving reply to request relative_jog(76, pos=XYZABC(x=0.0021999999999999997, y=0.03666666666666667, z=0.03666666666666667, a=None, b=None, c=None),) T_1669684367023363344=====
======RECEIVED GRBL REPLY TO REQUEST: b'$j=g91x0.0022y0.03667z0.03667f2829.06345\n'=====

0009968996>> $j=g91x0.0024y0.04z0.04f2829.06345
0009971107<<   <Idle|WPos:0.160,2.655,2.655|Bf:14,128|FS:0,0>
0009977878<<   <Run|WPos:0.160,2.655,2.655|Bf:15,128|FS:0,0>
0009978295<<   error:8
!!error_note!!  error:  RECEIVED ERROR: Command requires idle state  MachineServer()

0009987322<<   <Idle|WPos:0.165,2.690,2.690|Bf:15,128|FS:0,0>
$$$  MachineServer().jog_stop()
$$$  send_jog_cancel()
$$$  SHUTTING DOWN: MachineServer
$$$  MachineServer().jog_stop()
$$$  send_jog_cancel()
$$$  stop_track_status()
$$$  stop_read_loop()
$$$  SerialConn(1669684363.1).disconnect()
```

MitchBradley, 2022-11-30:
The first Run message appears when the X value finally changes from 0.000 to 0.005.  In the preceding jog commands, the X offset is smaller - sometimes much smaller - than the size of a machine step (.005).  I will keep investigating, but I do not see the point of issuing jogs for less than a step size.

ipsod, 2022-11-30:
> The first Run message appears when the X value finally changes from 0.000 to 0.005. In the preceding jog commands, the X offset is smaller - sometimes much smaller - than the size of a machine step [...]

Well, the intent is to implement GRBL1.1 jogging with independent speeds on multiple axes, and these really short motions happen when a faster axis is running along with a slower axis.  I'm not sure what sort of checks/whatever to implement to avoid that - I guess that all of the shorter motions will be grouped at the beginning/end of a motion, anyhow, so maybe I could just omit them without worrying about how it might effect the rest of the motion.

MitchBradley, 2022-11-30:
200 steps/mm = 0.005 mm/step

ipsod, 2022-11-30:
I see.  Experimentation shows that 0.0025 + 0.0025 = 0.005, so those short motions aren't entirely ignored.

Also, just issuing 2 `$j=g91x0.0025f1000` commands in a row does not trigger the error.

```
<Idle|WPos:0.195,2.705,2.705|Bf:15,128|FS:0,0>
$j=g91x0.0025f1000
ok
$j=g91x0.0025f1000
ok
<Jog|WPos:0.195,2.705,2.705|Bf:15,128|FS:0,0>
<Idle|WPos:0.200,2.705,2.705|Bf:15,128|FS:0,0>
$j=g91x0.0005f1000
ok
$j=g91x0.0005f1000
ok
$j=g91x0.0005f1000
ok
$j=g91x0.0005f1000
ok
$j=g91x0.0005f1000
ok
$j=g91x0.0005f1000
ok
$j=g91x0.0005f1000
ok
$j=g91x0.0005f1000
ok
<Jog|WPos:0.200,2.705,2.705|Bf:15,128|FS:0,0>
<Idle|WPos:0.205,2.705,2.705|Bf:15,128|FS:0,0>
```

Please let me know if there's anything further I can do to move this along.

MitchBradley, 2022-12-01:
It is a multi-core synchronization problem.  I am working on a fix.

ipsod, 2022-12-01:
Great, thank you.  Good hunting.

closed 2023-01-12 by bdring