bdring/FluidNC #727: Feature: UART for TFT24

ColPaulR, 2022-11-28:
### Machine Context

I'm using a MKS TinyBee board. GPIO.16 and GPIO.17 are Rx and Tx respectively. MKS TFT24 sends and receives g-code. It would be great to implement another UART. I'm getting familiar with the source code. It looks like it is a copy of uart.cpp with different pin assignments in Uart::begin. Am I on the right track here?

### Feature Description

Want to add another UART to Fluidnc to work with TFT touch display g-code terminal, in addition to USB and Wifi uarts.

### Other Approaches

NA

### How I Can Help

NA

MitchBradley, 2022-11-28:
Copy / edit is almost never the right track.

```
Uart MyUart(1, true);   \\ Hardware UART block 1, true for LF->CR-LF conversion
MyUart.setPins(17, 16);
MyUart.begin(115200, UartData::Bits8, UartStop::Bits1, UartParity::None);
```
This might conflict with UART usage by VFD Spindles or Trinamic UART drivers, but I doubt that you would have those with a Tinybee board.

ColPaulR, 2022-11-29:
Thanks for the pointer. I'll keep learning. I don't have Trinamic drivers, but I do (or will have) a Huanyang VFD. Having issues with the RS-485 but had the same issues with GRBL32 on xPro V5. May have to settle with 0-10V on FluidNC just like I did with xPro. When you say "conflict" is that a clash with the Uart code or possible contention with IO pins? I have the RS-485 on GPIO.13 and GPIO.4.

MitchBradley, 2022-11-29:
There are only 3 UART hardware blocks in total.  Since you don't need one for Trinamic drivers, you can use one for the main USB connection, one for your device, and (if needed), one for VFD.  If you say `Uart MyUart(-1, true)`, the -1 will auto-assign the hardware UART block number so each device gets a different one.

ColPaulR, 2022-11-29:
Understood. I tried with uart_num set to 1 and -1. No success yet, but it could be the wiring or the TFT configuration (or the idiot behind the keyboard). I'll troubleshoot tonight to see if I can ensure that the FluidNC change are not the source of my problems.

ColPaulR, 2022-11-30:
Found the issue (and created a couple more). I was not adding the new uart to allchannels. Once I registered the uart (and moved the declaration out of the try scope), everything starting working.

I've got the Uart MyUart(-1, true) declared in main.cpp outside of the setup function so that the reference doesn't get deallocated when the function ends. This seems contrary to the spirit of run-time configurable machine definitions as it claims 1 uart regardless of if the machine needs it. As you mentioned, that would cause a problem when both the non-USB UARTs are needed. Can you point me to where I need to start working to make this an item in the yaml file? Ideally, I would have a section entry with uart items for pin definitions and protocol params, like spindle does. I'm envisioning something like the below.

TFTuart:
    txd_pin: gpio.17
    rxd_pin: gpio.16
    rts_pin: NO_PIN
    its_pin: NO_PIN
    baud: 112000
    mode: 8N1

MitchBradley, 2022-11-30:
Search the source for some top-level section names and then continue searching and studying.  I don't want this to turn into a never-ending stream of "where do I find blah and how does it work".  Search, read, study, repeat is the only way to learn a code base.

closed 2022-11-30 by MitchBradley

justdue, 2022-12-23:
Hello @ColPaulR any news on this side? We are seeking for almost the same thing, not the display MKS TFT24, but another display with another esp to communicate with serial uart 1. Didi you manage to maker it work?

ColPaulR, 2022-12-23:
I did get it working but it the code is precompiled vs. runtime configurable thought YAML file. Take a look at https://github.com/ColPaulR/FluidNC. I think Main.cpp is the only change in my fork from the main code.

justdue, 2022-12-23:
Great, thanks. Will test. We plan to use this kind of display: https://pt.aliexpress.com/item/1005004936738935.html

FrankSchuby, 2024-01-12:
Hello Paul, I have successfully compiled your code with PlatformIO. I don't know anything about C or C++, 
I program my Atmels with Pascal.
Am I correct that everything that is sent to FluidNC is sent to the serial output Pin17 Pin16?

ColPaulR, 2024-01-12:
@FrankSchuby, it has been a while since I looked at this, but I don't think so. My edits just allowed another UART channel that was independent of all the other channels. Regardless, this is against a very old version of FluidNC. I think FluidNC has moved UART control into the YAML vs. compiled configurations.

FrankSchuby, 2024-01-12:
>>I think FluidNC has moved UART control into the YAML vs. compiled configurations.<<

Hello Paul, yes I have read a lot about that. At Wiki you can also see how the uart is integrated into config.yaml. What I don't understand is how I send commands to the uart (Pin17/Pin16) via FluidNC.
I'm currently programming my own GCode transmitter, that's looking good.
But how do I send data to the uart?

MitchBradley, 2024-01-12:
https://github.com/bdring/PendantsForFluidNC has many examples of how to connect external devices to FluidNC via UART.  There is a library "GrblParserC" that makes it easy.

What "commands" do you need to send?  The normal usage is that the external device acts like a conventional "GCode sender" program, sending commands to FluidNC and receiving status reports back.  GCode does not have any facilities for "sending commands" to external devices.

Herb01, 2024-03-04:
Tried to add a third uart with an uart-channel to work with a pendant. I can't get it working. Uart 2 is used to control my vfd but as soon as i try to configure the uart-channel the vfd is not working anymore. The uart-channel is on uart3. How can i get this to work? My board is a MKS tinybee and except uart3 everything works very good.

MitchBradley, 2024-03-04:
ESP32 has 3 UARTs.  uart0 is for the USB serial/console/sender channel.  That leaves uart1 and uart2 for other uses.  If you need more than that, you are out of luck.  Sorry.  There is only so much blood in this turnip.