bdring/FluidNC #730: Problem: WEBUI. In Tablet mode, Pause button is inactive

SnowBeaver, 2022-11-29:
### Controller Board

6-pack

### Machine Description

Gantry router

### Configuration file

```yaml
board: 6 Pack
name: 3 axis std_step X--ZYY v1.2
stepping:
  engine: I2S_STATIC
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 53.400
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 1050.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: -3.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 400.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.33:low:pu
      hard_limits: false
      pulloff_mm: 3.000
      tmc_2130:
        cs_pin: i2so.3
        r_sense_ohms: 0.110
        run_amps: 0.500
        hold_amps: 0.500
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0

  y:
    steps_per_mm: 53.400
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 960.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: -3.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 400.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.36:low
      hard_limits: false
      pulloff_mm: 3.000
      tmc_2130:
        cs_pin: i2so.19
        r_sense_ohms: 0.110
        run_amps: 0.500
        hold_amps: 0.500
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        step_pin: I2SO.18
        direction_pin: I2SO.17
        disable_pin: I2SO.16

    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.39:low
      hard_limits: false
      pulloff_mm: 3.000
      tmc_2130:
        cs_pin: i2so.22
        r_sense_ohms: 0.110
        run_amps: 0.500
        hold_amps: 0.500
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        step_pin: I2SO.21
        direction_pin: I2SO.20
        disable_pin: I2SO.23

  z:
    steps_per_mm: 808.000
    max_rate_mm_per_min: 500.000
    acceleration_mm_per_sec2: 50.000
    max_travel_mm: 360.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: -3.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 400.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.35:low
      hard_limits: false
      pulloff_mm: 3.000
      tmc_2130:
        cs_pin: i2so.14
        r_sense_ohms: 0.110
        run_amps: 0.6500
        hold_amps: 0.6500
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        step_pin: I2SO.13
        direction_pin: I2SO.12
        disable_pin: I2SO.15

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

probe:
  pin: gpio.2:high:pu
  check_mode_start: true

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

start:
  must_home: false
  
relay:
  output_pin: gpio.26
  enable_pin: NO_PIN
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0.000% 1=100.000%

laser:
  pwm_hz: 2000
  output_pin: gpio.13:low
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  tool_num: 100
  speed_map: 0=0.000% 1000=100.000%
  off_on_alarm: false
```

### Startup Messages

```text
Not required
```

### What happened?

When job run, attempts to pause the job using the Pause button In the Tablet panel does nothing. The Pause button the Dashboard panel does pause. Going back to the Tablet panel, job can be stopped using the Stop button

MitchBradley, 2022-12-02:
I fixed that last month but somehow the updated index.html.gz did not make it into -pre4 .  Release engineering is a very tedious business.

In the next release we will make sure that the correct index.html.gz is included.

SnowBeaver, 2022-12-14:
Pause button in tablet mode does not work. Another user mentioned it earlier this week. I reopened the issue to make sure it gets noticed.

MitchBradley, 2022-12-14:
It got noticed, but the time I spent replying to this reminder increased the time before it is deployed.

closed 2024-12-12 by bdring