bdring/FluidNC #732: Problem:

txg5230, 2022-12-01:
### Controller Board

6-pack

### Machine Description

Esp32 development board

### Configuration file

```yaml
board: 6 Pack
name: 6 Pack StepStick XYZABC
stepping:
  engine: I2S_STREAM
  idle_ms: 250
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 320.000
    max_rate_mm_per_min: 20000.000
    acceleration_mm_per_sec2: 600.000
    max_travel_mm: 1250.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.14:low:pu
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      stepstick:
       
        step_pin: I2SO.6
        direction_pin: I2SO.5
    

  y:
    steps_per_mm: 320.000
    max_rate_mm_per_min: 20000.000
    acceleration_mm_per_sec2: 600.000
    max_travel_mm: 2450.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.12:low:pu
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      stepstick:
        
        step_pin: I2SO.4
        direction_pin: I2SO.3
        

  z:
    steps_per_mm: 320.000
    max_rate_mm_per_min: 15000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.13:low:pu
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      stepstick:
        
        step_pin: I2SO.2
        direction_pin: I2SO.1
        
        
  a:
    steps_per_mm: 320
    max_rate_mm_per_min: 15000.000
    acceleration_mm_per_sec2: 600.000
    max_travel_mm: 960.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.15:low:pu
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      stepstick:
       
        step_pin: I2SO.0
        direction_pin: I2SO.7
     

        

i2so:
  bck_pin: gpio.17
  data_pin: gpio.4
  ws_pin: gpio.16

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

control:
  safety_door_pin: gpio.33:low:pu  
  feed_hold_pin: gpio.25:low:pu
  cycle_start_pin: gpio.26:low:pu
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

probe:
  pin: gpio.27:low:pu
  check_mode_start: true

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:

planner_blocks: 99

start:
  must_home: false

PWM:
  pwm_hz: 5000
  output_pin: gpio.2
  enable_pin: gpio.32
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 1000
  spindown_ms: 1000
  tool_num: 0
  speed_map: 0=0.000% 1000=100.000%
```

### Startup Messages

```text
[12:09:55.070]收←◆
[MSG:INFO: FluidNC v3.6.5-pre4]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:4595.yaml]
[MSG:INFO: Machine 6 Pack StepStick XYZABC]
[MSG:INFO: Board 6 Pack]
[MSG:INFO: I2SO BCK:gpio.17 WS:gpio.16 DATA:gpio.4]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 4]
[MSG:INFO: Axis X (150.000,1400.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.6 Dir:I2SO.5 Disable:NO_PIN]
[MSG:INFO:  X Neg Limit gpio.14:low:pu]
[MSG:INFO: Axis Y (150.000,2600.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.4 Dir:I2SO.3 Disable:NO_PIN]
[MSG:INFO:  Y Neg Limit gpio.12:low:pu]
[MSG:INFO: Axis Z (-150.000,150.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.2 Dir:I2SO.1 Disable:NO_PIN]
[MSG:INFO:  Z Neg Limit gpio.13:low:pu]
[MSG:INFO: Axis A (150.000,1110.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.0 Dir:I2SO.7 Disable:NO_PIN]
[MSG:INFO:  A Neg Limit gpio.15:low:pu]
[MSG:INFO: safety_door_pin gpio.33:low:pu]
[MSG:INFO: feed_hold_pin gpio.25:low:pu]
[MSG:INFO: cycle_start_pin gpio.26:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:gpio.32 Out:gpio.2 Dir:NO_PIN Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Probe Pin: gpio.27:low:pu]

[12:09:55.334]收←◆[MSG:INFO: Connecting to STA SSID:yygg]
[MSG:INFO: Connecting.]

[12:09:57.334]收←◆[MSG:INFO: Connecting..]

[12:09:59.343]收←◆[MSG:INFO: Connected - IP is 192.168.2.8]
[MSG:INFO: WiFi on]

[12:09:59.368]收←◆[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.6 [FluidNC v3.6.5-pre4 (wifi) '$' for help]
```

### User Interface Software

webul

### What happened?

Upload a file larger than 2 m on the webpage, the controller restarts, and cannot upload it.

MitchBradley, 2022-12-01:
Are you uploading to an SD card or to the local FLASH filesystem?

txg5230, 2022-12-01:
[12:30:54.991]发→◇[ESP210]
□
[12:30:55.070]收←◆[FILE: pvc.nc|SIZE:357732]
[FILE: 3.nc|SIZE:9272]
[FILE: 2.nc|SIZE:5524]
[FILE: 1.nc|SIZE:23896]
[FILE: t4x2.nc|SIZE:10691429]
[FILE: 鍏呯數鐩掑瓙寮€瀛?.nc|SIZE:805]
[DIR:dkj]
[FILE:  gd.nc|SIZE:57]
[FILE:  fd.nc|SIZE:41]
[FILE:  dk1.nc|SIZE:5]
[FILE:  dk2.nc|SIZE:5]
[FILE: ay.nc|SIZE:12406820]
[FILE: 1546.nc|SIZE:23]
[FILE: 5.nc|SIZE:253695]
[FILE: 6.nc|SIZE:86940660]
[FILE: 2240.nc|SIZE:23]
[FILE: fluidnc-v3.6.2-win64.zip|SIZE:13192021]
[FILE: 娴嬭瘯.nc|SIZE:0]

[12:30:55.179]收←◆[DIR:System Volume Information]

[12:30:55.259]收←◆[FILE:  WPSettings.dat|SIZE:12]

[12:30:55.342]收←◆[FILE:  IndexerVolumeGuid|SIZE:
[12:30:55.367]收←◆76]

[12:30:55.424]收←◆[FILE: 4.nc|SIZE:
[12:30:55.450]收←◆1337530]

[12:30:55.505]收←◆[FILE: NC80120.NC|SIZE:0]

[12:30:55.584]收←◆[FILE: 3560.nc|SIZE:23]

[12:30:55.665]收←◆[FILE: 1530.nc|SIZE:23]

[12:30:55.745]收←◆[FILE: 2250.nc|SIZE:
[12:30:55.771]收←◆23]

[12:30:55.825]收←◆[FILE: 2530.nc|SIZE:
[12:30:55.851]收←◆23]

[12:30:55.880]收←◆[/sd/ Free:1.66 GB Used:183.38 MB Total:1.84 GB]
ok
ok

[12:35:15.288]收←◆[MSG:INFO: Upload failed - file write failed]
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400f9262  PS      : 0x00060630  A0      : 0x800faab4  A1      : 0x3ffb1f90  
A2      : 0xff000000  A3      : 0x3ffb1fec  A4      : 0x00000000  A5      : 0x3ffcca38  
A6      : 0x00060620  A7      : 0x00000001  A8      : 0xff000100  A9      : 0x3ffb1f70  
A10     : 0x3ffb200c  A11     : 0x3ffb200c  A12     : 0x00000000  A13     : 0x0000000c  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000015  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xff00014c  LBEG    : 0x4008c8a9  LEND    : 0x4008c8b9  LCOUNT  : 0xfffffffe  

Backtrace:0x400f925f:0x3ffb1f900x400faab1:0x3ffb1fb0 0x400fd505:0x3ffb2040 0x400fd8f3:0x3ffb2170 0x400fda3a:0x3ffb21e0 0x401b926d:0x3ffb2200 0x40109806:0x3ffb2220 0x4010d746:0x3ffb2240 0x4010aa72:0x3ffb2280 0x4010b6e1:0x3ffb22c0 0x4010c436:0x3ffb23c0 0x4010d9a9:0x3ffb24c0 0x400f9c79:0x3ffb2510 0x4010279e:0x3ffb2560 0x400ef211:0x3ffb2580 0x400edb53:0x3ffb25a0 0x400e1de9:0x3ffb26e0 0x4011577d:0x3ffb2820 

ELF file SHA256: 0000000000000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12812
load:0x40080400,len:3032
entry 0x400805e4

[12:35:15.922]收←◆
[MSG:INFO: FluidNC v3.6.5-pre4]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:ERR: Skipping configuration file due to panic]
[MSG:INFO: Using default configuration]
[MSG:INFO: Axes: using defaults]
[MSG:INFO: Machine Default (Test Drive)]
[MSG:INFO: Board None]
[MSG:INFO: SPI not defined]
[MSG:INFO: No SD Card CS Pin]
[MSG:INFO: See http://wiki.fluidnc.com/en/config/sd_card#sdfallbackcs-access-sd-without-a-config-file]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-1000.000,0.000)]
[MSG:INFO: Axis Y (-1000.000,0.000)]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]

[12:35:16.107]收←◆[MSG:INFO: Connecting to STA SSID:yygg]
[MSG:INFO: Connecting.]

[12:35:18.107]收←◆[MSG:INFO: Connecting..]

[12:35:20.109]收←◆[MSG:INFO: Connected - IP is 192.168.2.8]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.6 [FluidNC v3.6.5-pre4 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]

txg5230, 2022-12-01:
sd卡

MitchBradley, 2022-12-01:
I cannot reproduce this problem.  I did the following on a 6-pack:
* Ran install-wifi.bat from fluidnc-v3.6.5-pre4-win64
* Ran install-fs.bat from fluidnc-v3.6.5-pre4-win64
* Uploaded your config file to the local filesystem and set $config/filename to that file
* Reset to use your config file
* Started WebUI
* Used WebUI to successfully upload several files in the 2 to 12 MB range, for example

[image]

[image]

Your log shows the following message:

> [12:35:15.288]收←◆[MSG:INFO: Upload failed - file write failed]

That indicates that something went wrong when FluidNC was trying to write the data to the SD card.  It is not possible to determine exactly what went wrong because the problem was detected deep inside a library routine.  To diagnose it further would require recompiling the library with a different debug level.  I wonder if your SD card might have a corrupted filesystem.

After the file write message there is this:

> Guru Meditation Error: Core 1 panic'ed (LoadProhibited). Exception was unhandled.
> ...

The following backtrace shows that something went wrong when FluidNC was trying to send a message back to WebUI.  It is as if the websocket connection went away.  I'm not sure why that happened.

MitchBradley, 2022-12-01:
If you can reproduce the problem, try sending "$message/level=debug".  That will give additional information about websocket connections and disconnections.  Perhaps that will help us discover why the crash occurs.

commit 6a4301f5 (2022-12-08): Merge pull request #733 from bdring/Fix732

Fix #732 - Crash when a WebUI upload fails
commit 8e8e0d3e (2022-12-08): Fix #732 - Crash when a WebUI upload fails
closed 2022-12-08 by bdring