bdring/FluidNC #736: Problem: SD Card not work.

zamby, 2022-12-05:
### Controller Board

WEMOS ESP32 LOLIN

### Machine Description

I am only  testwith  board ESP32.

### Configuration file

```yaml
name: "ESP32 Dev Controller V4"
board: "ESP32 Dev Controller V4"

stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 1
  pulse_us: 2
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.13:low
  
  x:
    steps_per_mm: 40
    max_rate_mm_per_min: 500
    acceleration_mm_per_sec2: 5
    max_travel_mm: 1000
    homing:
      cycle: 2
      mpos_mm: 10
      positive_direction: false
    
    motor0:
      limit_all_pin: gpio.17:low:pu
      stepstick:
        direction_pin: gpio.14
        step_pin: gpio.12
    motor1:
      null_motor:

  y:
    steps_per_mm: 800
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1000
    homing:
      cycle: 2
      mpos_mm: 10
      positive_direction: false

    motor0:
      limit_all_pin: gpio.4:low:pu
      stepstick:
        direction_pin: gpio.15
        step_pin: gpio.26
    motor1:
      null_motor:

  z:
    steps_per_mm: 800
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1000
    homing:
      cycle: 1
      mpos_mm: 10
      positive_direction: true

    motor0:
      limit_all_pin: gpio.16:low:pu
      stepstick:
        direction_pin: gpio.33
        step_pin: gpio.27
    motor1:
      null_motor:

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

coolant:
  flood_pin: gpio.25
  mist_pin:  gpio.21

        
probe:
  pin: gpio.32:low:pu

PWM:
  pwm_hz: 5000
  output_pin: gpio.2
  enable_pin: gpio.22
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0% 10000=100%
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.6.4]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine ESP32 Dev Controller V4]
[MSG:INFO: Board ESP32 Dev Controller V4]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.13:low]
[MSG:INFO: Axis X (10.000,1010.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.12 Dir:gpio.14 Disable:NO_PIN]
[MSG:INFO:  X All Limit gpio.17:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (10.000,1010.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.26 Dir:gpio.15 Disable:NO_PIN]
[MSG:INFO:  Y All Limit gpio.4:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-990.000,10.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.27 Dir:gpio.33 Disable:NO_PIN]
[MSG:INFO:  Z All Limit gpio.16:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:gpio.22 Out:gpio.2 Dir:NO_PIN Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Flood coolant gpio.25]
[MSG:INFO: Mist coolant gpio.21]
[MSG:INFO: Probe Pin: gpio.32:low:pu]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.6 [FluidNC v3.6.4 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
E (20073) sdmmc_sd: sdmmc_check_scr: send_scr returned 0x108
[MSG:ERR: sdmmc_card_init failed code 0x108]
No SD card
error:60
E (27021) sdmmc_io: sdmmc_io_reset: unexpected return: 0x109
[MSG:ERR: sdmmc_card_init failed code 0x109]
No SD card
error:60
E (31181) sdmmc_io: sdmmc_io_reset: unexpected return: 0x109
[MSG:ERR: sdmmc_card_init failed code 0x109]
No SD card
error:60
[MSG:ERR: sdmmc_card_init failed code 0x107]
No SD card
error:60
[MSG:ERR: sdmmc_card_init failed code 0x107]
No SD card
error:60
```

### What happened?

I configured a LOLIN ESP32 board with the SdCard (Micro SD TF Card Memory Shield Module SPI Micro SD Adapter)
From Web interface I only see NO SDCARD, or from terminal:
[MSG:ERR: sdmmc_card_init failed code 0x107]

 I tested the SD card with the standard ESP examples and it works fine.
Am I doing something wrong?

Thanks.

MitchBradley, 2022-12-05:
Please provide a pointer to "the standard ESP examples" so we know exactly what you mean, along with any extra SDK configuration files that you used, and information about the compilation environment that you used with those examples.

zamby, 2022-12-05:
[image]

Both SD 1GB and 16GB not working.

MitchBradley, 2022-12-05:
I ordered one of those adapter cards.  When it arrives I will look into the problem.

zamby, 2022-12-05:
Oh thanks. I didn't want you to spend money!

MitchBradley, 2022-12-05:
I ordered it some time ago.  My time is much more valuable than the cost of the adapter card.  Did you want me to spend that instead?

FlashMiks, 2022-12-09:
[image]
This is mod will work with this SD adapter. I think it is becouse too hight frequency SPI in new builds. I connect directly without buffer chip and 3.3 voltage regulator.

Alpusa, 2023-08-13:
> [image] This is mod will work with this SD adapter. I think it is becouse too hight frequency SPI in new builds. I connect directly without buffer chip and 3.3 voltage regulator.

You can also power VCC of the SD adapter using VIN or 5+.

closed 2024-02-03 by bdring

nithinkumarmnair, 2025-09-15:
I am also going to try this mod to the cheap SD card module by removing buffer chip. Just want clarity on. The voltage side why do we bypass the voltage regulator esp32 pins are 3.3v max right if we powerup with 5v will it get damaged?  Please suggest?