bdring/FluidNC #740: Problem: I am getting an error for Fluid Term when I try to open. "setConsoleColor failed"

DanAttema, 2022-12-08:
### Controller Board

ESP32 WROOM board.  Downloaded FluidNC fine.

### Machine Description

Using an ESP32 to control an XYYZ CNC Gantry Router.  

### Input Circuits

```yaml
None yet.  Just starting.
```

### Configuration file

```yaml
None yet.  Will be an XYYZ.
```

### Startup Messages

```text
Have got that far.
```

### User Interface Software

FluidTerm

### What happened?

I am getting an error for Fluid Term when I try to open. "setConsoleColor failed" The next line is "...press any key to continue".  It exits when I press a key.  Multiple attempt produce the same result.  It looks like a problem with Python or Windows or something.  Any suggestion will be helpful.

I am going to try another terminal program.

### Other Information

Screen shot FluidTerm error.

[image]

DanAttema, 2022-12-08:
Startup Messages

Should say "Have not got that far."

MitchBradley, 2022-12-08:
Ancient version of Windows?

DanAttema, 2022-12-08:
Windows 7

MitchBradley, 2022-12-08:
So yes.  Windows 7 does not have the new console subsystem.  Can you use a different machine?

DanAttema, 2022-12-08:
Not conveniently.  Does Windows 10 work?  What would be an alternative terminal program to run on Windows 7?

MitchBradley, 2022-12-08:
win 10 works just fine, especially if it is reasonably up to date, like say within the last year or so.  Any terminal program will work; I like PuTTY and TeraTermPro.

MitchBradley, 2022-12-08:
Arduino serial monitor works too.

DanAttema, 2022-12-08:
Thanks for the help and the quick response.  I may upgrade my computer to Windows 10.  For now I will try one of the alternatives.

MitchBradley, 2022-12-08:
I would not recommend upgrading a Windows 7 machine to windows 10.  Old machines often have problems with new operating systems.

MitchBradley, 2022-12-08:
windows 10 laptops go for about $100 on ebay

closed 2022-12-08 by MitchBradley

DanAttema, 2022-12-08:
A cheep laptop could work for FluidTerm.  

When I bought this machine it came with Window 10 upgrade as an option.  I installed Window 7 because I was running some legacy programs at the time.  Kind of regretting it now as Windows 10 is needed many new programs.

MitchBradley, 2022-12-08:
FluidTerm is not necessary.  It has some nice features for our internal uses, and it is a known starting point so we don't have to tell people how to set up half a dozen other serial terminals, but other serial terminals will do the job, and for long-term use you will want to use either WebUI or a sender like UGS.