bdring/FluidNC #744: Problem: Homing button greyed out in all senders

graham2550, 2022-12-10:
### Controller Board

ESP32 self-designed 4-axis board.

[image]

[ESP38PIN Laser schematic.pdf](https://github.com/bdring/FluidNC/files/10200973/ESP38PIN.Laser.schematic.pdf)

### Machine Description

Gantry router with external TB6600 drivers, motor on X, dual motors on Y and motor on Z.
All motors have neg limit switches NC.

### Input Circuits

```yaml
Only inputs are 4 x limit switches NC to GND and a probe input NO to GND.
```

### Configuration file

```yaml
name: "GRAHAMS_XYYZ Laser"
board: "ESP32_38PIN"
meta: G.Payne 11-Dec-2022 Rev 7

arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: true
report_inches: false
enable_parking_override_control: false
use_line_numbers: true
planner_blocks: 16

stepping:
  engine: RMT
  idle_ms: 255
  dir_delay_us: 0
  pulse_us: 3
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  shared_stepper_reset_pin: NO_PIN
  x:
    steps_per_mm: 80.000
    max_rate_mm_per_min: 15000.000
    acceleration_mm_per_sec2: 700.000
    max_travel_mm: 1300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      allow_single_axis: true
      mpos_mm: 0
      feed_mm_per_min: 200.000
      seek_mm_per_min: 1200.000
      settle_ms: 250
      seek_scaler: 1.000
      feed_scaler: 1.000
    motor0:
      limit_neg_pin: gpio.34:high
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: gpio.14
        direction_pin: gpio.2:low
        disable_pin: NO_PIN
  y:
    steps_per_mm: 80.000
    max_rate_mm_per_min: 15000.000
    acceleration_mm_per_sec2: 700.000
    max_travel_mm: 1300.000
    soft_limits: false
    homing:
      cycle: 3
      positive_direction: false
      allow_single_axis: true
      mpos_mm: 0
      feed_mm_per_min: 200.000
      seek_mm_per_min: 1200.000
      settle_ms: 250
      seek_scaler: 1.000
      feed_scaler: 1.000
    motor0:
      limit_neg_pin: gpio.35:high
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: gpio.26
        direction_pin: gpio.15:low
        disable_pin: NO_PIN
    motor1:
      limit_neg_pin: gpio.39:high
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: gpio.12
 #       direction_pin: gpio.15
        disable_pin: NO_PIN
  z:
    steps_per_mm: 200.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 500.000
    max_travel_mm: 50.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      allow_single_axis: true
      mpos_mm: 0
      feed_mm_per_min: 200.000
      seek_mm_per_min: 500.000
      settle_ms: 250
      seek_scaler: 1.000
      feed_scaler: 1.000
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.36:high
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: gpio.27
        direction_pin: gpio.33:low
        disable_pin: NO_PIN

probe:
  pin: gpio.32:low
  check_mode_start: false

start:
  must_home: true
  check_limits: false
  deactivate_parking: true

coolant:
  flood_pin: gpio.22:high
  mist_pin: gpio.21:high
  delay_ms: 0

Laser:
  pwm_hz: 5000
  output_pin: gpio.25
  enable_pin: NO_PIN
  disable_with_s0: true
  s0_with_disable: true
  tool_num: 0
  speed_map: 0=0.000% 1000=100.000%
  off_on_alarm: true

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
```

### Startup Messages

```text
FluidNC v1.2.0 using COM4
Exit: Ctrl-C, Ctrl-Q or Ctrl-], Clear screen: CTRL-W
Upload: Ctrl-U, Reset ESP32: Ctrl-R, Send Override: Ctrl-O
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12812
load:0x40080400,len:3032
entry 0x400805e4

[MSG:INFO: FluidNC v3.6.5-pre4]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:MY_laser.yaml]
[MSG:INFO: Machine GRAHAMS_XYYZ Laser]
[MSG:INFO: Board ESP32_30PIN]
[MSG:INFO: SPI not defined]
[MSG:INFO: No SD Card CS Pin]
[MSG:INFO: See http://wiki.fluidnc.com/en/config/sd_card#sdfallbackcs-access-sd-without-a-config-file]
[MSG:INFO: Stepping:RMT Pulse:3us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,1300.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.14 Dir:gpio.2:low Disable:NO_PIN]
[MSG:INFO:  X Neg Limit gpio.34]
[MSG:INFO: Axis Y (0.000,1300.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.26 Dir:gpio.15:low Disable:NO_PIN]
[MSG:INFO:  Y Neg Limit gpio.35]
[MSG:INFO:   Motor1]
[MSG:INFO:     standard_stepper Step:gpio.12 Dir:NO_PIN Disable:NO_PIN]
[MSG:INFO:  Y2 Neg Limit gpio.39]
[MSG:INFO: Axis Z (-50.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.27 Dir:gpio.33:low Disable:NO_PIN]
[MSG:INFO:  Z Pos Limit gpio.36]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Laser Ena:NO_PIN Out:gpio.25 Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: Flood coolant gpio.22]
[MSG:INFO: Mist coolant gpio.21]
[MSG:INFO: Probe Pin: gpio.32:low]
[MSG:INFO: Connecting to STA SSID:B936_CPE_954E]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connecting...]
[MSG:INFO: Connecting....]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connecting...]
[MSG:INFO: Connecting....]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]

Grbl 3.6 [FluidNC v3.6.5-pre4 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
```

### User Interface Software

UGS, OpenBuilds Control, Lightburn

### What happened?

Everything works great except in all my senders, UGS platform, OpenBuilds Control, and Lightburn, the homing button is greyed out.
I can home the machine with $H and individually with $HX, $HY, and $HZ, all work fine.
Obviously, I would like the Home button to work simply for convenience.

graham2550, 2022-12-14:
ADDITIONAL INFO:
Lightburn reports the reason as "no limit switches found", obvious nonsense. It finds them when it homes.
Could it be the optocouplers?
Anybody have a clue?

MitchBradley, 2022-12-15:
Have you asked the Lightburn people what that message means in detail?  I just tried running Lightburn but it is complaining the the license expired.  They gave me a complimentary license key to use for testing FluidNC, but apparently that license had some limits attached to it.  I suppose I could ask for another one, but I have a lot of other stuff I would rather spend my time on.

To evaluate this problem on other senders, e.g. UGS, I would need to know your UGS configuration settings, which version of UGS, which controller type for the UGS connection, and full console logs.  Debugging requires lots of detail, not just "the homing button is greyed out"

MitchBradley, 2022-12-15:
The older version of Lightburn continues to work with the old licesns.  Using your config file on a controller with pullups on the pins you use for limits, and with the limit switch config lines changed from :high to :low, the homing button is not grayed out.  Pressing it results in these messages with $message/level=debug .  Maybe the Lightburn people can tell you what causes the "no limit switches found" message.  The general rule is "ask for help from the people that took your money before asking the volunteers".

[image]

graham2550, 2022-12-15:
what happened to the reply I posted yesterday??

MitchBradley, 2022-12-16:
> what happened to the reply I posted yesterday??

I don't know.  I never saw it.  Maybe some sort of github glitch?

graham2550, 2022-12-16:
Well, good news this morning, The homing button is enabled now in Lightburn at least, but not the other senders.
I am truly puzzled, but I use Lightburn mostly so I think this problem has come to a close.
Thank you very much Mitch for all your efforts, keep up the good work.

closed 2022-12-16 by graham2550