bdring/FluidNC #748: Problem: RC Servo stutters and goes to different positions

Swarlo1, 2022-12-15:
### Controller Board

MKS-DLC32 flashed FluidNC

### Machine Description

Plotter with external stepper drivers, single x and y motor, mechanical endstops on x and y. RC Servo is for the Z axes.

### Configuration file

```yaml
board: MKS DLC32 2.1
name: Carlo CNC Laser/Plotter/Plasma
meta: (17/6/2022) by Swarlo
stepping:
  engine: I2S_STREAM
  idle_ms: 0
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: i2so.0
  x:
    steps_per_mm: 107.000
    max_rate_mm_per_min: 8000.000
    acceleration_mm_per_sec2: 500.000
    max_travel_mm: 530.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 3000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.36:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 5.000
      stepstick:
        step_pin:  i2so.1
        direction_pin: I2SO.2:low

  y:
    steps_per_mm: 107.000
    max_rate_mm_per_min: 8000.000
    acceleration_mm_per_sec2: 500.000
    max_travel_mm: 600.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 3000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 5.000
      stepstick:
        step_pin: I2SO.5
        direction_pin: I2SO.6

  z:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 2
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000

    motor0:
       rc_servo:
        pwm_hz: 50
        output_pin: gpio.22
        min_pulse_us: 600
        max_pulse_us: 2500

i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.15

probe:
  pin: gpio.2
  check_mode_start: true

start:
  must_home: false
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.6.5]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config 2mm.yaml]
[MSG:INFO: Machine CNC Laser/Plotter/Plasma]
[MSG:INFO: Board MKS DLC32 2.1]
[MSG:INFO: I2SO BCK:gpio.16 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.14 MOSI:gpio.13 MISO:gpio.12]
[MSG:INFO: SD Card cs_pin:gpio.15 detect:NO_PIN]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:0ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable I2SO.0]
[MSG:INFO: Axis X (0.000,530.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.1 Dir:I2SO.2:low Disable:NO_PIN]
[MSG:INFO:  X Neg Limit gpio.36:low]
[MSG:INFO: Axis Y (0.000,600.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.5 Dir:I2SO.6 Disable:NO_PIN]
[MSG:INFO:  Y Neg Limit gpio.35:low]
[MSG:INFO: Axis Z (0.000,2.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     rc_servo Pin:gpio.22 Pulse Len(600,2500 period:1048575)]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Probe Pin: gpio.2]
[MSG:INFO: Connecting to STA SSID:xxxxxxxxxxx]
[MSG:INFO: Connecting.]
[MSG:INFO: Connected - IP is xxx.xxx.x.xx]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
```

### What happened?

I recently flashed fluidnc to my mks-dlc32. All axes and endstops work so far but i cant get my RC Servo (the little ones 9gram 3 Pins PWM signal) to work correctly. Every time i home the servo and change the z position it stutters and sometimes it goes to different Z positions. Other times it doesn't move at all or the home position changes.

### Other Information

[MKS DLC32 V2.1_003 PIN.pdf](https://github.com/bdring/FluidNC/files/10239041/MKS.DLC32.V2.1_003.PIN.pdf)

MitchBradley, 2022-12-15:
According to my reading of the DLC32 V21._003 schematic, gpio.22 is the probe pin, an input with RC filtering between the ESP32 and the external pin.  That won't work with an RC servo because the RC filter will ruin the PWM signal.  I got an RC servo to work with a DLC32 V1.1 by using the LCD_CS pin on the EXP1 connector.  LCD_CS comes from gpio.25 and goes through an output buffer, thus providing a strong clean signal for the RC servo.

bdring, 2022-12-15:
I added the info above to the wiki page we have for that controller. Please let us know if we can close tih issue.

http://wiki.fluidnc.com/en/hardware/3rd-party/MKS_DLC32#faq

Swarlo1, 2022-12-15:
Thank you very much. When I connected the servo directly with the new pin 25, the stuttering stopped and the servo had a fixed home point in Z, but after i connected the servo to an extension cable of 2m lenght there was a little sutter but not like before.

MitchBradley, 2022-12-15:
We don't have time to debug your hardware.  It does sound like the cable could be too small a wire gauge, or perhaps a bad connection, but at this point you need to consider other resources for debugging.  It is no longer a FluidNC problem.

closed 2022-12-15 by MitchBradley