bdring/FluidNC #750: Problem: FluidNC v3.6.5 weird ack echo across channels

synman, 2022-12-16:
### Controller Board

Open Builds BlackBox X32

### Machine Description

Queen Bee Pro

### Input Circuits

```yaml
N/A
```

### Configuration file

```yaml
board: Open Builds BlackBox X32
name: BlackBox_X32_XYYZ-DOOR
meta: Handcrafted by Shell M Shrader
stepping:
  engine: RMT
  idle_ms: 255
  pulse_us: 2
  dir_delay_us: 1
  disable_delay_us: 0
  segments: 12
axes:
  shared_stepper_disable_pin: gpio.17:low
  shared_stepper_reset_pin: NO_PIN
  x:
    steps_per_mm: 200.0
    max_rate_mm_per_min: 3000.0
    acceleration_mm_per_sec2: 100.0
    max_travel_mm: 505.0
    soft_limits: true
    homing:
      cycle: 2
      allow_single_axis: true
      positive_direction: true
      mpos_mm: 0.0
      feed_mm_per_min: 100.0
      seek_mm_per_min: 2000.0
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: gpio.35
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 5.0
      standard_stepper:
        step_pin: gpio.12
        direction_pin: gpio.14
        disable_pin: NO_PIN
  y:
    steps_per_mm: 200.0
    max_rate_mm_per_min: 3000.0
    acceleration_mm_per_sec2: 100.0
    max_travel_mm: 505.0
    soft_limits: true
    homing:
      cycle: 2
      allow_single_axis: true
      positive_direction: true
      mpos_mm: 0.0
      feed_mm_per_min: 100.0
      seek_mm_per_min: 2000.0
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: gpio.34
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 5.0
      standard_stepper:
        step_pin: gpio.27
        direction_pin: gpio.26
        disable_pin: NO_PIN
    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.0
      standard_stepper:
        step_pin: gpio.33
        direction_pin: gpio.32
        disable_pin: NO_PIN
  z:
    steps_per_mm: 200.0
    max_rate_mm_per_min: 600.0
    acceleration_mm_per_sec2: 100.0
    max_travel_mm: 105.0
    soft_limits: true
    homing:
      cycle: 1
      allow_single_axis: true
      positive_direction: true
      mpos_mm: 0.0
      feed_mm_per_min: 100.0
      seek_mm_per_min: 300.0
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: gpio.39
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 5.0
      standard_stepper:
        step_pin: gpio.15
        direction_pin: gpio.2:low
        disable_pin: NO_PIN
kinematics:
  Cartesian: null
spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18
sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN
control:
  safety_door_pin: gpio.16:pu
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
coolant:
  flood_pin: NO_PIN
  mist_pin: gpio.21
  delay_ms: 0
probe:
  pin: gpio.22:low
  check_mode_start: true
macros:
  startup_line0: null
  startup_line1: null
  macro0: null
  macro1: null
  macro2: null
  macro3: null
start:
  must_home: true
  deactivate_parking: false
  check_limits: false
parking:
  enable: false
  axis: Z
  target_mpos_mm: -5.0
  rate_mm_per_min: 800.0
  pullout_distance_mm: 5.0
  pullout_rate_mm_per_min: 250.0
user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN
arc_tolerance_mm: 0.002
junction_deviation_mm: 0.01
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
planner_blocks: 16
PWM:
  pwm_hz: 5000
  direction_pin: gpio.4
  output_pin: gpio.25
  enable_pin: gpio.13
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 30000
  spindown_ms: 10000
  tool_num: 0
  speed_map: 0=0.000% 0=30.000% 6000=30.000% 24000=100.000%
  off_on_alarm: false
```

### Startup Messages

```text
Send: $startup/show
Recv: [MSG:INFO: FluidNC v3.6.5]
Recv: [MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
Recv: [MSG:INFO: Local filesystem type is spiffs]
Recv: [MSG:INFO: Configuration file:config.yaml]
Recv: [MSG:INFO: Machine BlackBox_X32_XYYZ-DOOR]
Recv: [MSG:INFO: Board Open Builds BlackBox X32]
Recv: [MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
Recv: [MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
Recv: [MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
Recv: [MSG:INFO: Axis count 3]
Recv: [MSG:INFO: Shared stepper disable gpio.17:low]
Recv: [MSG:INFO: Axis X (-505.000,0.000)]
Recv: [MSG:INFO:   Motor0]
Recv: [MSG:INFO:     standard_stepper Step:gpio.12 Dir:gpio.14 Disable:NO_PIN]
Recv: [MSG:INFO:  X Neg Limit gpio.35]
Recv: [MSG:INFO: Axis Y (-505.000,0.000)]
Recv: [MSG:INFO:   Motor0]
Recv: [MSG:INFO:     standard_stepper Step:gpio.27 Dir:gpio.26 Disable:NO_PIN]
Recv: [MSG:INFO:  Y Neg Limit gpio.34]
Recv: [MSG:INFO:   Motor1]
Recv: [MSG:INFO:     standard_stepper Step:gpio.33 Dir:gpio.32 Disable:NO_PIN]
Recv: [MSG:INFO: Axis Z (-105.000,0.000)]
Recv: [MSG:INFO:   Motor0]
Recv: [MSG:INFO:     standard_stepper Step:gpio.15 Dir:gpio.2:low Disable:NO_PIN]
Recv: [MSG:INFO:  Z Neg Limit gpio.39]
Recv: [MSG:INFO: safety_door_pin gpio.16:pu]
Recv: [MSG:INFO: Kinematic system: Cartesian]
Recv: [MSG:INFO: PWM Spindle Ena:gpio.13 Out:gpio.25 Dir:gpio.4 Freq:5000Hz Period:8191]
Recv: [MSG:INFO: Using spindle PWM]
Recv: [MSG:INFO: Mist coolant gpio.21]
Recv: [MSG:INFO: Probe Pin: gpio.22:low]
Recv: [MSG:INFO: Connecting to STA SSID:SHELLWLAN]
Recv: [MSG:INFO: Connecting.]
Recv: [MSG:INFO: Connected - IP is 10.151.51.35]
Recv: [MSG:INFO: WiFi on]
Recv: [MSG:INFO: Start mDNS with hostname:http://blackbox-x32.local/]
Recv: [MSG:INFO: SSDP Started]
Recv: [MSG:INFO: HTTP started on port 80]
Recv: [MSG:INFO: Telnet started on port 23]
```

### User Interface Software

CNCjs and WebUI

### What happened?

I'm seeing acks cross "channel" boundaries.   OKs from webui / telnet pop on serial and vice-versa.

Here's a video:

https://youtu.be/-xPGeBrDXko

MitchBradley, 2022-12-17:
This is fixed in the InputThread branch.  I don't know when we will promote that to the mainline.

synman, 2022-12-17:
I may just pull it and build it.   If FluidNC building is half as easy as Grbl_Esp32 using platformio, this is a no brainer.

MitchBradley, 2022-12-17:
Please do try it.  We need to get some outside testing on that branch.  I have been using it on my machines.

MitchBradley, 2022-12-17:
It attempts to do all of the UI IO - network, serial, whatnot - on one core, and the gcode and stepping on the other.  In principle that seems like a good idea to me, but I had to fix several synchronization issues to make it work.

synman, 2022-12-17:
wow, there's alot in this branch.

synman, 2022-12-23:
hey @MitchBradley

How do you want me to go about reporting / addressing regression issues in this branch (InputThread)?

For example, we've got some malformed yaml coming out of $CD:
```
stepping:<cr><lf>
  engine:RMT<cr><lf>
```
There should always be a space after a name in a name/value pair, like so:
```
stepping: <cr><lf>
  engine: RMT<cr><lf>
```
It gets even more complicated when yaml parsers try to deal with something like this:
```
  shared_stepper_disable_pin:gpio.17:low<cr><lf>
```
It will pass a syntax check and properly parse `gpio.17:low` as a value if done like so:
```
  shared_stepper_disable_pin: gpio.17:low<cr><lf>
```

MitchBradley, 2022-12-23:
00af513

synman, 2022-12-23:
Some additional observations:

From any of the 3 interface (serial, WebUI, and telnet):
- `$CD` response gets echoed to telnet and WebUI
- `$Settings/List` response gets echoed to telnet and WebUI
- other commands like `$axes` responses get echoed to telnet and WebUI
- status report `?` responses / acks stay in their channel --no leakage
- parser status `$G` responses / acks stay in their requester's channel -- no leakage

One last callout is specific to WebUI and is just an FYI.  It doesn't show acks (ok) for any status report requests, both auto reporting and ones you trigger.  It does appear to be showing them for everything else.

synman, 2022-12-23:
I'll run a couple jobs tomorrow (will likely carve air for this) and report back if anything odd happens.

synman, 2022-12-23:
> 00af513

nice... I'll pull this in and blackout my own hack

MitchBradley, 2022-12-23:
> One last callout is specific to WebUI and is just an FYI. It doesn't show acks (ok) for any status report requests, both auto reporting and ones you trigger. It does appear to be showing them for everything else.

Status reports are single-character "realtime" commands that are processed differently from line-oriented commands, per the GRBL protocol.  ok acks apply only to lines.

MitchBradley, 2022-12-23:
Realtime characters are "out of band" with respect to the line protocol.

synman, 2022-12-23:
understood.  Just calling it out because I can't verify the acks for them in the WebUI.   I understand why they are being suppressed.... the terminal window will fill up quickly with useless "ok" messages every other line.

other real-time commands like feed hold and resume (! and ~) behave the same... and I did verify they too are staying within their channels.

synman, 2022-12-23:
check this out:

[image]

I thought cycle start was a real-time command...

synman, 2022-12-23:
duh, I figured it out..... user error.

I was hitting enter.  That ok up there with the cycle start was trigged by the \r I sent right before it.  Nothing to see here.

MitchBradley, 2022-12-23:
bf002a23

synman, 2022-12-23:
Getting the following when I home.  I don't have any startup lines

```
Send: $H
Recv: [MSG:ERR: Startup line: Bad GCode number format]
Recv: [MSG:ERR: Startup line: Bad GCode number format]
```

Coming from here:
```
void settings_execute_startup() {
    Error status_code;
    for (int i = 0; i < config->_macros->n_startup_lines; i++) {
        String      str = config->_macros->startup_line(i);
        const char* s   = str.c_str();
        if (s && strlen(s)) {
            // We have to copy this to a mutable array because
            // gc_execute_line modifies the line while parsing.
            char gcline[256];
            strncpy(gcline, s, 255);
            status_code = gc_execute_line(gcline);
            // Uart0 << ">" << gcline << ":";
            if (status_code != Error::Ok) {
                log_error("Startup line: " << errorString(status_code) << " " << gcline);
            }
        }
    }
}
```

config->_macros->startup_line(i) is null:
```
Recv: [MSG:ERR: Startup line: Bad GCode number format for line: [0] gcline: [NULL] contents: [null]
Recv: [MSG:ERR: Startup line: Bad GCode number format for line: [1] gcline: [NULL] contents: [null]
```
Let's check my startup lines:
```
Send: $N
Recv: $N0=null
Recv: $N1=null
```
I can't clear them:
```
Send: $N0=
Recv: error:3
```
My guess is this just needs to be checked for null and treat it as no startup line.  I am curious about how you go about reseting them.  Let's say I do $N0=G91.  ... How do I undo that?

alarms and messages ([MSG:) are being broadcasted to all channels.  I'm thinking this is a good thing.  Stuff like:
```
[MSG:INFO: Disabling all motors]
[MSG:INFO: Enabling all motors]
[MSG:INFO: Caution: Unlocked]
[MSG:INFO: Soft limit on X target:50.000]
ALARM:2
[MSG:INFO: Reset to continue]

Grbl 3.6 [FluidNC v3.6.5-pre2 (InputThread-bf002a23-dirty) (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
[MSG:ERR: Startup line: Bad GCode number format]
[MSG:ERR: Startup line: Bad GCode number format]
[MSG:ERR: Startup line: Bad GCode number format]
[MSG:ERR: Startup line: Bad GCode number format]
```

MitchBradley, 2022-12-26:
The correct way to say it is:
```
macros:
  startup_line0:
  startup_line1:
  macro0:
```
not
```
   startup_line0: null
```
The FluidNC config file format is not real YAML.  It does not have data types at all, so there is no such thing as the "null" keyword.  Every value is just a string as far as the config parser is concerned.  The interpretation of that string as a number or whatever is done by the individual item, not at the parser level.

You might argue, and I would agree with you to some extent, that mentioning its similarity to YAML is misleading.  On the other hand, pretending that it is YAML does have the advantage that many editors and display agents will colorize and help with indentation.

I personally use the extension .flnc in my files, instead of .yaml, but others are not so fastidious.

MitchBradley, 2022-12-26:
Re $N0=G91 (for example) - at present the only way to change startup lines is by sending, for example
```
$/macros/startup_line0=G0 X0 Y0
```
That is not persistent; to persist it you have to change the config file.

synman, 2022-12-26:
null likely got inserted via the python yaml parser.  I'm seeing it in a couple other places within my "parsed" config file too:

```
kinematics:
  Cartesian: null
...
macros:
  startup_line0: null
  startup_line1: null
  macro0: null
  macro1: null
  macro2: null
  macro3: null
```

I can likely hack around this for now with some substitution / replace logic.   Another thought is to filter them out within FluidNC.   Having a "yaml" like config file is not all that appealing.

synman, 2022-12-27:
I've got this cleaned up in my own client / sender now.  I do hate the idea of having to parse out literal "null" anywhere.

MitchBradley, 2023-01-12:
Are we done? Can we close this?

synman, 2023-01-12:
yes!

closed 2023-01-12 by synman