bdring/FluidNC #751: MKS DLC32 V2.1 config

iherbak, 2022-12-20:
### Controller Board

MKS DLC v2.1

### Machine Description

Normal XY laser engraver with DLC32 board

### Configuration file

```yaml
board: MKS-DLC32 V2.1
name: K40 Laser
meta: (01.01.2022) by Skorpi

kinematics:
  Cartesian:

stepping:
  engine: I2S_STATIC
  idle_ms: 0
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
axes:
  shared_stepper_disable_pin: I2SO.0
  x:
    steps_per_mm: 157.750
    max_rate_mm_per_min: 18000.000
    acceleration_mm_per_sec2: 1500.000
    max_travel_mm: 325.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 5000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.36
      hard_limits: false
      pulloff_mm: 2.000
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2

  y:
    steps_per_mm: 157.750
    max_rate_mm_per_min: 12000.000
    acceleration_mm_per_sec2: 300.000
    max_travel_mm: 220.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 5000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35
      hard_limits: false
      pulloff_mm: 2.000
      stepstick:
        step_pin: I2SO.5
        direction_pin: I2SO.6:low

  z:
    steps_per_mm: 157.750
    max_rate_mm_per_min: 12000.000
    acceleration_mm_per_sec2: 500.000
    max_travel_mm: 80.000
    soft_limits: true
    homing:
      cycle: 0
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 1000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34
      hard_limits: false
      pulloff_mm: 1.000
      stepstick:
        step_pin: I2SO.3
        direction_pin: I2SO.4

i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14

sdcard:
  cs_pin: gpio.15
  card_detect_pin: gpio.39

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: gpio.33:low:pu
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

macros:
  startup_line0:
  startup_line1:
  macro0: $SD/Run=lasertest.gcode
  macro1: $SD/Run=home.gcode
  macro2:
  macro3:

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

probe:
  pin: gpio.22
  check_mode_start: true

Laser:
  pwm_hz: 5000
  #L on Beeper / IN on TTL
  output_pin: gpio.32
  enable_pin: I2SO.7
  disable_with_s0: false
  s0_with_disable: false
  tool_num: 0
  speed_map: 0=0.000% 0=12.500% 1700=100.000%
# 135=0mA 270=5mA 400=10mA 700=16mA
user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

start:
  must_home: false

# 5,18,19,22,23,25,26,27,32,33,39,I2SO.7
# SDA 0 / SCL 4
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.0.x (main-06f8f9d4-dirty)]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:MKS_DLC32_21_XYZ.yaml]
[MSG:INFO: Machine K40 Laser]
[MSG:INFO: Board MKS-DLC32 V2.1]
[MSG:INFO: I2SO BCK:gpio.16 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.14 MOSI:gpio.13 MISO:gpio.12]
[MSG:INFO: SD Card Detect gpio.39]
[MSG:INFO: SD Card cs_pin:gpio.15 detect:gpio.39]
[MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:0ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable I2SO.0]
[MSG:INFO: Axis X (0.000,325.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.1 Dir:I2SO.2 Disable:NO_PIN]
[MSG:INFO:  X Neg Limit gpio.36]
[MSG:INFO: Axis Y (0.000,220.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.5 Dir:I2SO.6:low Disable:NO_PIN]
[MSG:INFO:  Y Neg Limit gpio.35]
[MSG:INFO: Axis Z (0.000,80.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.3 Dir:I2SO.4 Disable:NO_PIN]
[MSG:INFO:  Z Neg Limit gpio.34]
[MSG:INFO: macro0_pin gpio.33:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Laser Ena:I2SO.7 Out:gpio.32 Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: Probe Pin: gpio.22]
[MSG:INFO: Connecting to STA SSID:]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is ]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
```

### User Interface Software

FluidTerm

### What happened?

I guess the config should have   

card_detect_pin: gpio.39

in order to be able work with the SD card

bdring, 2022-12-20:
We have 2 example configs here.

https://github.com/bdring/FluidNC/tree/main/example_configs

Both use `no_pin` for detect. FluidNC does not use the detect anyway.

What exactly is your issue?

iherbak, 2022-12-20:
I based the info on the MKS DLC32 released firmware where the detect pin is set. I did not know the detection is not used.

However if i open a second fluidterm it kills the whole firmware? Is it limited to one session?

Thanks

bdring, 2022-12-20:
FluidTerm uses the USB/serial port. You can only make one connection via the serial port.

Do you mean the WebUI? I think the latest version of the firmware supports multiple connections. It does not look like you are using a release, so I cannot determine the revision.

MitchBradley, 2022-12-20:
> if i open a second fluidterm it kills the whole firmware?

If you really mean FluidTerm and not WebUI, then we need to know which host operating system you are using.  On Windows, it is not possible to open the same serial port from two different processes, so trying to start FluidTerm when it is already running will just fail to open  the serial port.  On Linux, it is possible to open the same port from two different processes, resulting in much confusion because there is no way to know which process will receive the data sent by FluidNC, and characters that are sent by the processes can be interleaved.  This is a problem with Linux and the only solution is to be careful to run only one serial program at a time.

iherbak, 2022-12-20:
OK in this case i meant the WebUI. However when i install the release 3.6.5 i alwas get [MSG:ERR: sdmmc_card_init failed code 0x102].

Can it be the result of not matching flash parameters. 
My board is detected as 40M and not 80M and MKS suggest to use DOUT not DIO

Thanks

MitchBradley, 2022-12-20:
https://github.com/bdring/FluidNC/pull/742

If you build from the Devt branch you will get that feature

iherbak, 2022-12-20:
I actually made a fork, 
- modified the settings in the install scripts, 
- grabbed the dout_40m bootloader from the espressif, 
- built the wifi firmware 
- tried to build the filesystem as well ( that resulted in  a 2MB bin have no idea why) so instead i uploaded using the default tool UI

Then i start the webui and start a gcode file. Then i open a second tab and try to connect
fluidterm shows this

Grbl 3.0 [FluidNC v3.0.x (main-06f8f9d4-dirty) (wifi) '$' for help]
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed).

Core  1 register dump:
PC      : 0x400e97bc  PS      : 0x00060035  A0      : 0x8008227e  A1      : 0x3ffc036c
A2      : 0x3ffb4c18  A3      : 0x00000001  A4      : 0x3ffb1cc0  A5      : 0x3ffb1c80
A6      : 0x3ffc15c4  A7      : 0x3ffb29b4  A8      : 0x8008a820  A9      : 0x000000c0
A10     : 0x3ffb4c6c  A11     : 0x00000001  A12     : 0x3ffb1c80  A13     : 0x3ffb1c00
A14     : 0x3ffc15d8  A15     : 0x3ffb2a74  SAR     : 0x00000002  EXCCAUSE: 0x00000007
EXCVADDR: 0x00000000  LBEG    : 0x4008bc14  LEND    : 0x4008bc1f  LCOUNT  : 0x00000000

Backtrace:0x400e97b9:0x3ffc036c |<-CORRUPTED

ELF file SHA256: 0000000000000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:12784
load:0x40080400,len:3032
entry 0x400805e4

[MSG:INFO: FluidNC v3.0.x (main-06f8f9d4-dirty)]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:ERR: Skipping configuration file due to panic]

MitchBradley, 2022-12-20:
We only support our own releases.  We do not have time to help with custom builds.  I have version 3.6.5 running on a DLC32 V1.1 right now, accessing files on SD.  I started WebUI with no problems, then connected with FluidTerm from Windows, and it is still working.  Here is the config file I am using for testing:

[dlc32.txt](https://github.com/bdring/FluidNC/files/10272173/dlc32.txt)

iherbak, 2022-12-20:
I understand. However could you try start a gcode on the webui and then
connect with a second webui?

I will look into the settings

Mitch Bradley ***@***.***> (időpont: 2022. dec. 20., K,
21:26) ezt írta:

MitchBradley, 2022-12-20:
Connecting with a second WebUI while running a GCode program is problematic.  When you start WebUI, the http server sends the WebUI code index.html.gz to the browser.  index.html.gz is stored in the local FLASH filesystem.  FLASH accesses are very slow and they interfere with code execution because the code is also stored in FLASH.  The instruction cache often hides this conflict, but if there is a cache miss, the code cannot be fetched until the FLASH access for sending index.html.gz has finished.  In many cases that delay is not a problem, but when you are running a GCode program, the processor is spending a lot of time servicing interrupts.  Due to a limitation of the ESP32 processor, an instruction cache miss in an interrupt service routine will cause a crash.  To mitigate this, we place ISR routines in IRAM, but it is very difficult to make sure that every possible bit of code that can be touched during an ISR is properly placed in IRAM.  One particularly troublesome area is that the compiler, by default, puts switch dispatch tables and vtables for virtual methods in FLASH.  We have some linker hacks to move them into RAM, but it is very difficult to ensure that we got them all.

iherbak, 2022-12-20:
Thank you so it is due to design and something specific to my build

Please close the issue

Thank you
Mitch Bradley ***@***.***> (időpont: 2022. dec. 20., K,
21:46) ezt írta:

closed 2022-12-20 by MitchBradley