bdring/FluidNC #753: Problem:

marvin42blue, 2022-12-21:
### Controller Board

my own board: ESP32 wroom + TMC2130 drivers (bigtreetech v3)

### Machine Description

self-built XY Laser gantry
two Nema-14 stepper motors (Nanotec SC3518L1204-A) 24V powered

### Configuration file

```yaml
name: "ESP32 Dev Controller V4"
board: "ESP32 Dev Controller V4"

stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 1
  pulse_us: 4
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.15:high
  
  x:
    steps_per_mm: 100
    max_rate_mm_per_min: 4000
    acceleration_mm_per_sec2: 100
    max_travel_mm: 130
    soft_limits: false
    homing:
      cycle: 2
      mpos_mm: 0
      positive_direction: false
      seek_mm_per_min: 1200
      feed_mm_per_min: 100
      settle_ms: 400
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.14:high:pu
      tmc_2130:
        direction_pin: gpio.12:low
        step_pin: gpio.13
        cs_pin: gpio.4
        spi_index: -1
        r_sense_ohms: 0.110
        run_amps: 1.200
        hold_amps: 0.80
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false

  y:
    steps_per_mm: 100
    max_rate_mm_per_min: 4000
    acceleration_mm_per_sec2: 100
    max_travel_mm: 120
    soft_limits: false
    homing:
      cycle: 2
      mpos_mm: 105
      positive_direction: true
      seek_mm_per_min: 1200
      feed_mm_per_min: 100
      settle_ms: 400
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_pos_pin: gpio.27:high:pu
      tmc_2130:
        direction_pin: gpio.25:low
        step_pin: gpio.26
        cs_pin: gpio.16
        spi_index: -1
        r_sense_ohms: 0.110
        run_amps: 1.200
        hold_amps: 0.80
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false

  z:
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
        
Laser:
  pwm_hz: 5000
  output_pin: gpio.2:high
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  tool_num: 0
  speed_map: 0=0% 1000=100%

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

parking:
  enable: true
  axis: Y
  pullout_distance_mm: 0
  pullout_rate_mm_per_min: 250.000
  target_mpos_mm: 104
  rate_mm_per_min: 800.000
```

### Startup Messages

```text
none
```

### User Interface Software

WebUI, LaserGrbl

### What happened?

FluidNC runs great and the TMC drivers are very nice, in terms of smooth motion and quietness.
But I have issues loosing steps after a few minutes of running the system and I assume its because the TMC2130 are heating up. Using a fan seems to help, but I'm looking for a way to detect this in software. Is it possible to get access to the DIAG0 and DIAG1 pins of the TMC and have FluidNC set an alarm in such a case ?
I was looking into the TMC datasheet (chap 17) and it seems that some diag....-settings are required in order to drive the DIAG pins properly, but this is not configurable in the config.yaml - did I miss something ? or is this not yet implemented ?
Alternatively, how about using a simple 1-wire temperature sensor such as DS1820 - are there any plans to support this ?
thanks, Matthias

MitchBradley, 2022-12-21:
Can you attach an LED to the DIAG pins?

Do you really want a hard-realtime app like FluidNC to bit-bang a 1-wire temperature sensor while it is doing high-speed step generation?

bdring, 2023-04-27:
Stale...closing

closed 2023-04-27 by bdring