bdring/FluidNC #756: Feature: Backlash Compensation

nsiatras, 2022-12-26:
### Machine Context

Every CNC machines

### Feature Description

I often look to see if the backlash compensation feature has been Implemented in FluidNC. 
But the FluidNC project development says 

<i>"Our current position...

We do not support backlash compensation at this time. It requires that the firmware know which side of travel the backlash is at all times. Lightweight machines that are prone to chatter tend to scramble the backlash. Backlash compensation can make things worse. Physical means of limiting backlash like preloading tend to work better. We prefer to spend our time on other features.

<b>If you know of any research papers addressing backlash on lightweight machines, please send them our way.</b>"</i>

After a google search for "lightweight cnc machines and backlash compensation" I found nothing. This makes perfect sense as there is no reason to have a lightweight CNC. CNC machines are machines that are built to perform precision movements. No sane person would build a CNC machine with the intention of being "light".

Anyway... The 2 photos I upload were taken by me. As you will see Backlash compensation is something that works. Siemens, Fanuc, Heidenhain and all CNC controller manufacturers worldwide have the same opinion, as they all have Backlash Compensation in their software.

Backlash compensation is a great feature that is still been missing from FluidNC. Hope FluidNC devs can see the photo.

[image]

PS. My tests are done using GRBL_ESP32. I implemented backlash compensation to GRBL_ESP32 but unfortunately I cannot make it work with Arcs. It only works with G0 and G1 commands.

### Other Approaches

There is no other way to solve that problem.

### How I Can Help

I can run all tests on my CNC machine and get involved with the code.
You can see my CNC machine on the video
https://www.youtube.com/watch?v=OFZef09i1mk

Pastitas, 2023-08-04:
In case this is not getting much attention, I believe this would make sense for a lot of reasons.
I don't have the time (nor the expertise in this field) to program it myself but I'll see to it to get this to someone that might

nsiatras, 2023-08-04:
> In case this is not getting much attention, I believe this would make sense for a lot of reasons. I don't have the time (nor the expertise in this field) to program it myself but I'll see to it to get this to someone that might

I stopped asking for the backlash compensation feature because I implemented it into RabbitGRBL
https://github.com/SourceRabbit/RabbitGRBL

Perhaps you can use it. It is compatible with ESP32

RealTadango, 2024-02-16:
This needs to be added because it makes all cnc machines SO much better.... i can't believe this is not implemented yet......

nsiatras, 2024-02-16:
> This needs to be added because it makes all cnc machines SO much better.... i can't believe this is not implemented yet......

If you add backlash compensation on a lightweight CNC router you will have no difference. The performance will be exactly the same. Backlash compensation is a feature for machines with rigid structure like milling machines with friction slides.
You can change your firmware to Rabbit GRBL (https://github.com/SourceRabbit/RabbitGRBL). I implemented the backlash compensation feature and the results are great 
Video:
https://www.youtube.com/watch?v=-t-U46T7R6s

RealTadango, 2024-02-16:
> If you add backlash compensation on a lightweight CNC router you will have no difference. The performance will be exactly the same. Backlash compensation is a feature for machines with rigid [...]

I really like the config options from FluidNC and i don't want to get something designed for another machine to run on mine. Does RabiitGRBL have a flexible spindle driver? I need one for pwm with 50hz 1100 of 1900 100% power for my machine.

MitchBradley, 2024-02-16:
This is turning into a discussion about another bit of software, inappropriate here.  If you want to discuss RabbitGRBL, do it on a RabbitGRBL forum.

RealTadango, 2024-02-16:
True. We can discuss the need for backlash compensation and in my opinion it really helps with all machines for doing small detailed work. My Y axes has 0.070mm backlash but is still very stiff and solid. It is more a drive backlash. In small details I see this in the result. Compensation would make this perfect. Think about machining PCBs where you want 0.1mm thin traces... I cannot do that now. I really like the setup of FluidNC with the configuration file and would like to use it but without the compensation it will limit me to much.

nsiatras, 2024-02-18:
> I really like the config options from FluidNC and i don't want to get something designed for another machine to run on mine. Does RabiitGRBL have a flexible spindle driver? I need one for pwm with [...]

You might need to adjust the PWM min value to achieve that.
Take a look at the screenshot. It is the Settings editor of Focus (https://www.sourcerabbit.com/Shop/pr-i-91-t-focus-cnc-control-software.htm)
[image]

You can join our discord if you need further assistance
https://discord.com/invite/nRKETyjJ7E

@MitchBradley sorry for the 'inappropriate' content Mitch.

bdring, 2025-01-21:
http://wiki.fluidnc.com/en/development/open_issues

closed 2025-01-21 by bdring