bdring/FluidNC #759: Feature: Support for MKS DLC32 board plus TS35-R LCD display.

AndWalu, 2022-12-27:
### Machine Context

MKS DLC32 board plus TS35-R LCD display.

### Feature Description

Makerbase makes it's firmware open source and published source on GitHub:
  https://github.com/makerbase-mks/MKS-DLC32-FIRMWARE
Maybe there is a chance to integrate TS35-R display with FluidNC?

### Other Approaches

-

### How I Can Help

I have DLC32 and TS35-R and can help with testing.

bdring, 2022-12-27:
We are currently overhauling our communication systems. This is how the firmware sends status and receives user inputs. It does not make sense for us to consider things like this until we are done with that.

That will probably take a few more weeks.

solawc, 2022-12-30:
This is obviously not a difficult thing. Of course, the more important thing is the stability of movement and communication. It is really not difficult to move the screen UI.

MitchBradley, 2022-12-30:
@solawc Okay, so do it.

MitchBradley, 2022-12-30:
One consideration is that this particular feature benefits only one vendor's boards.

msy021, 2023-01-04:
> One consideration is that this particular feature benefits only one vendor's boards.

 Yes but the board is open source and there are lots of users for these boards. Also there are other vendors with similar boards, for example Atomstack fits its laser engravers with boards similar with these and the firmware is fully compatible.
The main thing is that having an lcd touchscreen is great. Best thing for me is using it as a control pendant.
And if you start supporting lcd touchscreens, I am sure that more board manufacturers will start to adopt this feature.

elf128, 2023-03-09:
FYI, I do have a slightly "hacky" implementation of LCD on MKS DLC32. Screen and touch works. Once I'll have UI running I'll share it.

However, please don't expect it to happen next week. I agree with @MitchBradley, doing it right takes time. The implementation of UI in the code base above is emmm, hair risinglly bad, to say the least. It goes against overall philosophy of FluidNC.

And @solawc, This is obviously A Very difficult thing to untangle C-style script ( naming intended ) where every single button position is hardcoded in multiple places and even simple move of a button takes a weekend. Let alone making this code resolution-independent.

MitchBradley, 2023-03-09:
The TFT-aware firmware from MKS appears to require a "big application" partitioning scheme that is not compatible with OTA firmware updates.  To deploy that, we would need to rethink the way we do partitioning, possibly lose OTA, and possibly end up with a explosion of different installation choices.  All of those would cause serious support headaches.

PatrikRindlisbacher, 2023-03-09:
Many controls that are viewed as one from the outside. Have more than one CPU inside for different tasks.

FLUID NC must control the machine. :-)

But it's a few people's turn to solve the operation with additional hardware.

PatrikRindlisbacher, 2023-03-09:
My wish would be to open a department in the **Discord** that takes care of the external operation of FluidNC.

elf128, 2023-03-09:
@MitchBradley , MKS code is not optimal for size in any way. I hope I can fit it without going "big". 
Also, we don't need everything.  Just having status screen ( which in a way present already on machines with OLED screen ) and jog control would cover 99% of the stuff you need a screen for. 
Rest is implemented already through WEB.

MitchBradley, 2023-03-09:
Duplicating the OLED status display on the LCD would be a good start.  I wonder if there is some way to save space by factoring out the font infrastructure so that it does not have to be duplicated between the SSD1306 library and whatever the LCD support uses?  I would be pleased if that library could be restructured/rewritten to support better re-use of its interior components.

MitchBradley, 2023-03-09:
... but maybe I am overthinking it.  The fonts in the SSD1306 library use about 2700 bytes, and some - perhaps all - of them are probably too small to be of much use on a bigger display.

Skorpi08, 2023-03-09:
[image]

elf128, 2023-03-09:
Those are 1bit fonts, which are good for saving memory, but might be a bit of overkill, cause for 16bit screen you have to unpack it on the fly. Not sure it's a good idea.

LVGL has it's own font system, which I'm not a huge fan of, but it is what it is. They're not small.

MitchBradley, 2023-03-09:
The size of the .c files is not the same as the size of the compiled object code.  C source often represents a single byte as a string like "0xFA, ", plus boilerplate, so .c files are often at least 6 times larger than the object code.

elf128, 2023-03-09:
Also, store of a binary data in a C file is a questionable approach.

MitchBradley, 2023-03-09:
Storing binary data in a .c file is just not done.  It is always represented in ascii form then compiled to binary.

Skorpi08, 2023-03-09:
is this compiled files?
[image]

elf128, 2023-03-09:
Which is a questionable approach. 
I would prefer to keep binary data as binary data, build obj from it and link directly.
Otherwise in theory you need a compile step when you binary data is built into .c which require external tool to be part of build system.

MitchBradley, 2023-03-09:
.d is a dependency file.  .o is the compiled file but it also contains symbol information so the linker knows how to connect the binary information into the program.  To find the size of the actual binary data in a .o file, you can run the command "size foo.o" - but the "size" command might not be easy to find on a given system.  In a platformio installation, it is buried in a toolchain directory somewhere in the platformio data directory.

You can represent a monochrome, fixed-width, bitmapped font glyph in about 50 bytes or less.  If you only have glyphs for the 95 printable ASCII characters, you are looking at less than 5K.  Unpacking on the fly is trivial and fast compared to sending data over I2C.

MitchBradley, 2023-03-09:
> I would prefer to keep binary data as binary data, build obj from it and link directly. Otherwise in theory you need a compile step when you binary data is built into .c which require external tool to be part of build system.

In practice nobody ever does it that way, at least in the context of C toolchains.  For stuff like fonts, the binary data hardly ever changes so the "convert to .c" step is often done exactly once and the result is checked in to the source control system, never to be modified.  There is a Linux tool "xxd -i" that will generate a suitable .c file from a binary file.

This is a solved problem with a lot of history.  Please can we not waste time reinventing an ancient wheel?

Skorpi08, 2023-03-09:
Fonts and Pics can be used from SD.

elf128, 2023-03-09:
The screen in question use SPI and most of the operations are sending command to set memory region and then pushing a bunch of pixels. The amount of data you have to push for those 16bit LCDs is much bigger then monochrome OLED, so you set it up as DMA transfer, otherwise you ending up with slideshow. So you render you font into memory buffer and then DMA it to the screen. That's also what LVGL does. But then it has it's own font system and they have variable-size fonts with Unicode support.

I wrote systems like this in the past and it's a lot of work. Especially when you start to worry about symbols outside of ASCII, like degree symbol ° for temperatures, euro-sign and so on and so forth. And then localization people knock your door and.... well. 

So for this particular case, I would rather take existing library which cover all those corner-cases already and optimized for embedded.  Plus you can just keep it as dependency outside of your code base.

elf128, 2023-03-09:
> In practice nobody ever does it that way, at least in the context of C toolchains. For stuff like fonts, the binary data hardly ever changes so the "convert to .c" step is often done exactly once [...]

For fonts, yes. For Images - i disagree, but would not insist.

elf128, 2023-03-09:
> Fonts and Pics can be used from SD.

This is a bad idea. Your machine will simply not boot without card, card removal for g-code upload will most likely cause crash. And also, like I mention above. You have to change LVGL for that, which mean that from this moment entire LVGL has to be moved into codebase and someone have to support it.

Skorpi08, 2023-03-09:
those who want to use the display will then have to put up with it.

MitchBradley, 2023-03-10:
The more I think about it, the less I like the idea of adding additional UI code to FluidNC.  The way to handle this opportunity is to add an extra MCU to run the UI.  ESP32-S2 modules in the mini form factor are going for about $2 on AliExpress  ($20 for 10, plus $3 shipping).  You can get with or without PSRAM for essentially the same price.

So the thing to do is design a little interface board - either as a 6-pack module or not - that adapts the mini pinout to whatever LCD/touchscreen you want.  The "standard" interface that MKS uses would be a reasonable choice.  The UI processor can talk to FluidNC over any channel you want - uart, websocket, telnet, bluetooth, ....  If you want to use WiFi without an external AP, the UI processor can run the AP code so FluidNC is isolated from the flakiness that results from running the FluidNC ESP32 in AP mode.

If I were doing this, I would be tempted to do the entire UI in MicroPython.  LVGL has MicroPython bindings, and there is also https://github.com/peterhinch/micropython-micro-gui .

elf128, 2023-03-10:
I understand your reasoning and I do see sense in it. 
But look at this from customer perspective.  You know that people want. They are not looking at this as an upgrade of machine without LCD. They just want to buy the board with screen, upload firmware and expect it to "just work". Or maybe they already have whole machine assembled with the screen and doesn't want to rebuild it. 

You will loose people right at the line: "design a little interface board". They expect plug'n'play solution. That board already comes pre-flashed and work with the screen out of the box. So, no one will expect any less.

One more time, I'm not trying to convince anyone. I totally see your perspective. 

Anyway, I started implementation of a screen mid-January just for myself, simply cause I have build machine based on that board a while ago. Then I notice bugs, failed miserably to fix them in their codebase and switched to FluidNC. Now, everything is running, but having black screen all the time is less then ideal. The screen is running already with text, pannels, layouts and buttons. It's just doesn't do much. Ones I'll have anything useful we can talk and look at the numbers, otherwise right now it's just a lot of hand waving. 

BTW, I found this thread yesterday by total accident. I think I've found it too early.

msy021, 2023-03-10:
>

I am looking forward to your release.

MitchBradley, 2023-03-10:
I am not expecting random users to design a board - I am expecting you or someone like you to do it.  "Customers" is an interesting word in light of the fact that the vast majority of users do not pay for FluidNC and do not donate to its support - but they do seem to expect a substantial amount of that support. The small number of core developers can barely get any work done.

elf128, 2023-03-10:
It's actually a good point about definition of a "Customer". 

Well, if one would like to go the path of a screen with dedicated CPU, I would note, that AliExpress has a selection of LCDs with ESP32 on the same board for about $20. They already have populated pins for various peripherals. Not sure it's right to post links here. Search for "esp32 lcd touch screen" you'll have all sizes and colors. 

You can connect it to first 4 pins in EXP1 where you already have 2 io, 5V and ground and there you have it.

MitchBradley, 2023-03-10:
Yep.  It all boils down to "who is going to take responsibility for support", especially considering that UIs attract a lot of "can we just have this one extra feature, it should be easy but I am not a programmer so please can you do the work for free?".  China, Inc is really adept at not providing support.

elf128, 2023-03-10:
This is exactly why I wanted to define UI the same way you guys decided to define entire machine. It's kinda brilliant idea I like about FluidNC to have an external definition in yaml file. 
I wanted to define whole UI like this, starting with transport definition, pins. Then initialisation sequences for LCD driver,  and all the way to control definition, pannels, buttons events. 

This way UI in the firmware become a simple state machine which just follows config definition and doesn't have to be changed every single time users ask to add a button. You don't even have to recompile FW. 

Once it's done - users can mess around UI all day long, making new screens, layouts, adding features and so on and don't bother core team.

MitchBradley, 2023-03-10:
You can certainly define a UI in a yaml file, but don't expect us to integrate it with the FluidNC yaml file.  We just don't have the bandwidth to maintain and support all that stuff.  FluidNC has a defined channel interface so you can attach whatever you want.

Feel free to crib parsing, pin definition and other code from FluidNC if it helps.  It might not.

I just went through the exercise of updating a Marlin config for an old 3D printer.  The number of UI options that are in the Marlin config file is just out of control.  We have nowhere near the funding or the team size to get into that realm.  Go wild, but do it in your own project.  Keep things modular - FluidNC will concentrate on motion and machine control, leaving others to handle UI refinements.  The channel interface is the connection.

MitchBradley, 2023-03-10:
Nextion displays have a UI builder and I think LVGL might have one too.

elf128, 2023-03-10:
LVGL has that tool, but I would rather not use it cause the output is just a wall of plain-C.

solawc, 2023-04-15:
> 

I have added the LCD control UI for MKS DLC32, but in fact, I can briefly introduce it:

1. Adding a set of MKS like UI to add icons and font libraries is not difficult using LVGL, and the implementation logic is relatively clear. I will handle it independently using a FreeRTOS task. In fact, the FreeRTOS of ESP32 is very powerful, but in ESP32_ GRBL, I personally believe it has not played the role of FreeRTOS, such as IPC communication.

2. But in the end, due to my personal programming skills, after I joined the UI, the highest processing speed will become slower, which means high-speed engraving will fail. Therefore, I designed a serial screen program for MKS DLC32 and DLC using ESP32, and I found that this not only ensures ESP32_ In addition to the performance of GRBL or FluidNC, the UI experience is also great.

solawc, 2023-04-15:
> I have added the LCD control UI for MKS DLC32, but in fact, I can briefly introduce it:
>
> 1. Adding a set of MKS like UI to add icons and font libraries is not difficult using LVGL, and the [...]

Of course, there are still many areas for optimization in the program. Recently, I have been researching the data flow of FluidNC, and I need to understand the direction of serial port data so that I can add a serial port screen better

macmimy, 2023-05-29:
no updates :(

bdizzle01, 2023-08-10:
Any Updates? I saw in the wiki "LCD TouchScreen data is available to be ported into a special "only MKS DLC32" build." and didnt know if that meant that there is a fluidnc that supports the LCD.

MitchBradley, 2023-08-10:
There is no LCD version of FluudNC from the core developers.

MitchBradley, 2024-01-03:
We are very unlikely to do this.  The developers are working on a full-featured pendant based on an M5Dial, which will work with a variety of different controller boards, not just the MKS DLC32.

souhail-abdallah, 2024-01-28:
> FYI, I do have a slightly "hacky" implementation of LCD on MKS DLC32. Screen and touch works. Once I'll have UI running I'll share it.
>
> However, please don't expect it to happen next week. I agree [...]

Hello professor, I would like some help, please
I have a problem with the MKS DLC32 card, which is that I cannot reach slightly high speeds with it. Is there a software solution, or perhaps an electronic board to amplify or accelerate the signal perhaps?

souhail-abdallah, 2024-01-28:
> We are very unlikely to do this. The developers are working on a full-featured pendant based on an M5Dial, which will work with a variety of different controller boards, not just the MKS DLC32.

souhail-abdallah, 2024-01-28:
Hello professor, I would like some help, please
I have a problem with the MKS DLC32 card, which is that I cannot reach slightly high speeds with it. Is there a software solution, or perhaps an electronic board to amplify or accelerate the signal perhaps?

flixmyswitch, 2024-07-03:
Reading through this. Is there anything working. I have seen one YouTube video with a TFT showing long box buttons.

MitchBradley, 2024-07-03:
There is no support for TS35-R LCD display, none of the primary developers are working on it, and if someone else were to write the code, we would not add it to our mainline code base because it would require more FLASH storage that is available on most of the controllers that we support.  The pendant solutions that we support are described at http://wiki.fluidnc.com/en/hardware/official/M5Dial_Pendant  and http://wiki.fluidnc.com/en/hardware/official/CYD_Dial_Pendant .

closed 2024-07-03 by MitchBradley