bdring/FluidNC #760: spinup_ms and down is not work right.

RDS356, 2022-12-28:
### Controller Board

Self made 4 axis and analog out 0-10.
[image]

### Machine Description

Router with vfd 400hz and spindle 2.2 kW.

### Configuration file

```yaml
name: "RDS CNC"
board: "RDS ESP32 V.1.1"
# Oct 2022 by Rei

kinematics:
  Cartesian:

stepping:
  engine: I2S_STATIC
  idle_ms: 255
  dir_delay_us: 4
  pulse_us: 4
  disable_delay_us: 0
  
axes:
  shared_stepper_disable_pin: gpio.27
  x:
    steps_per_mm: 80.000
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 255.000
    max_travel_mm: 500.000
    soft_limits: true
    homing:
      cycle: 0
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 1000.000
      settle_ms: 100
      seek_scaler: 1.000
      feed_scaler: 1.000

    motor0:      
      limit_neg_pin: gpio.33      
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
       step_pin: I2SO.1
       direction_pin: I2SO.2:low
       
  y:
    steps_per_mm: 80.000
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 255.000
    max_travel_mm: 500.000
    soft_limits: true
    homing:
      cycle: 0
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 1000.000
      settle_ms: 100
      seek_scaler: 1.000
      feed_scaler: 1.000

    motor0:      
      limit_neg_pin: gpio.32      
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
       step_pin: I2SO.3
       direction_pin: I2SO.4:low
  
  z:
    steps_per_mm: 80.000
    max_rate_mm_per_min: 800.000
    acceleration_mm_per_sec2: 255.000
    max_travel_mm: 80.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 1000.000
      settle_ms: 100
      seek_scaler: 1.000
      feed_scaler: 1.000

    motor0:      
      limit_neg_pin: gpio.35    
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
       step_pin: I2SO.5
       direction_pin: I2SO.6:low
  
  a:
    steps_per_mm: 80.000
    max_rate_mm_per_min: 2000.000
    acceleration_mm_per_sec2: 255.000
    max_travel_mm: 500.000
    soft_limits: true
    homing:
      cycle: 0
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 1000.000
      settle_ms: 100
      seek_scaler: 1.000
      feed_scaler: 1.000

    motor0:      
      limit_neg_pin: gpio.34
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
       step_pin: I2SO.7
       direction_pin: I2SO.0:low
       
i2so:
  ws_pin: gpio.12
  bck_pin: gpio.13
  data_pin: gpio.14  

#spi:
#  miso_pin: gpio.19
#  mosi_pin: gpio.23
#  sck_pin: gpio.18

#sdcard:
#  cs_pin: gpio.5
#  card_detect_pin: NO_PIN

control:
  safety_door_pin: gpio.25:low
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:

coolant:
  flood_pin: gpio.15
  mist_pin: gpio.21
  delay_ms: 0

probe:
  pin: gpio.26
  check_mode_start: true
  
PWM:
  pwm_hz: 5000
  output_pin: gpio.2
  enable_pin: gpio.4
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 6000
  spindown_ms: 6000
  tool_num: 0
  speed_map: 0=0% 24000=100%

start:
  must_home: false
  deactivate_parking: true
  check_limits: true
```

### Startup Messages

```text
m3s8000
```

### User Interface Software

bcnc

### What happened?

Simple, the acceleration ramp configs in VFDs  is note proportional with frequency  setuped.(frequency is relative with rotation).
Eg.:  if in VFD the configurations in acceleration ramp is 6 seconds, no matter the frequency , the time to came the frequency setuped will be the same, in this case 6 seconds.
But the FluidNC make a proportional time with speed, it is wrong.
How must be ?
- If the value in spinup_ms is 6000, after m3, only m3, FluidNC must wait 6 seconds before start rest of gcode.

The same in spindown.

Regards, 
Reinaldo de Souza

bdring, 2022-12-28:
What VFD do you have?

RDS356, 2022-12-28:
Weg cfw300.
Look P100 and P101.
https://static.weg.net/medias/downloadcenter/h51/h9c/WEG-CFW300-programming-manual-10007849716-en.pdf

I'm CNC builder and all VFD that I knew works like that.

bdring, 2022-12-28:
All of my VFDs are proportional. You are the first to ask for this change. Most simple PWM spindles would be proportional due to the physics involved.

I suggest you add `G4 P6` to your gcode and set the `spinup_ms:` to 0.

RDS356, 2022-12-28:
> All of my VFDs are proportional. You are the first to ask for this change. Most simple PWM spindles would be proportional due to the physics involved.
> 
> I suggest you add `G4 P6` to your gcode and set the `spinup_ms:` to 0.
Yes, I did. But in my opinion is need change this  to make FluidNC better and more compatibile.

Look, if you set your VFD to 400hz max, when 400hz = 24000rpm and 5 second Acceleration Time. Why running 12000rpm(200 hz) the acceleration is 2.5 Secs ? I disagre, there is no sense cause this resource is used to keep the bearing life spam.

If the acceleration ramp is 5 seconds ever will be 5 seconds to came any rpm/frequency.

I suggest make this correction.
Thank you.

bdring, 2022-12-28:
Acceleration is a rate, not a fixed time (ask Newton). Here is the manual from the Huanyang, which is probably the most popular VFD for DIY users.

[image]

We cannot change just for you. You could add another config item or a new spindle type to the firmware. We are not fond of rarely used config items though.

RDS356, 2022-12-29:
Hi..
I'm Sorry, you are right.
Yes, all VFD is proportional acceleration time.  Off course, is pure math.
Sorry to open this unnecessary issue.
Thank you.

RDS356, 2023-01-03:
I'm re opening this issue to make just one question. Not about Spindle, but laser.
In my board i have a PWM to ANOLOG output 0-5 Volts to control my laser power supply. When the job start ,  is needed a little time(around 50ms) to analogic out stabilsh.  Without this the drawing or cut start after the point. So, my question is;
Is possible create a new parameter like below ?
PWM:
  pwm_hz: 5000
  output_pin: gpio.2
  enable_pin: gpio.4
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  proportional_up_down: false   <=== default true
  spinup_ms: 50
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0% 100=100%

bdring, 2023-01-03:
No, you should use the G4 command in your gcode add specific delays

MitchBradley, 2023-01-03:
Also, you might consider using "laser:" instead of "pwm:".  If you use laser:, the system will adjust the power to compensate for the speed of motion during acceleration, deceleration, and cornering.  That can prevent burn marks at the ends of line segments when the motion is slower.

bdring, 2023-01-03:
It sounds like the analog control is too slow to respond to any real time speed changes.

RDS356, 2023-01-03:
> [image]

RDS356, 2023-01-03:
> Also, you might consider using "laser:" instead of "pwm:". If you use laser:, the system will adjust the power to compensate for the speed of motion during acceleration, deceleration, and cornering. That can prevent burn marks at the ends of line segments when the motion is slower.

No, the problem is really PWM to ANALOG conversion. Need a delay. I'm using this circuit, works very well..

RDS356, 2023-01-03:
[image]

RDS356, 2023-01-03:
Vout in arduino (PWM 5 Volts) is 0-10 in esp32 is 0-7 Volts.

bdring, 2023-01-03:
You could change the gain and the time constant of that circuit.
Google that.

RDS356, 2023-01-03:
> You could change the gain and the time constant of that circuit. Google that.

To solve this ? I think no. Tomorrow I will put pictures here showing this problem.

MitchBradley, 2023-01-04:
To smooth out a 5Khz pwm signal (200 us period), a time constant of 1 ms is more than adequate.  Your circuit has a TC of about 50 ms.  I don't know how fast the rest of the power supply can respond, but your PWM to voltage converter is really slow, much slower than is typical.

MitchBradley, 2023-01-04:
For controlling a real spindle, there is some advantage to a long time constant, to prevent it from starting or stopping too fast and stressing the mechanical system.  For example, I have a spindle with a threaded tool holder that will sometimes spin off the threads if you stop too fast.  But for a laser without mechanical inertia, it is often (usually) better to have faster control.  Many lasers can adjust the power in a millisecond or less.

RDS356, 2023-01-05:
This is my problem..
[image]

RDS356, 2023-01-05:
I don't know how to solve. 
I edit my gcode to run with spindle(picture attached), is just one line, but in this case there is a lot of lines, became impossible the edition.
[image]

RDS356, 2023-01-05:
> To smooth out a 5Khz pwm signal (200 us period), a time constant of 1 ms is more than adequate. Your circuit has a TC of about 50 ms. I don't know how fast the rest of the power supply can respond, but your PWM to voltage converter is really slow, much slower than is typical.

Maybe it is a little slow, but is very precise.See the video below;
Map 0-0% 100-100% - out 0 to 5 Volts.

https://youtu.be/AdrG9grHuFw

bdring, 2023-01-05:
You have a hardware problem. It likely causes turn off problems as well. I see excess power issues at the ends of your lines. Delays cannot help this. You should be able to change the timing without affecting precision.

Our current spindle and G4 delays were never intended for this level of timing accuracy.

MitchBradley, 2023-01-05:
If you reduce the capacitor from 10 uF to 0.47uF, the speed will increase without affecting the precision. With 0.47 uF, your PWM to voltage circuit would respond in roughly 2.5 ms.  You could use 0.22 uF to get the response time down to about 1 ms at the expense of a little more ripple on the output voltage. Ripple typically does not cause problems for laser power modulation.

You are asking us to add a feature to FluidNC to compensate for a bad design choice in your circuit. We are not going to do that. It is always better to fix a problem at its source, instead of hiding the problem with software warts.

RDS356, 2023-01-05:
Hi..
Before your answers I remade the RC calculate. C=2.2uF reduced the time in almost five times. Now is working much better.(picture attached).  My laser is 80 Watts , the upper write is 12% and downer 10%.
I will try with the values that you suggested.
Thank you for help.

[image]

closed 2023-01-06 by RDS356