bdring/FluidNC #762: Occasional freeze while running

deranku, 2022-12-30:
### Controller Board

Pen/Laser Controller SPI 

### Machine Description

String art machine with external stepper driver on x, solenoid on y and servo on z.

### Configuration file

```yaml
name: "Stringer"
board: "FluidNC Pen/Laser"

start:
  must_home: false
  deactivate_parking: false

stepping:
  engine: RMT
  # 255: dauerhaft an
  idle_ms: 255
  # dir_delay_us: 5 us
  dir_delay_us: 10
  # pulse_us: not less than 2.5 us
  pulse_us: 3
  # Some motors need a delay from when they are enabled to when they can take the first step.: at least 5 us
  disable_delay_us: 10

axes:
  shared_stepper_disable_pin: gpio.13
  
  # Reifen/Stepper
  x: 
    # actually steps per nail
    # 37.5 steps for 256 nails (600 mm ring)
    # 30   steps for 320 nails (800 mm ring)
    # 32   steps for 300 nails (800 mm ring)
    steps_per_mm: 32
    max_rate_mm_per_min: 17000
    acceleration_mm_per_sec2: 500
    max_travel_mm: 1000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      allow_single_axis: true
      mpos_mm: 0
      seek_mm_per_min: 1500.000
      feed_mm_per_min: 100.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100
    
    motor0:
      limit_pos_pin: gpio.15:pu
      hard_limits: false
      pulloff_mm: 2.0
      standard_stepper:
        step_pin: gpio.14
        direction_pin: gpio.12

  # Fädler/Solenoid
  y:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 2000.000
    acceleration_mm_per_sec2: 500.000
    max_travel_mm: 5.000
    soft_limits: false
    homing:
      cycle: 0
      positive_direction: true
      mpos_mm: 0
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      solenoid:
        output_pin: gpio.27
        pwm_hz: 1000
        off_percent: 0.000
        pull_percent: 100.000
        hold_percent: 100.000
        pull_ms: 100

  # Bohrer/Servo
  z:
    steps_per_mm: 80
    max_rate_mm_per_min: 1500
    acceleration_mm_per_sec2: 100
    max_travel_mm: 40
    soft_limits: false
    homing:
      cycle: 0
      positive_direction: true
      mpos_mm: 2
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100
    
    motor0:
      rc_servo:
        pwm_hz: 50
        output_pin: gpio.22
        min_pulse_us: 540
        max_pulse_us: 2470

relay:
  output_pin: gpio.2
  spinup_ms: 500
  spindown_ms: 1000
  tool_num: 0
  speed_map: 0=0.000% 0=100.000% 1=100.000%

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

control:
  # feed_hold_pin: gpio.32:low
  cycle_start_pin: gpio.33:low:pu  # no pu for 3.4.4
  safety_door_pin: gpio.32:low:pu  # no pu for 3.4.4
  # reset_pin: NO_PIN
```

### Startup Messages

```text
$ss
[MSG:INFO: FluidNC v3.6.5]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:stringer_config.yaml]
[MSG:INFO: Machine Stringer]
[MSG:INFO: Board FluidNC Pen/Laser]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:3us Dsbl Delay:10us Dir Delay:10us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.13]
[MSG:INFO: Axis X (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.14 Dir:gpio.12 Disable:NO_PIN]
[MSG:INFO:  X Pos Limit gpio.15:pu]
[MSG:INFO: Axis Y (-5.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     solenoid Pin: gpio.27 Off: 0.000 Hold: 100.000 Pull:100.000 Duration:100 pwm hz:1000 period:1000]
[MSG:INFO: Axis Z (-38.000,2.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     rc_servo Pin:gpio.22 Pulse Len(540,2470 period:1048575)]
[MSG:INFO: safety_door_pin gpio.32:low:pu]
[MSG:INFO: cycle_start_pin gpio.33:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Relay Spindle Ena:NO_PIN Out:gpio.2 Dir:NO_PIN]
[MSG:INFO: Using spindle Relay]
[MSG:INFO: Connecting to STA SSID:Hakaniemi]
[MSG:INFO: Connecting.]
[MSG:INFO: Connected - IP is 192.168.178.28]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://stringer.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
```

### User Interface Software

WebUI

### What happened?

Machine works perfectly under FNC 3.4.4 with WebUI only. Exept for crashes, when the WebUI went away. I saw that this was fixed in 3.6.5, so I updated firmware and WebUI.

Under both versions of FNC, I ran the same G-Code file, which is around 400 KB.

Under 3.6.5, the WebUI going away does not seem like a problem anymore. But three times during that run, the machine stopped hard, without deceleration, and was dead for around 20 seconds. After that, the program continued, but position on x (stepper) was lost.

The G-Code used is output by a modification of Barts python script for the string art machine. It looks like this:

```
G90
G21
G0 Y0.00
G10 L20 P0 X264

(MSG Starting line 0)
(Go from 0 to 264)
G0 X264.50 Y0.00
G4 P0.1
G0 Y5.00
G1 X263.50 F500
G0 Y0.00
G4 P0.2

(MSG Starting line 1)
(Go from 264 to 184)
G0 X184.50 Y0.00
G4 P0.1
G0 Y5.00
G1 X183.50 F500
G0 Y0.00
G4 P0.2

(MSG Starting line 2)
(Go from 184 to 27)
G0 X327.50 Y0.00
G4 P0.1
G0 Y5.00
G1 X326.50 F500
G0 Y0.00
G4 P0.2

(MSG Starting line 3)
(Go from 27 to 179)
G0 X179.50 Y0.00
G4 P0.1
G0 Y5.00
G1 X178.50 F500
G0 Y0.00
G4 P0.2
```
and so on. 

Exept for the firmware update, nothing was changed.

### Other Information

More of a sidenote: under 3.6.5, `cycle_start_pin` and `safety_door_pin` need to have Pull Ups activated. But not under 3.4.4. That's odd.

bdring, 2022-12-31:
Are you always running via the WebUI? Can you try running via a serial port sender or by starting an SD file using FluidTerm. Do not have the WebUI loaded anywhere.

This could help narrow down the problem.

deranku, 2023-01-01:
Yes, WebUI only. I'll try and report back 👍

bdring, 2023-01-01:
It is possible something is causing the WebUI to reload during a job, like a refresh from a wakeup, etc. That can mess up the timing.

The next version will block that.

https://discord.com/channels/780079161460916227/1055694404494561320/1058813451566334043

deranku, 2023-01-02:
I ran the same G-Code file from the SD card over Telnet: `$SD/Run=/pinguini4-300-3500-knitter.nc` WebUI was not loaded. 

A freeze occured again, but the stepper motor was not interrupted during motion. Maybe a coincidence. It happened at a random position. I noticed that a couple of lines were skipped on the console, but I can't tell if they were really skipped or just not sent to the console.

```
[MSG:INFO: GCode Comment...Starting line 380]
[MSG:INFO: GCode Comment...Starting line 381]
[MSG:INFO: GCode Comment...Starting line 386]
[MSG:INFO: GCode Comment...Starting line 387]
```

I put the computer to sleep a couple of times, but that did not cause the freeze. It happened while the computer was asleep, though.

MitchBradley, 2023-01-02:
It would be nice to have a complete copy of the gcode file that causes the problem

deranku, 2023-01-02:
You are right Mitch, here is the file:
[pinguini4-300-3500-knitter.nc.zip](https://github.com/bdring/FluidNC/files/10333230/pinguini4-300-3500-knitter.nc.zip)

deranku, 2023-01-27:
Tested under 3.6.6 (WebUI) with the same issue. Ran with the attached GCode.
[drill_300_holes.nc.zip](https://github.com/bdring/FluidNC/files/10518968/drill_300_holes.nc.zip)

villamany, 2023-01-28:
Hi, im newbe using FluidNC, but after first tests same issues here.
I use a MKS DLC32 controller with a laser diode.
-If if launch a SD file using WebUI, after some time i start to get freezings but if select the radioButton "GRBL Report None" file continue without issues, also using the "poll every 3s" seems to be ok. 
The issue seems to be a buffer overflow or similar getting the reports at 50ms.

Some more notes:
-If after launch the flile i close the Web UI also file continues without freezers.
-But if i open again the Web page while runing the ESP hangs.
-Take care the "Auto" radioButton is automatically selected after lauching a file if you have selected the "none" radio so you need to select it again or simply leave it in "poll" .

The issue leaves the laser head stopped and powered ON so you can get fire.

Attached a file for testings.
[ojo100.zip](https://github.com/bdring/FluidNC/files/10528135/ojo100.zip)

MitchBradley, 2023-01-28:
What happens if you increase the auto-poll interval to, say, 500 ms?

villamany, 2023-01-28:
Nice, seems to be the autoPoll mode.
After some test the same file at Auto Poll 4000ms still freezing. While if i use simple Poll @100ms (0.1s) is ok. I think you will noticed because the output Web console lines in autoPoll mode are not time constant while using simple poll are constant.

lkarlslund, 2023-10-03:
FYI: Same problem here with 3.7.8, intermittent hangs and I lose steps. Not sure when it started happening, but it seems like it was a recent firmware upgrade. I'm doing all jobs via upload+WebUI run. I'll try simple poll and see if it changes anything.