bdring/FluidNC #763: Problem: Flood Coolant toggle doesn't work

balthisar, 2022-12-31:
### Controller Board

MKS TinyBee

### Machine Description

n/a - controls related

### Input Circuits

```yaml
n/a
```

### Configuration file

```yaml
# HE0 - i2so.17 - mist enable (dust) M7
# HE1 - i2so.18 - flood enable (coolant) M8
coolant:
  mist_pin: i2so.17
  flood_pin: i2so.18
  delay_ms: 0
```

### Startup Messages

```text
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
$SS
[MSG:INFO: FluidNC v3.6.5]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:DBG: Running after-parse tasks]
[MSG:DBG: Checking configuration]
[MSG:INFO: Machine Queenbee Pro]
[MSG:INFO: Board MKS TinyBee V1.0]
[MSG:INFO: I2SO BCK:gpio.25 WS:gpio.26 DATA:gpio.27]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:I2S_static Pulse:10us Dsbl Delay:2us Dir Delay:1us Idle Delay:25ms]
[MSG:INFO: User Digital Output:0 on Pin:I2SO.16]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,750.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.1 Dir:I2SO.2 Disable:I2SO.0]
[MSG:INFO:  X Neg Limit gpio.33:low:pu]
[MSG:DBG:  X Neg Limit 0]
[MSG:INFO: Axis Y (0.000,1260.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.4 Dir:I2SO.5 Disable:I2SO.3]
[MSG:INFO:  Y Neg Limit gpio.32:low:pu]
[MSG:DBG:  Y Neg Limit 0]
[MSG:INFO:   Motor1]
[MSG:INFO:     standard_stepper Step:I2SO.13 Dir:I2SO.14 Disable:I2SO.12]
[MSG:INFO: Axis Z (-130.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.7 Dir:I2SO.8 Disable:I2SO.6]
[MSG:INFO:  Z Pos Limit gpio.22:low:pu]
[MSG:DBG:  Z Pos Limit 0]
[MSG:INFO: reset_pin gpio.39:low]
[MSG:INFO: feed_hold_pin gpio.34:low]
[MSG:INFO: cycle_start_pin gpio.36:low]
[MSG:INFO: macro0_pin gpio.16:low:pu]
[MSG:INFO: macro1_pin gpio.17:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:gpio.21 Out:gpio.4 Dir:I2SO.21 Freq:2500Hz Period:16383]
[MSG:INFO: Laser Ena:NO_PIN Out:gpio.2:pd Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Flood coolant I2SO.18]
[MSG:INFO: Mist coolant I2SO.17]
[MSG:INFO: Probe Pin: gpio.35:low]
[MSG:INFO: Connecting to STA SSID:DerryNet]
[MSG:INFO: Connecting.]
[MSG:INFO: Connected - IP is 192.168.1.109]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
```

### User Interface Software

WebUI

### What happened?

When pressing Flood in the WebUI, the designated output does not toggle.

Notes:

- Pressing Mist in the WebUI toggles correctly.
- Issuing M8 turns on the Flood output correctly.
- Issuing M9 turns off both outputs correctly.
- M7, of course, turns on Mist appropriately.
- Only "Flood" does not work.
- Outputs enabled/disabled verified via red LED on board and o'scope on terminals.

The WebUI correctly tries to issue `SendRealtimeCmd(0xA0);`; however it has no effect.

Using 0xA1 as a macro works properly for Mist (added via binary editor).
Using 0XA0 as a macro doesn't not work at all.

FluidNc is latest version as shown in `$SS` output, above.

Because M8 works, my assumption is that this is a FluidNc bug.

MitchBradley, 2023-01-03:
Can you try with FluidTerm - CTRL-O ft  and CTRL-O mt

balthisar, 2023-01-03:
Ctrl-O doesn't seem to do anything.

~~~
--- FluidTerm v1.2.0 on /dev/cu.usbserial-1410  115200,8,N,1 ---
--- Quit: Ctrl+] or Ctrl+Q | Upload: Ctrl+U | Reset: Ctrl+R | ClearScreen: Ctrl+W ---
mt
[MSG:DBG: Bad GCode: MT]
error:2
~~~

I also tried FluidTerm from its dedicated repo, although it has older latest commit dates:

~~~
--- Fluidterm on /dev/cu.usbserial-1410  115200,8,N,1 ---
--- Quit: Ctrl+] or Ctrl+Q | Menu: Ctrl+T | Help: Ctrl+T followed by 'H' ---
mt
[MSG:DBG: Bad GCode: MT]
~~~

A quick view through the source doesn't seem to indicate that Ctrl-O is mapped to anything, and nothing else seems to suggest that an escape sequence is setup in FluidTerm.

My default python 3:

~~~
/usr/bin/env python3      
Python 3.10.3 (main, May  5 2022, 21:55:55) [Clang 13.1.6 (clang-1316.0.21.2.3)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
~~~

Shell is zsh, and it's not configured to intercept Ctrl-O for any other purpose.

MitchBradley, 2023-01-03:
Only the Windows native version of FluidTerm supports the ctrl-o commands.  I don't spend much effort maintaining the Python version because comparatively few people use it.

balthisar, 2023-01-03:
~~~
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100|A:FM>
~~~

From the Windows version of fluidterm, the report indicates that the Ctrl-O commands both work. As this is a Windows VM nowhere near my lab, I didn't verify with the o'scope, and the board's LED's don't illuminate running on USB power. Generally the report output is reliable, so I'd suggest that they work, and it's only the WebUI that's not working.

MitchBradley, 2023-01-03:
String.fromCharCode(0xa0) returns ' ' i.e. a string containing the blank character.  I need to find some different way of transforming from binary.

MitchBradley, 2023-01-03:
Actually, the return value of fromCharCode(0xa0) is a string containing a non-breaking space, which would be sent properly as the UTF-8 sequence 0xc3, 0xa0 - except that the string input to SendPrinterCommand() is trimmed, and .trim removes spaces, including non-breaking ones.

MitchBradley, 2023-01-03:
I think the solution is to remove .trim() from this line in printercmd.js SendPrinterCommand():
```
    if (cmd.trim().length == 0) return;
```
so it reads 
```
    if (cmd.length == 0) return;
```
I cannot test it because I am traveling with an ancient laptop and I'm in node/npm/gulp version hell on that machine, so when I try to run the gulp package command, I get all sorts of errors about versions being incompatible and certificates expiring and commands missing and ...

MitchBradley, 2023-01-04:
https://github.com/MitchBradley/ESP3D-WEBUI/blob/revamp/index.html.gz might work.  I managed to solve the compilation problem.

erikbuch, 2023-02-06:
Release 3.6.7 fixed the problem for me! 
Thanks alot for your work!