bdring/FluidNC #764: Problem: spindle at maximum RPM starts and in half a second goes off

axsmyth, 2023-01-06:
### Controller Board

MKS DLC32

### Machine Description

Ordinary MKS DLC32, XYZ, spindle 775 motor 20000rpm 40W @24V

### Input Circuits

```yaml
Standart for MKS DLC32.
```

### Configuration file

```yaml
board: MKS-DLC32
name: CNC3020 Pro MAX

kinematics:
  Cartesian:

stepping:
  engine: I2S_STATIC
  idle_ms: 254
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
axes:
  shared_stepper_disable_pin: I2SO.0
  x:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 1200.000
    acceleration_mm_per_sec2: 80.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.36
      hard_limits: false
      pulloff_mm: 5.000
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2:high

  y:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 1200.000
    acceleration_mm_per_sec2: 80.000
    max_travel_mm: 200.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35
      hard_limits: false
      pulloff_mm: 5.000
      stepstick:
        step_pin: I2SO.5
        direction_pin: I2SO.6:low

  z:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 500.000
    acceleration_mm_per_sec2: 80.000
    max_travel_mm: 50.000
    soft_limits: false
    homing:
      cycle: 0
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34
      hard_limits: false
      pulloff_mm: 5.000
      stepstick:
        step_pin: I2SO.3
        direction_pin: I2SO.4

i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14

sdcard:
  cs_pin: gpio.15
  card_detect_pin: gpio.39

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: gpio.33:low:pu
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

macros:
  startup_line0:
  startup_line1:
  macro0: $SD/Run=lasertest.gcode
  macro1: $SD/Run=home.gcode
  #These are examples
  macro2:
  macro3:

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

probe:
  pin: gpio.22:low:pu
  check_mode_start: true

PWM:
  pwm_hz: 8000
  output_pin: gpio.32
  enable_pin: NO_PIN
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0.000% 20000=100.000%
  off_on_alarm: false

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

start:
  must_home: false
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.6.5]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:tmc2209_AxS.yaml]
[MSG:INFO: Machine CNC3020 Pro MAX]
[MSG:INFO: Board MKS-DLC32]
[MSG:INFO: I2SO BCK:gpio.16 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.14 MOSI:gpio.13 MISO:gpio.12]
[MSG:INFO: SD Card Detect gpio.39]
[MSG:INFO: SD Card cs_pin:gpio.15 detect:gpio.39]
[MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:254ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable I2SO.0]
[MSG:INFO: Axis X (0.000,300.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.1 Dir:I2SO.2 Disable:NO_PIN]
[MSG:INFO:  X Neg Limit gpio.36]
[MSG:INFO: Axis Y (0.000,200.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.5 Dir:I2SO.6:low Disable:NO_PIN]
[MSG:INFO:  Y Neg Limit gpio.35]
[MSG:INFO: Axis Z (0.000,50.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.3 Dir:I2SO.4 Disable:NO_PIN]
[MSG:INFO:  Z Neg Limit gpio.34]
[MSG:INFO: macro0_pin gpio.33:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:NO_PIN Out:gpio.32 Dir:NO_PIN Freq:8000Hz Period:8191]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Probe Pin: gpio.22:low:pu]
[MSG:INFO: Connecting to STA SSID:AxS]
[MSG:INFO: Connecting.]
[MSG:INFO: Connected - IP is 192.168.1.19]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
ok
```

### User Interface Software

WebUI, Candle.

### What happened?

I bought a controller to replace the burned one. My old controller worked with WiFi-adaptor to give me remote access. So I purchased MKS DLC32. Unfortunately, its “telnet” connection has code errors and couldn’t be useful for me.
I had several hard days to setup FNC and tune it, but it successfully works now and I am very happy with it. Thank you!
 Nevertheless, there is an issue when trying to start spindle at maximum RPM. It starts and in half a second goes off. With MKS firmware it started well. In my case max speed is about 20000 RPM (21300 in fact) so I placed in config.yaml “speed_map: 0=0.000% 20000=100.000%”. And spindle refuses to start normally with S > 19000. 
Now I have to turn on the spindle at 19000 first and then run G-code where S=20000.
I couldn’t find out what’s wrong with my settings. Could you help me?

bdring, 2023-01-06:
Check the voltage on the PWM pin. That will validate the firmware.

axsmyth, 2023-01-06:
[image]
That is view of PWM voltage @19k rpm. @20k there is no spur down. The stright line of voltage only at 24V.

bdring, 2023-01-06:
See what actual max rpm value works. Do that then go to the full 20k from the previous value. You may have an over current fault if you go full all at once.

axsmyth, 2023-01-06:
It works fine only lower than 19248 Hz i.e. 19247 works, 19248 doesn't.
There is console log with those attempts. Note that after "M03 S19248" there is no response neither alarm. Motor just stops.
As I described, with MKS firmware the motor started at 100%pwm with no problem. And there was no any board or MOSFET failure. Obviously, the power reserve on the board is quite sufficient.

M05 < ok
M03 S19245 < [GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S19245]
ok
M05 < ok
M03 S19249 < [GC:G0 G54 G17 G21 G90 G94 M3 M9 T0 F0 S19000]
ok
M03 S19248 < ok
M3 S19000 < ok
M5 < [GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S19247]
ok
M03 S19248
That's the end of log.
And console stays working. I could run Candle etc.

axsmyth, 2023-01-06:
Does this indicate that there is overcurrent problem? Indeed, in parallel with this "failure", the console in the Candle works quite normally. And continues to receive commands to turn on or turn off the motor, with the exception of no response to the "failed command".

bdring, 2023-01-06:
Try changing the spindle control to the ***relay*** config item using the pin you have for the PWM as the relay output. This will skip any PWM things and go to full on with any speed value. I suspect it is a current issue, but this will confirm that the firmware is not an issue.

MKS may be ramping up the PWM or something to mitigate the current.

bdring, 2023-01-06:
Also, use Fluidterm to send all commands. Candle might be masking some messages.

axsmyth, 2023-01-06:
Thank you. I'll check that later. I want to say that you assure me there is a overcurrent issue not firmware one. I think I would study G-code to find a pause insertion into sequence to start to max rpm softly.

bdring, 2023-01-06:
That is a common solution. Some people feel it is less stressful on the bearings to start it at a slower RPM for a few seconds to distribute the grease before going full speed.

M3S3000 ; goto 3000RPM
G4P2.5 ; wait 2.5 seconds
S20000 ; goto 20000 RPM

<http://wiki.fluidnc.com/en/features/supported_gcodes#g4-dwell>

axsmyth, 2023-01-06:
That's the way! Thank you. I will definetly check the relay option and give a response. It is nice to have such responsive support.

axsmyth, 2023-01-08:
Sorry, there is a delay for checking ooption describimg above. Main chip of my DLC32 seems to be burned out. I need time to change it with new one. I suspect this could be due to power surges when starting the spindle. Now it does not start at all (no messages over comport) and gets very hot after power on.

alimnette, 2023-01-16:
i have a same issue.
i will test the relay in config section.
but for me i must add rampeup for start the spindle

alimnette, 2023-01-19:
I tested with a soft start, it works.

M03 S5000
G04 P0.5
M03 S8000
G04 P0.5
M03 S11000
G04 P0.5
M03 S15000
G04 P0.5
M03 S17000
G04 P0.5
M03 S20000

axsmyth, 2023-02-20:
I've repared the board and checked all over. It was issue with limited power of power supply. Thank you for duly support.

closed 2023-02-20 by axsmyth