bdring/FluidNC #765: Not working SD card

alvarotorijano, 2023-01-07:
### Controller Board

Bare ESP32+SD card reader / MKS32

### Machine Description

Testing only SD Card
[image]

### Configuration file

```yaml
board: test
name: test

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN
```

### Startup Messages

```text
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12812
load:0x40080400,len:3032
entry 0x400805e4

[MSG:INFO: FluidNC v3.6.5]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Axes: using defaults]
[MSG:INFO: Machine test]
[MSG:INFO: Board test]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle 
Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-1000.000,0.000)]
[MSG:INFO: Axis Y (-1000.000,0.000)]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Connecting to STA SSID:HackFi]
[MSG:INFO: Connecting.]
[MSG:INFO: Connected - IP is 192.168.1.54]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://Cnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.6 [FluidNC v3.6.5 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
```

### User Interface Software

WebUI

### What happened?

There is no way at all to make the SD card to work.
Always returns:
[MSG:ERR: sdmmc_card_init failed code 0x107] 
[MSG:ERR: sdmmc_card_init failed code 0x108]

Both values appear randomly. Also observed 0x109 sometimes and 0x104 more rarely. 
Tried with 3 different sdCards formatted with Fat32. Two of them 32GB and other one of 16GB.

Tried soldering cables directly to an sdcard adapter as described here: https://github.com/bdring/FluidNC/issues/659

Tried with older versions: v3.4.4, v3.5.0, v3.6.0 and Devt branch

Tried with different SPI speeds with 0.1MHz, 1MHz, 10MHz and the default speed 20MHz as described here: https://github.com/bdring/FluidNC/pull/742

The same SD card adapter works with the SD example for ESP32
What am I missing?

alvarotorijano, 2023-01-07:
Also tried this way with no difference
[image]

bdring, 2023-01-07:
Try this

``` Yaml
sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN
  frequency_hz: 10000000
  ```
  
  http://wiki.fluidnc.com/en/config/sd_card

alvarotorijano, 2023-01-07:
Nothing. This is the output:
$G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|WCO:0.000,0.000,0.000>
[MSG:ERR: sdmmc_card_init failed code 0x107]
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100>
[MSG:ERR: sdmmc_card_init failed code 0x107]
<Alarm|MPos:0.000,0.000,0.000|FS:0,0>
[MSG:ERR: sdmmc_card_init failed code 0x107]
[MSG:ERR: sdmmc_card_init failed code 0x107]
[MSG:ERR: sdmmc_card_init failed code 0x107]
[MSG:ERR: sdmmc_card_init failed code 0x106] <--------------- First time seen
<Alarm|MPos:0.000,0.000,0.000|FS:0,0>
[MSG:ERR: sdmmc_card_init failed code 0x108]
[MSG:ERR: sdmmc_card_init failed code 0x107]
[MSG:ERR: sdmmc_card_init failed code 0x108]

keeps failing but a new error is reported. All those messages appears when try to update the SD window on the webUI

bdring, 2023-01-07:
It does not like those adapters. We do not use the same Espressif library as the the Arduino setup for the examples. This library appears to have issues with something. All the hardware on this page with SD cards work fine.

http://wiki.fluidnc.com/en/hardware/existing_hardware

Sorry, but we can't help you much.

alvarotorijano, 2023-01-07:
Works on the MKS DLC32. Could worth it to comment on the wiki that this **** adapters does not work with this firmware. This drove me completely crazy.

bdring, 2023-01-07:
Sure, when I get some time.

pedrolapute, 2023-01-07:
Hi, i had the same issue ! I found that many off my sd want work! Try this, use SD CARD FORMATER for your sd’s and put a capacitor between Gnd and 5V.

Pierre Desroches

[image]<https://user-images.githubusercontent.com/11181850/211125533-2467f750-8372-4b26-ad7e-55d665d8bad3.png>

MitchBradley, 2023-01-07:
Try reducing frequency_hz to 5000000 or even lower to 1000000. Dupont wires between the ESP32 and the SD card degrade the signal, reducing the frequency that will work. The Arduino library uses a low clock frequency which works with a lot of DIY setups but also reduces the performance. We use the ESP-IDF library which defaults to a high clock frequency. That gives high performance with the on board SD adapters that are common on commercial boards. For DIY lashups, you must reduce the clock frequency, often by a lot, to account for poor signal quality.

alvarotorijano, 2023-01-07:
Thank you very much @MitchBradley.
@pedrolapute which capacitor are you using? Could you attach a photo?

Gos2two, 2023-01-07:
Hello!

I actually had the same issue. FluidNC Pen Laser reader works just fine. 

On top of this, I have a question about the library being used to handle SD Card file writing. @MitchBradley @bdring, how do you do SD File writing from the FluidNC side without involving the ESP3D-webui side of things? In other words is ther any writeFile(SD, "/hello.txt", "Hello ") type function implemented. Using SD.h library like you would do on other ESP32 projects intereferes with the SPI pins. 

Thanks

MitchBradley, 2023-01-07:
@Gos2two - please do not hijack an existing issue about SD not working by asking a programming question. Create a separate issue or ask the question on Discord

PatrikRindlisbacher, 2023-02-15:
With my M5Stach Core2 I have not gotten the SD card to work either. Not even with adjustments in the code.
Fehlercode 0x107 Fehlercode 0x108 0x109 0x104 

I had to give up after 7-8 hours
Problem solved with new Bdring 6 CNC card.

PatrikRindlisbacher, 2023-02-19:
[image]

19-Sergey-92, 2023-02-25:
Hi
Currently I have the configuration:
...
spi:
  miso_pin: gpio.21
  mosi_pin: gpio.18
  sck_pin: gpio.19

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN
  frequency_hz: 10000000
...
Everything works, but I'm missing 2-3 GPIO pins.
Can I connect an SD-Card to GPIO6-11, freeing up GPIO 5,18,19,21?

alvarotorijano, 2023-02-26:
SD Card uses an SPI bus so you must use MOSI, MISO, SCK, and CS pins. It is
mandatory to use at least the first three pins BUT it is also possible to
get rid of the CS (Chip select pin) pulling this line down, but it is also
mandatory that your SPI library supports this. I'm not sure if it is
possible using the implementation of the framework but you can try and
report the results.

19-Sergey-92, 2023-02-26:
[MSG:ERR: Configuration error at /machine/spi: ERR: gpio.7 - Unusable GPIO]
[MSG:ERR: Configuration error at /machine/spi: ERR: gpio.10 - Unusable GPIO]
[MSG:ERR: Configuration error at /machine/spi: ERR: gpio.6 - Unusable GPIO]
[MSG:ERR: Configuration error at /machine/sdcard: ERR: gpio.9 - Unusable GPIO]
[MSG:INFO: Configuration is invalid]
[MSG:INFO: Using default configuration]
[MSG:INFO: Axes: using defaults]
[MSG:INFO: Configuration is invalid]

Is there any way to fix it?
I want to configure the SDcard specifically for these GPIOs

19-Sergey-92, 2023-02-26:
...
spi:
  miso_pin: gpio.7
  mosi_pin: gpio.10
  sck_pin: gpio.6

sdcard:
  cs_pin: gpio.9
  card_detect_pin: NO_PIN
  frequency_hz: 10000000
...

MitchBradley, 2023-02-26:
No you cannot use those GPIOs.  The ESP32 module uses them for other purposes.  http://wiki.fluidnc.com/en/hardware/esp32_pin_reference#unusable-gpio
Please do not hijack an existing issue by asking a different question.

closed 2023-02-26 by MitchBradley