bdring/FluidNC #768: Feature:

cncyar, 2023-01-11:
### Machine Context

I connected a Nextion display to control my machine. It works fine for sending commands and running Gcode, but there is one problem - there is no feedback about the operation of the controller and the whole machine as a whole.

### Feature Description

Add an event register (Alarm, Hold, Door, Limits, etc.) as an output for LEDs or other actuators

### Other Approaches

In my case, the lack of indication does not provide information about the state of the machine, and adding this function, for example via I2C or SPI, will allow organizing a full-fledged Offline controller without Wifi connection to WEBui. We press the reset button in the Nextion when the accident LED is on, etc.

### How I Can Help

?

MitchBradley, 2023-01-11:
Our intention is to support displays using the auto reporting capability in the InputThreadMerge branch, which will soon be promoted to the development line.

Any Channel instance (for example a Uart channel, Websocket channel, or Telnet channel) can issue `$Report/Interval=100` to say that, for that channel, reports should be sent automatically on any state change, or at 100 ms intervals during motion.  The reports are either the `<...>` report that is issued in response to a `?`, or, in the case of GCode mode changes, the `[GC:...]` report that is issued in response to `$G`.

A software module to drive a display can create a Uart class and override its write(uint8_t) method (Uart inherits from Channel .. Stream .. Print, and Print::write(uint8_t) is virtual). That overridden method can parse the report strings to do whatever it wants.

For interfaces other than Uart, like I2C or SPI, you could create a suitable class that inherits from Channel and similarly provide a suitable override for write(uint8_t).

wu5, 2023-02-16:
I like the idea of displaying the status (like alarm) on LED lights.

[image]
[image]

bobbybumfluff, 2023-02-16:
> LEDIndicatorLight
> tower
> [ { ***@***.***": "http://schema.org", ***@***.***": "EmailMessage",
> "name": "View Issue" }, "description": "View this Issue on GitHub",

Probably best to invite the developers of FluidNC, to quote you a price 
for writing this code for you. Or you could perhaps code your 
requirement, yourself, for free....😁.

wu5, 2023-02-16:
> Probably best to invite the developers of FluidNC, to quote you a price for writing this code for you. Or you could perhaps code your requirement, yourself, for free.....

Why so toxic?

I do not demand anything from the developers of FluidNC and am grateful to them. I'm just giving my opinion that I like this feature. It would be great if it was ever implemented.

cncyar, 2023-02-16:
Yes. I got the following: I transfer anything from nextion to 5 axes. I receive only status from FluidNC, using an additional Arduino Nano (LEDs). There is no time to make a test board yet. As soon as I have time, I'll post the result here.

Gaab88, 2024-03-21:
Hi!

I enabled the second serial port for nextion in config.yaml:

uart1:
  txd_pin: gpio.22
  rxd_pin: gpio.21
  baud: 115200
  mode: 8N1
uart_channel1:
   uart_num: 1
   report_interval_ms: 75
   all_messages: true

Unfortunately, nothing other than the status of the security door is received on that port. I would like to use nextion as a display and control interface if possible. I found a video showing the Nextion display connected to the ESP32 GRBL. The problem is that I can send commands to fluidnc, but not read from it.  Could you help me? :)

bdring, 2024-03-21:
I recommend using our starter code. There are several examples. 

http://wiki.fluidnc.com/en/hardware/pendants_displays

Gaab88, 2024-03-22:
> I recommend using our starter code. There are several examples.
> 
> http://wiki.fluidnc.com/en/hardware/pendants_displays

Thank you very much for your reply!

DisplayToTerminal fails to compile with Platformio under VScode. Reinstalling did not fix the problem.

bdring, 2024-03-22:
DisplayToTerminal on the main branch compiles for me.

```
 *  Executing task: C:\Users\barto\.platformio\penv\Scripts\platformio.exe run 

Processing megaatmega2560 (platform: atmelavr; board: megaatmega2560; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/megaatmega2560.html
PLATFORM: Atmel AVR (4.2.0) > Arduino Mega or Mega 2560 ATmega2560 (Mega 2560)
HARDWARE: ATMEGA2560 16MHz, 8KB RAM, 248KB Flash
DEBUG: Current (avr-stub) External (avr-stub, simavr)
PACKAGES:
 - framework-arduino-avr @ 5.1.0
 - toolchain-atmelavr @ 1.70300.191015 (7.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 8 compatible libraries
Scanning dependencies...
Dependency Graph
|-- GrblParserC
Building in release mode
Compiling .pio\build\megaatmega2560\src\main.cpp.o
...
...
Compiling .pio\build\megaatmega2560\FrameworkArduino\wiring_shift.c.o
Archiving .pio\build\megaatmega2560\libFrameworkArduino.a
Linking .pio\build\megaatmega2560\firmware.elf
Checking size .pio\build\megaatmega2560\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  23.5% (used 1924 bytes from 8192 bytes)
Flash: [          ]   3.3% (used 8408 bytes from 253952 bytes)
Building .pio\build\megaatmega2560\firmware.hex
```

I did see a few errors on the LessArduino branch.

MitchBradley, 2024-03-22:
Give us exact messages, not vague descriptions.  You said "nothing other than the status of the security door is received on that port".  We need to know complete details.  It is okay that your English is not good. Cut and paste output from your computer, or attach screenshots.

Gaab88, 2024-03-23:
Hi!

I took 3 pictures of the states that I receive messages about.
I am attaching 2 txts, one of which contains the Fluidnc configuration and the other the message of the Webinstaller terminal.

The MCU used for the DisplayToTerminal project cannot read any data, so I did not take a picture of it.

The two units were connected according to the description, together with resistors.

[image]
[image]
[image]
[config.txt](https://github.com/bdring/FluidNC/files/14730659/config.txt)
[fluidnc_web_terminal.txt](https://github.com/bdring/FluidNC/files/14730690/fluidnc_web_terminal.txt)

(The configuration file is of a test nature, since I have to design my own panel, so my Fluidnc motherboard is assembled on a test panel. This panel contains 4 buttons with which I imitate the control and status indicator LEDs)

MitchBradley, 2024-03-23:
Your config file has this:
```yaml
control:
  safety_door_pin: gpio.34:low
```
If gpio 34 is at a low level, or is picking up electrical noise that makes it go low periodically, FluidNC will issue the safety door message and go into safety door lockout state.

This issue is an old one that was started by someone else.  Most of its commentary is related to the other person's situation.  Please start a new issue and provide all of the information there, so we can track your specific situation.

closed 2024-03-23 by MitchBradley