bdring/FluidNC #772: Problem:

Nossolov, 2023-01-12:
### Controller Board

Wemos ESP32 + CNC shield V3.0
Fluidnc 3.2.7 and 3.6.5

### Machine Description

DIY Router like CNC 3018

### Configuration file

```yaml
name: "ESP32 cnc shield v3"
board: "ESP32 cnc shield v3"

stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 1
  pulse_us: 2
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.12:low

  x:
    steps_per_mm: 35.555
    max_rate_mm_per_min: 360
    acceleration_mm_per_sec2: 5
    max_travel_mm: 360
    soft_limit: true
    homing:
      cycle: 2
      allow_single_axis: true
      mpos_mm: 180
      positive_direction: false

    motor0:
      limit_neg_pin: gpio.32:low:pu
      stepstick:
        direction_pin: gpio.16
        step_pin: gpio.26
    motor1:
      null_motor:

  y:
    steps_per_mm: 35.555
    max_rate_mm_per_min: 360
    acceleration_mm_per_sec2: 5
    max_travel_mm: 180
    soft_limit: true
    homing:
      cycle: 2
      allow_single_axis: true
      mpos_mm: 90
      positive_direction: false

    motor0:
      limit_all_pin: gpio.33:low:pu
      stepstick:
        direction_pin: gpio.27
        step_pin: gpio.25
    motor1:
      null_motor:

  z:
    steps_per_mm: 200
    max_rate_mm_per_min: 500
    acceleration_mm_per_sec2: 25
    max_travel_mm: 400
    homing:
      cycle: 1
      mpos_mm: 10
      positive_direction: true

    motor0:
      limit_all_pin: gpio.15:low:pu
      stepstick:
        direction_pin: gpio.14
        step_pin: gpio.17
    motor1:
      null_motor:

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

coolant:
  flood_pin: NO_PIN
  mist_pin:  NO_PIN

probe:
  pin: gpio.39:low

PWM:
  pwm_hz: 5000
  output_pin: gpio.21
  enable_pin: gpio.22
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0% 10000=100%

control:
  safety_door_pin: NO_PIN
  reset_pin: gpio.34
  feed_hold_pin: gpio.36
  cycle_start_pin: gpio.35
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.2.7]
[MSG:INFO: Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:INFO: Configuration file:SolarTracker_config.yaml]
[MSG:WARN: Ignored key planner_blocks]
[MSG:WARN: Ignored key soft_limit]
[MSG:WARN: Ignored key allow_single_axis]
[MSG:WARN: Ignored key soft_limit]
[MSG:WARN: Ignored key allow_single_axis]
[MSG:INFO: Machine ESP32 cnc shield v3]
[MSG:INFO: Board ESP32 cnc shield v3]
[MSG:INFO: SPI not defined]
[MSG:ERR: SD needs SPI defined]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.12:low]
[MSG:INFO: Axis X (180.000,540.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     Standard Stepper Step:gpio.26 Dir:gpio.16 Disable:NO_PIN]
[MSG:INFO:     Neg Limit gpio.32:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (90.000,270.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     Standard Stepper Step:gpio.25 Dir:gpio.27 Disable:NO_PIN]
[MSG:INFO:     All Limit gpio.33:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-390.000,10.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     Standard Stepper Step:gpio.17 Dir:gpio.14 Disable:NO_PIN]
[MSG:INFO:     All Limit gpio.15:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: Reset gpio.34]
[MSG:INFO: FeedHold gpio.36]
[MSG:INFO: CycleStart gpio.35]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:gpio.22 Out:gpio.21 Dir:NO_PIN Freq:5000Hz Res:13bits]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Probe Pin: gpio.39:low]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.2 [FluidNC v3.2.7 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
```

### User Interface Software

I use my own interface in Labview, but same pb with Fluidterm

### What happened?

I send G1 X10 Fxx cmde. It is working well for Feedrate > 60. If i use Feedrate < 50 / 60, everything seems to be ok, status report give the good value  of feedrate, axis go to the right place, but too fast. With fluidterm, it take about 10 / 12s for 10mm travel on X & Y at feedrate 3mm/min. After many tries, i find that real minimum feedrate is between 50 and 60mm/min. If you want more, it's ok, but whatever you want less, feedrate stay at 50/60 mm/min. Same with Z, but not with the same minimum real value of feedrate (about 10 to 12mm/min). X & Y have the same mechanical config, not Z. May be it's important.
Same limitation if i use feedrate override during travel.
Sure i do something wrong, but i don't see what.

MitchBradley, 2023-01-12:
60 mm/min is 1 mm/sec.  At 200 steps/mm, that is one step every 5 milliseconds.  The GRBL stepping engine inside FluidNC has a maximum segment time of a few milliseconds, and must do at least one step in each segment.

If you need to go slower, one alternative is to use drivers with a high degree of microstepping, so steps/mm is higher.  Another alternative is mechanical gearing, again to increase step/mm.

MitchBradley, 2023-01-12:
For a special application that needs to go very slow it would probably be better to write your own program. Acceleration is unlikely to be necessary at such speeds. For each motor, just calculate the time to the next step and wait until the next step time.

MitchBradley, 2023-01-12:
Can we close this?

Nossolov, 2023-01-12:
Yes, I close