bdring/FluidNC #775: Problem: Compiling with oled basic errors out on infile not declared in scope. line 161

Rad3d21, 2023-01-14:
### Controller Board

ESP32 DIY board. 

### Machine Description

DIY laser board based on ESP32.

### Configuration file

```yaml
Not relevant. Compiling is not working.
```

### Startup Messages

```text
Not relevant. Compiling is not working.
```

### What happened?

Tried compiling the software using visual studio code & platform IO. Uncommented the relevant lines to  include OLED support for the basic oled 0.96".  Got an error 
FluidNC/src/Custom/oled_basic.cpp:161:13: error: 'infile' was not declared in this scope

robercy, 2023-01-14:
Hello !

More info about this problem:

-----------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (4.4.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20003.220626 (2.0.3) 
 - tool-esptoolpy @ 1.30300.0 (3.3.0) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 36 compatible libraries
Scanning dependencies...
Dependency Graph
|-- TMCStepper @ 0.7.3
|   |-- SPI @ 2.0.0
|-- ArduinoOTA @ 2.0.0
|   |-- Update @ 2.0.0
|   |-- WiFi @ 2.0.0
|   |-- ESPmDNS @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|-- DNSServer @ 2.0.0
|   |-- WiFi @ 2.0.0
|-- ESPmDNS @ 2.0.0
|   |-- WiFi @ 2.0.0
|-- Update @ 2.0.0
|-- WebServer @ 2.0.0
|   |-- WiFi @ 2.0.0
|   |-- FS @ 2.0.0
|-- WiFi @ 2.0.0
Compiling .pio\build\wifi-oled-basic\src\src\Configuration\JsonGenerator.cpp.o
Compiling .pio\build\wifi-oled-basic\src\src\Configuration\LegacySettingRegistry.cpp.o
Compiling .pio\build\wifi-oled-basic\src\src\Configuration\Parser.cpp.o
Compiling .pio\build\wifi-oled-basic\src\src\Configuration\RuntimeSetting.cpp.o
Compiling .pio\build\wifi-oled-basic\src\src\Configuration\Tokenizer.cpp.o
Compiling .pio\build\wifi-oled-basic\src\src\Configuration\Validator.cpp.o
Compiling .pio\build\wifi-oled-basic\src\src\Control.cpp.o
Compiling .pio\build\wifi-oled-basic\src\src\ControlPin.cpp.o
Compiling .pio\build\wifi-oled-basic\src\src\CoolantControl.cpp.o
Compiling .pio\build\wifi-oled-basic\src\src\CustomCode.cpp.o
Compiling .pio\build\wifi-oled-basic\src\src\Custom\oled_basic.cpp.o
Compiling .pio\build\wifi-oled-basic\src\src\Custom\oled_io.cpp.o
Compiling .pio\build\wifi-oled-basic\src\src\Custom\oled_tiny.cpp.o
Compiling .pio\build\wifi-oled-basic\src\src\Error.cpp.o
Compiling .pio\build\wifi-oled-basic\src\src\FileStream.cpp.o
Compiling .pio\build\wifi-oled-basic\src\src\FluidPath.cpp.o
Compiling .pio\build\wifi-oled-basic\src\src\GCode.cpp.o
Compiling .pio\build\wifi-oled-basic\src\src\I2SOut.cpp.o
Compiling .pio\build\wifi-oled-basic\src\src\InputFile.cpp.o
Compiling .pio\build\wifi-oled-basic\src\src\Jog.cpp.o
Compiling .pio\build\wifi-oled-basic\src\src\Kinematics\Cartesian.cpp.o
Compiling .pio\build\wifi-oled-basic\src\src\Kinematics\CoreXY.cpp.o
Compiling .pio\build\wifi-oled-basic\src\src\Kinematics\Kinematics.cpp.o
Compiling .pio\build\wifi-oled-basic\src\src\Kinematics\Midtbot.cpp.o
Compiling .pio\build\wifi-oled-basic\src\src\Kinematics\WallPlotter.cpp.o
Compiling .pio\build\wifi-oled-basic\src\src\Limits.cpp.o
Compiling .pio\build\wifi-oled-basic\src\src\Logging.cpp.o
Compiling .pio\build\wifi-oled-basic\src\src\Machine\Axes.cpp.o
Compiling .pio\build\wifi-oled-basic\src\src\Machine\Axis.cpp.o
Compiling .pio\build\wifi-oled-basic\src\src\Machine\EventPin.cpp.o
Compiling .pio\build\wifi-oled-basic\src\src\Machine\Homing.cpp.o
Compiling .pio\build\wifi-oled-basic\src\src\Machine\I2SOBus.cpp.o
Compiling .pio\build\wifi-oled-basic\src\src\Machine\LimitPin.cpp.o
Compiling .pio\build\wifi-oled-basic\src\src\Machine\MachineConfig.cpp.o
Compiling .pio\build\wifi-oled-basic\src\src\Machine\Macros.cpp.o
Compiling .pio\build\wifi-oled-basic\src\src\Machine\Motor.cpp.o
Compiling .pio\build\wifi-oled-basic\src\src\Machine\SPIBus.cpp.o
FluidNC/src/Custom/oled_basic.cpp: In function 'void oledUpdate(void*)':
FluidNC/src/Custom/oled_basic.cpp:161:13: error: 'infile' was not declared in this scope
         if (infile) {
             ^~~~~~
FluidNC/src/Custom/oled_basic.cpp:161:13: note: suggested alternative: 'init'
         if (infile) {
             ^~~~~~
             init
*** [.pio\build\wifi-oled-basic\src\src\Custom\oled_basic.cpp.o] Error 1
========================================================= [FAILED] Took 7.67 seconds =========================================================

Environment      Status    Duration
---------------  --------  ------------
wifi-oled-basic  FAILED    00:00:07.671
==================================================== 1 failed, 0 succeeded in 00:00:07.671 ====================================================

Thanks.

MitchBradley, 2023-01-14:
The recent input channel changes broke oled-basic.  I will fix it when I get time.

bdring, 2023-01-14:
This was expected. The channel refactor was to make things like displays work in a new way.

robercy, 2023-01-14:
Nice to hear, this bad behavior is for a good cause.
Keep this great piece of software, growing! Take your time.

robercy, 2023-01-17:
Hi, Mitch.

Compiled, loaded and running like a charm. Nice job.  Thanks.

Rad3d21, 2023-01-18:
It works with the mentioned OLED branch.

PR #777 (2023-02-26): Channel-based I2C_OLED
This is a start.  It needs some work to display limit switches and whatnot.

It is configurable via:

```
i2c0:
  sda_pin: gpio.14
  scl_pin: gpio.13
  frequency: 400000

oled:
  i2c_num: 0
  i2c_address: 60
  width: 128
  height: 64
  radio_delay_ms: 2000
```
You can also have an i2c1 bus.

The i2c_address 60 is 0x3c in hex.  That is the default value so it can be omitted.
The display layout is hardcoded, so changing the width and height is probably not helpful.

This might resolve #775
closed 2023-01-18 by Rad3d21