bdring/FluidNC #778: FluidNC 3.6.6 don't work with bCNC

RDS356, 2023-01-16:
### Controller Board

RDS ESP32

### Machine Description

Laser CO2 80 Watts, DM452 and Router With spindle 2.2kW.

### Configuration file

```yaml
name: "RDS Laser - Sergio"
board: "RDS ESP32 V.1.1"
# Oct 2022 by Rei

kinematics:
  Cartesian:

stepping:
  engine: I2S_STATIC
  idle_ms: 255
  dir_delay_us: 4
  pulse_us: 4
  disable_delay_us: 0
  
axes:
  shared_stepper_disable_pin: gpio.27
  x:
    steps_per_mm: 240.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 255.000
    max_travel_mm: 815.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 500.000
      seek_mm_per_min: 1500.000
      settle_ms: 250
      seek_scaler: 1.000
      feed_scaler: 1.000

    motor0:      
      limit_neg_pin: gpio.33      
      hard_limits: false
      pulloff_mm: 2.000
      standard_stepper:
       step_pin: I2SO.1
       direction_pin: I2SO.2
       
  y:
    steps_per_mm: 240.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 255.000
    max_travel_mm: 1425.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 500.000
      seek_mm_per_min: 1500.000
      settle_ms: 250
      seek_scaler: 1.000
      feed_scaler: 1.000

    motor0:      
      limit_neg_pin: gpio.32      
      hard_limits: false
      pulloff_mm: 2.000
      standard_stepper:
       step_pin: I2SO.5
       direction_pin: I2SO.6
  
  z:
    steps_per_mm: 18.957
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 255.000
    max_travel_mm: 720.000
    soft_limits: false
    homing:
      cycle: 0
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 1000.000
      settle_ms: 100
      seek_scaler: 1.000
      feed_scaler: 1.000

    motor0:      
      limit_neg_pin: gpio.35    
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
       step_pin: I2SO.7
       direction_pin: I2SO.0
       
i2so:
  ws_pin: gpio.12
  bck_pin: gpio.13
  data_pin: gpio.14  

#spi:
#  miso_pin: gpio.19
#  mosi_pin: gpio.23
#  sck_pin: gpio.18

#sdcard:
#  cs_pin: gpio.5
#  card_detect_pin: NO_PIN

control:
  safety_door_pin: gpio.25
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:

coolant:
  flood_pin: gpio.15
  mist_pin: gpio.21
  delay_ms: 0

probe:
  pin: gpio.26
  check_mode_start: true
  
pwm:
  pwm_hz: 5000
  output_pin: gpio.2
  enable_pin: gpio.4
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0% 100=100%

start:
  must_home: false
  deactivate_parking: true
  check_limits: true
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.6.6]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:RDS_esp32_laser.yaml]
[MSG:INFO: Machine RDS Laser - Sergio]
[MSG:INFO: Board RDS ESP32 V.1.1]
[MSG:INFO: I2SO BCK:gpio.13 WS:gpio.12 DATA:gpio.14]
[MSG:INFO: SPI not defined]
[MSG:INFO: No SD Card CS Pin]
[MSG:INFO: See http://wiki.fluidnc.com/en/config/sd_card#sdfallbackcs-access-sd-without-a-config-file]
[MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:0us Dir Delay:4us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.27]
[MSG:INFO: Axis X (0.000,815.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.1 Dir:I2SO.2 Disable:NO_PIN]
[MSG:INFO:  X Neg Limit gpio.33]
[MSG:INFO: Axis Y (0.000,1425.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.5 Dir:I2SO.6 Disable:NO_PIN]
[MSG:INFO:  Y Neg Limit gpio.32]
[MSG:INFO: Axis Z (0.000,720.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.7 Dir:I2SO.0 Disable:NO_PIN]
[MSG:INFO:  Z Neg Limit gpio.35]
[MSG:INFO: safety_door_pin gpio.25]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:gpio.4 Out:gpio.2 Dir:NO_PIN Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Flood coolant gpio.15]
[MSG:INFO: Mist coolant gpio.21]
[MSG:INFO: Probe Pin: gpio.26]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
```

### User Interface Software

bCNC

### What happened?

bCNC is my preferred , but didn't  work with new release 3.6.6.
I tried with other config file without sucess.
Am i making something wrong ?

[image]

kdomanski, 2023-01-16:
Looks more like bCNC doesn't work.

RDS356, 2023-01-16:
bCNC Works well with 3.6.5 and early.

bdring, 2023-01-16:
That is a python traceback from a crashing bCNC. I would ask for support from them. There is likely a parsing error from some subtle, but probably legal, change in the FluidNC output.

aripyk, 2023-01-17:
I have exactly the same problem.

ipsod, 2023-01-17:
Here's a link to their code.  Section is commented `# Update G variables from "G" string`.  Looks like they're expecting an int after T, but instead there is nothing.

https://github.com/vlachoudis/bCNC/blob/master/bCNC/CNC.py#L783

MitchBradley, 2023-01-17:
@ipsod - thanks for finding that.  The fix was trivial once I knew where to look.

closed 2023-01-29 by bdring

MitchBradley, 2023-01-29:
See also https://github.com/cncjs/cncjs/issues/805

commit 0e7730df (2023-01-27): Merge pull request #780 from bdring/Fix778

Fix #778 - T field in GC report was missing the integer
commit 886531b6 (2023-01-27): Fix #778 - T field in GC report was missing the integer