bdring/FluidNC #779: Web UI on widescreen, G20 features misaligned

maxcel2, 2023-01-17:
### Controller Board

N/a

### Machine Description

N/a

### Configuration file

```yaml
N/a
```

### Startup Messages

```text
N/a
```

### User Interface Software

WebUI

### What happened?

When the WebUI is viewed on a wide screen, the buttons/boxes in the G20 panel are all out of position and don't make any logical sense. If the width of the window is progressively reduced, the buttons start to compact to logical positions. There appears to be an optimum screen x resolution for the x, y & z to be in the correct positions, but then the text in other buttons is truncated.

MitchBradley, 2023-01-17:
What is the "G20 panel".  I see no panel with that name or label.

bdring, 2023-04-27:
Closing stale, incomplete issue

closed 2023-04-27 by bdring