bdring/FluidNC #781: Feature: Add outputs to indicate status: Idle/Run/Halt/Alarm

errolt, 2023-01-18:
### Machine Context

Useful for:
Adding a fan to cool the steppers when the machine is not in idle mode(Idle pin, active low)
Adding a cnc signal tower(green light,yellow light, red light)

### Feature Description

If machine is in idle state, set IDLE pin, all other status pins unset
If machine is in running state set RUNNING pin, all others status pins unset
etc
etc

### Other Approaches

Can be done with an external controller monitoring fluidnc's output, but doing in in fluidnc would be a lot cleaner.

### How I Can Help

Can try to do the code, but have not looked at the fluidnc code framework yet.

mani-fa, 2023-01-18:
This one "Adding a fan to cool the steppers when the machine is not in idle mode(Idle pin, active low)" can be done with coolant or mist and they are already in FluidNC code, You can use them for any purposes like cooling steppers or controlling a fan .

errolt, 2023-01-18:
> This one "Adding a fan to cool the steppers when the machine is not in idle mode(Idle pin, active low)" can be done with coolant or mist and they are already in FluidNC code, You can use them for any purposes like cooling steppers or controlling a fan .

That would rely on gcode at the start/end of the run to turn the fan/mist on. If it is missing then hardware damage can occur, and with the spindle running it can be hard to verify that the fan is running.
And stopping the gcode would leave the fan running.

I'll look at the code to see if I can add it myself, but at this point I don't know where would be best.

sjonholle, 2023-01-18:
My DRV8825 are controlled by the disable_pin in FluidNC. This pin also controls via a transistor, the fan.

errolt, 2023-01-19:
> My DRV8825 are controlled by the disable_pin in FluidNC. This pin also controls via a transistor, the fan.

That is a really good idea. When I design my own board I'll keep that in mind and add a dedicated fan header. I'm currently using an MKS DLC32 though, and adding a FET would be a huge pain.

I have added a StatusOutput class(with all 10 status types) to FluidNC, but I'm struggling to find the appropriate place to update the status pins. I've added it to the protocol_main_loop, and it works for Idle/Jog/Alarm. But Home and Hold seems to not loop in protocol_main_loop, and the pin status does not update.

closed 2023-01-23 by errolt

wu5, 2023-02-16:
I don't know why the issue was closed. 

I like the idea of lamps showing Alarm status. I would like to have such a feature in FluidNC.

I don't know why the issue was prematurely closed. I like the idea of indicator lamps showing alarm status. I would like to have such a feature in FluidNC.

MitchBradley, 2023-02-16:
With https://github.com/bdring/FluidNC/pull/787 , people can add auxiliary UART-connected devices that can monitor the status.

PatrikRindlisbacher, 2023-02-16:
I started a project that could fulfill such wishes. I will put the first version online in about 1-2 weeks.

https://github.com/PatrikRindlisbacher/FluidNC-or-GRBL-UART-Gateway-an-IO-Module-

errolt, 2023-02-17:
> I don't know why the issue was prematurely closed.
> 
> I like the idea of lamps showing Alarm status. I would like to have such a feature in FluidNC.

Because:

1. Everyone told me to do it with extra/external hardware, or extra gcode.
2. I couldn't find a place in the code to get the status under all circumstances. 

I have a branch here: https://github.com/errolt/FluidNC/tree/StatusOutput
It added outputs for all the possible states, but as mentioned earlier in this thread, HOME and HOLD doesn't work.
Due to those states not working, I never did a merge/pull request.

PatrikRindlisbacher, 2023-02-17:
Hello errollt
I recently took up this topic. In my view, the matter should be divided into two parts.

- FLUID NC reports the correct status via the URAT
- Individual hardware for display

This division is important so that FluidNC is not overloaded.

PatrikRindlisbacher, 2023-02-19:
> Because:
>
> 1. Everyone told me to do it with extra/external hardware, or extra gcode.
> 2. I couldn't find a place in the code to get the status under all circumstances.
>
> I have a branch here: [...]

Hi errolt --> I'm busy working on your wish. get in touch with me.

PatrikRindlisbacher, 2023-02-20:
The first version is ready.

https://github.com/PatrikRindlisbacher/grbl-Status-light-tower-M5-Atom-Matrix