bdring/FluidNC #782: Crashing when disconnect from WiFi & random move with micro pin

mani-fa, 2023-01-18:
### Controller Board

2 Axis pen/Laser

### Machine Description

2 Axis pen/Laser with a4988 driver

### Input Circuits

```yaml
for input pin I just put a push button for "macro0_pin: gpio.34:low" to run "1.nc" file from SD, which I use it as a reset pin before reconfiguration.
```

### Configuration file

```yaml
name: "ESP32 Pen/Laser V3"
board: "ESP32 Pen/Laser V3"
meta: M@N!_F@

stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 1
  pulse_us: 2
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.13
  
  x:
    steps_per_mm: 80
    max_rate_mm_per_min: 12000
    acceleration_mm_per_sec2: 1000
    max_travel_mm: 320
    homing:
      cycle: 2
      mpos_mm: 10
      positive_direction: false
    
    motor0:
      limit_all_pin: gpio.15:low:pu
      stepstick:
        direction_pin: gpio.26
        step_pin: gpio.12
    motor1:
      null_motor:

  y:
    steps_per_mm: 80
    max_rate_mm_per_min: 12000
    acceleration_mm_per_sec2: 1000
    max_travel_mm: 240
    homing:
      cycle: 2
      mpos_mm: 10
      positive_direction: false

    motor0:
      limit_all_pin: gpio.4:low:pu
      stepstick:
        direction_pin: gpio.25
        step_pin: gpio.14
    motor1:
      null_motor:

  z:
    steps_per_mm: 400
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 500
    max_travel_mm: 5.00
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 5
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100
    
    motor0:
      rc_servo:
        pwm_hz: 50
        output_pin: gpio.27
        min_pulse_us: 2100
        max_pulse_us: 1000

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: gpio.36:low
  cycle_start_pin: gpio.39:low
  macro0_pin: gpio.34:low
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

coolant:
  flood_pin: NO_PIN
  mist_pin:  NO_PIN
  delay_ms: 0
        
probe:
  pin: NO_PIN

macros:
  startup_line0: 
  startup_line1:
  macro0: $sd/run=1.nc
  macro1:
  macro2:
  macro3:

start:
  must_home: false

Laser:
  pwm_hz: 1000
  output_pin: gpio.17
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0% 1000=100%
```

### Startup Messages

```text
rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12812
load:0x40080400,len:3032
entry 0x400805e4

[MSG:INFO: FluidNC v3.6.6]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:pen_laser_v3.yaml]
[MSG:WARN: Ignored key direction_pin]
[MSG:WARN: Ignored key spinup_ms]
[MSG:WARN: Ignored key spindown_ms]
[MSG:INFO: Machine ESP32 Pen/Laser V3]
[MSG:INFO: Board ESP32 Pen/Laser V3]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.13]
[MSG:INFO: Axis X (10.000,330.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.12 Dir:gpio.26 Disable:NO_PIN]
[MSG:INFO:  X All Limit gpio.15:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (10.000,250.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.14 Dir:gpio.25 Disable:NO_PIN]
[MSG:INFO:  Y All Limit gpio.4:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (0.000,5.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     rc_servo Pin:gpio.27 Pulse Len(2100,1000 period:1048575)]
[MSG:INFO: feed_hold_pin gpio.36:low]
[MSG:INFO: cycle_start_pin gpio.39:low]
[MSG:INFO: macro0_pin gpio.34:low]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Laser Ena:NO_PIN Out:gpio.17 Freq:1000Hz Period:65535]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID ESP32.T-Laser IP 192.168.4.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 8080]

Grbl 3.6 [FluidNC v3.6.6 (wifi) '$' for help]
```

### User Interface Software

WebUI , LasrGrbl

### What happened?

1. everything works ok when connecting and selecting file from sd card but after selecting a file to work ,if before closing browser I disconnect wifi , then esp32 stop working and freezes (On PC or Phone) , re-connecting may solve the problem but sometimes it crashes and I cannot connect to it , reboot required to back to normal state.

2. another problem is when connecting again to working machine , it freeses for a second.

3.I put a push button to run a file from sd card. When It runs and done ,I push it again to do job again, but somehow it crashes and just process bar never ended it goes up and down and just do some gcode randomly. (maybe after 4 or 5 time that I press it)

4. MKSlaser and Atomstack app for android works Ok with FluidNc , I can control and take picture and upload it to SD card , after uploading it ask to start working and by confirming, it shows proccess bar , pause and stop works as well , only part that it worked before and now not supported is showing files on SD card. with these apps anything can be done over phone.

MitchBradley, 2023-01-29:
Foiled again by the Arduino Framework:

framework-arduinoespressif32/cores/esp32/esp-hal-gpio.c:
 ```c
extern void ARDUINO_ISR_ATTR __digitalWrite(uint8_t pin, uint8_t val)
{
	gpio_set_level((gpio_num_t)pin, val);
}
```
framework-arduinoespressif32/cores/esp32/esp-hal.h:
```c
#if CONFIG_ARDUINO_ISR_IRAM
#define ARDUINO_ISR_ATTR IRAM_ATTR
#define ARDUINO_ISR_FLAG ESP_INTR_FLAG_IRAM
#else
#define ARDUINO_ISR_ATTR
#define ARDUINO_ISR_FLAG (0)
#endif
```
framework-arduinoespressif32/tools/sdk/esp32/sdkconfig:
```
# CONFIG_ARDUINO_ISR_IRAM is not set
```

mani-fa, 2023-01-30:
I test all problems with 3.6.7 but issues are still there

bdring, 2023-01-30:
The fix is in a branch, not 3.6.7

Prelo96, 2023-02-01:
> The fix is in a branch, not 3.6.7

Same problem and other issues but... where i can find the righ branch? Or simply a stable version? 
Can i have the discord link maybe i can find more inf myself...

bdring, 2023-02-01:
https://github.com/bdring/FluidNC/tree/Fix797

You will need to compile yourself

MitchBradley, 2023-02-01:
I made a test build - https://github.com/bdring/FluidNC/releases/tag/v3.6.8-pre1

mani-fa, 2023-02-01:
> I made a test build - https://github.com/bdring/FluidNC/releases/tag/v3.6.8-pre1

I test with 3.6.8-pre1, I don't know this fix is for my cases or not but problem not solved. I test uploading 17mb file for #797 issue and successfully uploaded.

Prelo96, 2023-02-02:
> I made a test build - https://github.com/bdring/FluidNC/releases/tag/v3.6.8-pre1

3.6.8 tested, seems to be ok, no crash at all but I loose some step like 1.605mm in 20 minute of working  I will try to increase step pulse and dir delay tomorrow cause was set to 2us, pretty strange cause with the 3.6.7 i got some crash but never loose a step...

Prelo96, 2023-02-04:
Tested again and no loosing steps. Maybe ground connection issue. 
Also no crash at all

mani-fa, 2023-02-04:
> Tested again and no loosing steps. Maybe ground connection issue. Also no crash at all

check case 1 please , I tested again and freeze happened again

Prelo96, 2023-02-05:
> 
Happened again, also a lot of error in z, loose 16mm during a parallel carving.
If I close the webui I have no problem, 
Crashed again when I pause and stop a work, starting to not respond to all commands and act really slow in responds then completely freeze, restarted 
Worst issue the loosing of stwp in z but I'm not really sure it caused by the firmware or interferences...

Prelo96, 2023-02-05:
Another things the tep loosing seems to be all at the same time or better In 2/3 times during the parallel.

MitchBradley, 2023-02-05:
@Prelo96 Start another issue with you own information instead of using this one.  We need the information about your configuration, not the original poster's information.

mani-fa, 2023-04-27:
Problem listed above has been solved in new release?
I did not test it yet.

bdring, 2023-04-27:
Fix797 is in the latest release.

mani-fa, 2023-04-27:
> Fix797 is in the latest release.

thanks
I mean #782 problem ,  disconnecting from wifi make machine to stop and random gcode send from micro key

MitchBradley, 2023-04-27:
What does "disconnect wifi" mean exactly.

MitchBradley, 2023-04-27:
In the future, please file a separate issue for each problem.  This issue has 5 separate problems listed, and another person also hijacked it, so figuring out what is going on is very confusing.

mani-fa, 2023-04-27:
> What does "disconnect wifi" mean exactly. Are you unplugging your wifi router, or telling your PC to disconnect from the wifi that it is associated with, or what?

I connect to FluidNc wifi via PC or Phone and from WebUI I select a file in SD card , machine start to run gcode , If my computer goes to sleep or I restart PC or any other way to cause my PC to disconnect from wifi like you mentioned "telling me PC to disconnect from the wifi" ,FluidNC stop streaming gcode

mani-fa, 2023-04-27:
> In the future, please file a separate issue for each problem. This issue has 5 separate problems listed, and another person also hijacked it, so figuring out what is going on is very confusing.

Yes
your Right, my mistake
I was testing FluidNC and face all these problems , 
I'll separate next time
sorry for this confusing listed issues

MitchBradley, 2023-04-28:
I cannot reproduce the "wifi disconnect freeze" problem.  I did this
* Loaded v3.7.0 onto a test machine based on a 6-pack board
* Started WebUI on a PC running Windows 11, under the Chrome browser
* Ran an SD test program consisting of 
```
g21
g91
g1 f5000
Y.1
<13000 more lines of Y.1>
```
* During the test run, I closed the WebUI browser window.  The test program continued tor run.
* I opened another browser window and reconnected to WebUI.  There was a pause in the motion while FluidNC responded to the browser's request for index.html.gz - as expected - then the motion resumed.
* I put the PC to sleep via the Windows start menu.  Motion continued
* I woke up Windows by typing Shift on the keyboard and logging back in.  Motion continued the entire time.

The pause in motion when reconnecting WebUI is caused by serving index.html.gz.  When you browse to the FluidNC IP address, the browser fetches the index.html.gz file - the file that implements the WebUI user interface - from the local FLASH filesystem.  The ESP32 also uses FLASH for fetching the program instructions.  When it is reading index.html.gz from FLASH and sending it to the browser, that competes with the processor for instruction cycles, so motion pauses or stutters during the time that FLASH filesystem access locks out the processor's instruction fetches.

mani-fa, 2023-04-28:
> I cannot reproduce the "wifi disconnect freeze" problem. I did this
>
> * Loaded v3.7.0 onto a test machine based on a 6-pack board
> * Started WebUI on a PC running Windows 11, under the Chrome browser [...]

"During the test run, I closed the WebUI browser window. The test program continued tor run."

you're right , closing WebUI before restart or put PC to sleep makes no problem
but consider this situation: WebUI is open and suddenly computer restarts or goes to sleep or somehow Wifi disconnects (in all examples WebUI is open)!
In this condition FluidNC keep running or stops?

miki310101, 2023-04-29:
У меня была похожая проблема. Запускаю G код через браузер смартфона, начинается работа, но через 30 секунд тайм аут экран гаснет и работа замирает до тех пор пока снова не активирую экран смартфона. Я решил её путем изменения в браузере  в окне GRBL Reports:   🔘None     🔘Auto 50ms    🔵Poll  3 sec . После этого при выключенном экране  работа продолжается.

mani-fa, 2023-04-29:
> /I had a similar issue. I launch the G code from the SD card through the smartphone browser, work starts, but after 30 seconds the screen goes blank and the work freezes until I activate the smartphone screen again. I solved it by changing in the browser in the window - GRBL Reports: 🔘None 🔘Auto 50ms 🔵Poll 3 sec . After that, when the screen is off, work continues.

let's try it with your settings

MitchBradley, 2023-04-29:
I have tracked the problem down to the Arduino websocket library stalling when FluidNC is trying to send a position report.  I may have a fix but I have some more work to do before I can release it.

MitchBradley, 2023-04-29:
Switching to non-auto status polling does help, at the expense of slower updates of the DRO and visualizer.

commit 05088727 (2023-05-02): Merge pull request #882 from bdring/Fix782

Fix #782 - auto-report websocket stalls when wifi goes down
commit fef6076e (2023-05-02): Fix #782 - auto-report websocket stalls when wifi goes down

In the process, I moved a lot of websocket code from WebServer.cpp
into WSChannel.cpp because WebSocket.cpp is too complex.
closed 2023-06-08 by bdring